respeaker/seeed-voicecard #190: dkms build error against 4.19.75-v8+ header, latest Buster

HinTak, 2019-10-11:
The raspberrypi people has finally put out the matching kernel header packages yesterday afternoon:

```
ii  raspberrypi-bootloader     1.20190925-2 armhf        Raspberry Pi bootloader
ii  raspberrypi-kernel         1.20190925-2 armhf        Raspberry Pi bootloader
ii  raspberrypi-kernel-headers 1.20190925-2 armhf        Header files for the Raspberry Pi Linux kernel
```

but it seems to have some incompatibility with the array driver code:

```
In file included from ./include/asm-generic/int-ll64.h:11,
                 from ./arch/arm/include/uapi/asm/types.h:5,
                 from ./include/uapi/linux/types.h:5,
                 from ./include/linux/types.h:6,
                 from ./include/linux/list.h:5,
                 from ./include/linux/module.h:9,
                 from /var/lib/dkms/seeed-voicecard/0.3/build/wm8960.c:13:
./include/uapi/asm-generic/int-ll64.h:12:10: fatal error: asm/bitsperlong.h: No such file or directory
 #include <asm/bitsperlong.h>
          ^~~~~~~~~~~~~~~~~~~
compilation terminated.
```

I think I can probably fix this, so stay tuned.

HinTak, 2019-10-11:
Encountering problem with 1.20190925-2 as described in
https://github.com/raspberrypi/linux/issues/3279

Pillar1989, 2019-10-11:
@HinTak  You are too great.

It's hard to believe that the raspberry faction keeps having this problem.

HinTak, 2019-10-11:
Following the instructions in 
https://forum.loverpi.com/discussion/555/how-to-fix-dkms-error-bin-sh-1-scripts-basic-fixdep-exec-format-error ,
and do all 3:

```
cd /usr/src/linux-headers-4.19.75+/
...
cd /usr/src/linux-headers-4.19.75-v7+/
...
cd  /usr/src/linux-headers-4.19.75-v7l+/
...
```

do the job.

There are some other issues with `/usr/src/linux-headers-4.19.75-v8+/`, but the respeaker code is not configured to build against that at the moment, so that's not important for the time being.

HinTak, 2019-10-11:
Just confirming the procedure outlined earlier worked - got a recording and play back, and stuff. This can close properly when the upstream bug ( https://github.com/raspberrypi/linux/issues/3279 ) get fixed. For now, there is a functional workaround as detailed earlier.

HinTak, 2019-10-14:
https://github.com/raspberrypi/linux/issues/3279 was fixed with 1.20190925+1-1

closed 2019-10-14 by HinTak