respeaker/seeed-voicecard #191: v8+ build

HinTak, 2019-10-11:
The top level install script does not use the v8+ headers (only base, base-v7+, and base v7l+). But trying to do so, for completeness, result in this:

```
/usr/src/linux-headers-4.19.75-v8+ $ sudo make KERNELRELEASE=4.19.75-v8+ -C /lib/modules/4.19.75-v8+/build M=/var/lib/dkms/seeed-voicecard/0.3/build
make: Entering directory '/usr/src/linux-headers-4.19.75-v8+'
  CC [M]  /var/lib/dkms/seeed-voicecard/0.3/build/wm8960.o
In file included from ./arch/arm/include/asm/unistd.h:16,
                 from ./include/uapi/linux/unistd.h:8,
                 from ./include/asm-generic/seccomp.h:14,
                 from ./arch/arm/include/generated/asm/seccomp.h:1,
                 from ./include/linux/seccomp.h:14,
                 from ./include/linux/sched.h:21,
                 from ./include/linux/ratelimit.h:6,
                 from ./include/linux/device.h:25,
                 from ./include/linux/i2c.h:30,
                 from /var/lib/dkms/seeed-voicecard/0.3/build/wm8960.c:19:
./arch/arm/include/uapi/asm/unistd.h:21:10: fatal error: asm/unistd-eabi.h: No such file or directory
 #include <asm/unistd-eabi.h>
          ^~~~~~~~~~~~~~~~~~~
compilation terminated.
make[1]: *** [scripts/Makefile.build:304: /var/lib/dkms/seeed-voicecard/0.3/build/wm8960.o] Error 1
make: *** [Makefile:1519: _module_/var/lib/dkms/seeed-voicecard/0.3/build] Error 2
make: Leaving directory '/usr/src/linux-headers-4.19.75-v8+'
```

This seems to be a separate issue from 
https://github.com/respeaker/seeed-voicecard/issues/190 , so filing separate.

I know you don't currently build against v8+ at the moment.

turmary, 2019-10-23:
The lastest package raspberrypi-kernel-headers (1.20190925+1-1) doesn't contain v8+ headers,
Where did you find it?

HinTak, 2019-10-23:
You are right - I had the earlier 1.20190925-2 and did a bit of hacking around that (see earlier threads in the other issue). There are some left-over bits on it:

du -s /usr/src/*
86912 /usr/src/linux-headers-4.19.75+
87064 /usr/src/linux-headers-4.19.75-v7+
87324 /usr/src/linux-headers-4.19.75-v7l+
800 /usr/src/linux-headers-4.19.75-v8+
1188 /usr/src/seeed-voicecard-0.3

The v8+ headers were available for 4.19.66 also. It seems that v8+ is 64-bit and substantially different, so it may not be well-supported yet.

closed 2020-01-15 by Pillar1989