respeaker/seeed-voicecard #192: Error! Bad return status for module build on kernel: 4.19.75-v7l+ (armv7l)

gregorek123, 2019-10-11:
Hi,

I am unable to install ReSpeaker 2-mic array on Raspberry PI 4. 
When running install script, I get:

`0 upgraded, 0 newly installed, 0 to remove and 10 not upgraded.
Error! There are no instances of module: seeed-voicecard
0.3 located in the DKMS tree.

Creating symlink /var/lib/dkms/seeed-voicecard/0.3/source ->
                 /usr/src/seeed-voicecard-0.3

DKMS: add completed.

Kernel preparation unnecessary for this kernel.  Skipping...

Building module:
cleaning build area...
make -j4 KERNELRELEASE=4.19.75+ -C /lib/modules/4.19.75+/build M=/var/lib/dkms/seeed-voicecard/0.3/build.....(bad exit status: 2)
Error! Bad return status for module build on kernel: 4.19.75+ (armv7l)
Consult /var/lib/dkms/seeed-voicecard/0.3/build/make.log for more information.

Kernel preparation unnecessary for this kernel.  Skipping...

Building module:
cleaning build area...
make -j4 KERNELRELEASE=4.19.75-v7+ -C /lib/modules/4.19.75-v7+/build M=/var/lib/dkms/seeed-voicecard/0.3/build.....(bad exit status: 2)
Error! Bad return status for module build on kernel: 4.19.75-v7+ (armv7l)
Consult /var/lib/dkms/seeed-voicecard/0.3/build/make.log for more information.

Kernel preparation unnecessary for this kernel.  Skipping...

Building module:
cleaning build area...
make -j4 KERNELRELEASE=4.19.75-v7l+ -C /lib/modules/4.19.75-v7l+/build M=/var/lib/dkms/seeed-voicecard/0.3/build........(bad exit status: 2)
Error! Bad return status for module build on kernel: 4.19.75-v7l+ (armv7l)
Consult /var/lib/dkms/seeed-voicecard/0.3/build/make.log for more information.
git init
Initialized empty Git repository in /etc/voicecard/.git/
git add --all
git commit -m "origin configures"
[master (root-commit) 66cd77f] origin configures
 7 files changed, 1476 insertions(+)
 create mode 100644 ac108_6mic.state
 create mode 100644 ac108_asound.state
 create mode 100644 asound_2mic.conf
 create mode 100644 asound_4mic.conf
 create mode 100644 asound_6mic.conf
 create mode 100644 dkms.conf
 create mode 100644 wm8960_asound.state
Created symlink /etc/systemd/system/sysinit.target.wants/seeed-voicecard.service → /lib/systemd/system/seeed-voicecard.service.
------------------------------------------------------
Please reboot your raspberry pi to apply all settings
Enjoy!
`

`pi@raspberrypi:~/seeed-voicecard $ ls /lib/modules/
4.19.75+  4.19.75-v7+  4.19.75-v7l+  4.19.75-v8+`

I have originally started with Latest Raspbian which produced below errors during installation:
`Error! echo
Your kernel headers for kernel 4.19.75+ cannot be found at
/lib/modules/4.19.75+/build or /lib/modules/4.19.75+/source.
Error! echo
Your kernel headers for kernel 4.19.75-v7+ cannot be found at
/lib/modules/4.19.75-v7+/build or /lib/modules/4.19.75-v7+/source.
Error! echo
Your kernel headers for kernel 4.19.75-v7l+ cannot be found at
/lib/modules/4.19.75-v7l+/build or /lib/modules/4.19.75-v7l+/source.
`
Then I deployed Raspbian from July with older kernel to sort out lack of latest kernel headers.
Now I have latest kernel headers:
`ii  raspberrypi-kernel                    1.20190925-2                          armhf        Raspberry Pi bootloader
ii  raspberrypi-kernel-headers            1.20190925-2                          armhf        Header files for the Raspberry Pi Linux kernel`
but I'm still not able to install respeaker (see output from beginning of the post).

`pi@raspberrypi:~/seeed-voicecard $ cat /var/lib/dkms/seeed-voicecard/0.3/build/make.log
DKMS make.log for seeed-voicecard-0.3 for kernel 4.19.75-v7l+ (armv7l)
Fri Oct 11 06:06:50 HDT 2019
make: Entering directory '/usr/src/linux-headers-4.19.75-v7l+'
  CC [M]  /var/lib/dkms/seeed-voicecard/0.3/build/wm8960.o
  CC [M]  /var/lib/dkms/seeed-voicecard/0.3/build/ac108.o
  CC [M]  /var/lib/dkms/seeed-voicecard/0.3/build/ac101.o
  CC [M]  /var/lib/dkms/seeed-voicecard/0.3/build/seeed-voicecard.o
/bin/sh: 1: scripts/basic/fixdep: Exec format error
/bin/sh: 1: scripts/basic/fixdep: Exec format error
make[1]: *** [scripts/Makefile.build:304: /var/lib/dkms/seeed-voicecard/0.3/build/seeed-voicecard.o] Error 2
make[1]: *** Waiting for unfinished jobs....
make[1]: *** [scripts/Makefile.build:304: /var/lib/dkms/seeed-voicecard/0.3/build/wm8960.o] Error 2
/bin/sh: 1: scripts/basic/fixdep: Exec format error
/bin/sh: 1: scripts/basic/fixdep: Exec format error
make[1]: *** [scripts/Makefile.build:304: /var/lib/dkms/seeed-voicecard/0.3/build/ac101.o] Error 2
make[1]: *** [scripts/Makefile.build:304: /var/lib/dkms/seeed-voicecard/0.3/build/ac108.o] Error 2
make: *** [Makefile:1519: _module_/var/lib/dkms/seeed-voicecard/0.3/build] Error 2
make: Leaving directory '/usr/src/linux-headers-4.19.75-v7l+'`

gregorek123, 2019-10-11:
Sorry I'm new to github. I found "Issues" tab :) and found solution under:
https://forum.loverpi.com/discussion/555/how-to-fix-dkms-error-bin-sh-1-scripts-basic-fixdep-exec-format-error

closed 2019-10-11 by gregorek123

HinTak, 2019-10-14:
It has been fixed upstream . See https://github.com/raspberrypi/linux/issues/3279

Ywyder, 2021-02-19:
The problem is that this version is not compatible with Kernel 5.10.11+
There are multiple pull requests, where the problem is fixed. 

I used this code and it worked. 
- https://github.com/pguyot/wm8960/pull/9

Probably this one works too, but i didn't test this one:
- https://github.com/waveshare/WM8960-Audio-HAT/pull/25

HinTak, 2021-02-19:
This works with 5.10 - https://github.com/HinTak/seeed-voicecard . If you have the 2-mics device, indeed there are alternatives; if you find my fork useful, consider clicking donate.

HinTak, 2021-02-19:
I don't / didn't put in a pull, as Ubuntu is at 5.4/ 5.8. But in my repo there are branches for older kernel versions.