respeaker/seeed-voicecard #193: Issue with latest snips version

Sothymos, 2019-10-11:
Hi,
I just did a fresh install of snips followed by the seeed-voicecard drivers. As soon as I install the drivers, the snips-audio-server sits around 70-90% cpu load all the time.

I'm using the respeaker 4 on a raspi 3 B+ with the recommended debian stretch:
```
pi@raspberrypi:~ $ uname -a
Linux raspberrypi 4.19.66-v7+ #1253 SMP Thu Aug 15 11:49:46 BST 2019 armv7l GNU/Linux
pi@raspberrypi:~ $ lsb_release -a
No LSB modules are available.
Distributor ID: Raspbian
Description:    Raspbian GNU/Linux 9.8 (stretch)
Release:        9.8
Codename:       stretch
```
And the latest snips
```
OS version ................... Raspbian GNU/Linux 9 (stretch)
Installed assistant .......... MyAssistant
Language ..................... de
Hotword ...................... hey_snips
ASR engine ................... snips
Status ....................... Live

Service status:

snips-asr .................... 0.64.0 (running)
snips-audio-server ........... 0.64.0 (running)
snips-dialogue ............... 0.64.0 (running)
snips-hotword ................ 0.64.0 (running)
snips-nlu .................... 0.64.0 (running)
snips-skill-server ........... 0.64.0 (running)
snips-tts .................... 0.64.0 (running)
```
When I downgrade the audio-server and tts to 0.63.3 the cpu load goes back to normal.

What I noticed running dmesg:
```
[  965.589278] OF: overlay: WARNING: memory leak will occur if overlay removed, property: /soc/i2c@7e804000/status
[  965.779389] i2c /dev entries driver
[  966.139315] OF: overlay: WARNING: memory leak will occur if overlay removed, property: /soc/spi@7e204000/status
[  967.949259] OF: overlay: WARNING: memory leak will occur if overlay removed, property: /soc/i2s@7e203000/#sound-dai-cells
[  967.949274] OF: overlay: WARNING: memory leak will occur if overlay removed, property: /soc/i2s@7e203000/status
[  967.949300] OF: overlay: WARNING: memory leak will occur if overlay removed, property: /soc/i2c@7e804000/status
[  967.949315] OF: overlay: WARNING: memory leak will occur if overlay removed, property: /soc/sound/compatible
[  967.949323] OF: overlay: WARNING: memory leak will occur if overlay removed, property: /soc/sound/seeed-voice-card,format
[  967.949328] OF: overlay: WARNING: memory leak will occur if overlay removed, property: /soc/sound/seeed-voice-card,name
[  967.949333] OF: overlay: WARNING: memory leak will occur if overlay removed, property: /soc/sound/status
[  967.949338] OF: overlay: WARNING: memory leak will occur if overlay removed, property: /soc/sound/seeed-voice-card,bitclock-master
[  967.949346] OF: overlay: WARNING: memory leak will occur if overlay removed, property: /soc/sound/seeed-voice-card,frame-master
[  967.949357] OF: overlay: WARNING: memory leak will occur if overlay removed, property: /soc/sound/seeed-voice-card,channels-playback-override
[  967.949363] OF: overlay: WARNING: memory leak will occur if overlay removed, property: /soc/sound/seeed-voice-card,channels-capture-override
[  967.967828] snd_soc_seeed_voicecard: loading out-of-tree module taints kernel.
[  967.990236]  ac10x i2c_id number: 0
[  967.990243]  ac10x data protocol: 0
[  968.118108]  ac10x codec count  : 1
[  968.118741] ac108_set_sysclk  :24000000
[  968.124721] seeed-voicecard soc:sound: ac10x-codec.1-003b <-> 3f203000.i2s mapping ok
[  970.512451] bcm2835-i2s 3f203000.i2s: I2S SYNC error!
```

Let me know what more information you need.

Thanks in advance!

cpoisson, 2019-10-17:
Hi @Sothymos,

I am from Snips. I think your issue should be addressed [here](https://github.com/snipsco/snips-issues/issues)

Did you tried to set your the base sampling rate of your capture interface to 16000 samples per seconds? This way, the snips-audio-server will not resample the sound captured and thus the cpu consumption will be reduced.

Sothymos, 2019-10-20:
Hi cpoisson,

thanks a lot for taking the time to write such a elaborate answer! :)
I opened the issue here as it all started with the seeed drivers.

When I tried _sudo -u _snips snips-audio-server -v --portaudio_capture="default" --portaudio_playback="default"_ I got the following output and it didn't change the situation:
```
pi@raspberrypi:~ $ sudo -u _snips snips-audio-server -v --portaudio_capture="default" --portaudio_playback="default"
[22:22:14.446600] DEBUG:snips_common_cli::cli: conf lookup: snips-audio-server.no_exit_on_all_panics -> Ok(None)
[22:22:14.447356] DEBUG:snips_common_cli::cli: conf lookup: snips-common.no_exit_on_all_panics -> Ok(None)
[22:22:14.447640] DEBUG:snips_common_cli::cli: conf lookup: snips-audio-server.bus -> Ok(None)
[22:22:14.447929] DEBUG:snips_common_cli::cli: conf lookup: snips-common.bus -> Ok(None)
[22:22:14.448190] DEBUG:snips_common_cli::cli: conf lookup: snips-audio-server.mqtt -> Ok(None)
[22:22:14.448449] DEBUG:snips_common_cli::cli: conf lookup: snips-common.mqtt -> Ok(None)
[22:22:14.448700] DEBUG:snips_common_cli::cli: conf lookup: snips-audio-server.mqtt_username -> Ok(None)
[22:22:14.448863] DEBUG:snips_common_cli::cli: conf lookup: snips-common.mqtt_username -> Ok(None)
[22:22:14.449047] DEBUG:snips_common_cli::cli: conf lookup: snips-audio-server.mqtt_password -> Ok(None)
[22:22:14.449566] DEBUG:snips_common_cli::cli: conf lookup: snips-common.mqtt_password -> Ok(None)
[22:22:14.449681] DEBUG:snips_common_cli::cli: conf lookup: snips-audio-server.mqtt_tls_hostname -> Ok(None)
[22:22:14.449877] DEBUG:snips_common_cli::cli: conf lookup: snips-common.mqtt_tls_hostname -> Ok(None)
[22:22:14.452466] DEBUG:snips_common_cli::cli: conf lookup: snips-audio-server.disable_playback -> Ok(None)
[22:22:14.453002] DEBUG:snips_common_cli::cli: conf lookup: snips-common.disable_playback -> Ok(None)
[22:22:14.453312] DEBUG:snips_common_cli::cli: conf lookup: snips-audio-server.alsa_playback -> Ok(None)
[22:22:14.453578] DEBUG:snips_common_cli::cli: conf lookup: snips-common.alsa_playback -> Ok(None)
[22:22:14.453850] DEBUG:snips_common_cli::cli: conf lookup: snips-audio-server.portaudio_playback -> Ok(Some("default                                                                                                                        "))
[22:22:14.454118] DEBUG:snips_common_cli::cli: conf lookup: snips-audio-server.output -> Ok(None)
[22:22:14.454541] DEBUG:snips_common_cli::cli: conf lookup: snips-common.output -> Ok(None)
[22:22:14.454816] DEBUG:snips_common_cli::cli: conf lookup: snips-audio-server.frame -> Ok(None)
[22:22:14.455078] DEBUG:snips_common_cli::cli: conf lookup: snips-common.frame -> Ok(None)
[22:22:14.455337] DEBUG:snips_common_cli::cli: conf lookup: snips-audio-server.disable_capture -> Ok(None)
[22:22:14.455572] DEBUG:snips_common_cli::cli: conf lookup: snips-common.disable_capture -> Ok(None)
[22:22:14.455734] DEBUG:snips_common_cli::cli: conf lookup: snips-audio-server.nomike -> Ok(None)
[22:22:14.455891] DEBUG:snips_common_cli::cli: conf lookup: snips-common.nomike -> Ok(None)
[22:22:14.456084] DEBUG:snips_common_cli::cli: conf lookup: snips-audio-server.portaudio_capture -> Ok(None)
[22:22:14.456258] DEBUG:snips_common_cli::cli: conf lookup: snips-common.portaudio_capture -> Ok(None)
[22:22:14.456416] DEBUG:snips_common_cli::cli: conf lookup: snips-audio-server.mike -> Ok(None)
[22:22:14.456795] DEBUG:snips_common_cli::cli: conf lookup: snips-common.mike -> Ok(None)
[22:22:14.457185] DEBUG:snips_common_cli::cli: conf lookup: snips-audio-server.alsa_capture -> Ok(None)
[22:22:14.457541] DEBUG:snips_common_cli::cli: conf lookup: snips-common.alsa_capture -> Ok(None)
[22:22:14.457818] DEBUG:snips_common_cli::cli: conf lookup: snips-audio-server.hijack -> Ok(None)
[22:22:14.458079] DEBUG:snips_common_cli::cli: conf lookup: snips-common.hijack -> Ok(None)
[22:22:14.458338] DEBUG:snips_common_cli::cli: conf lookup: snips-audio-server.bind -> Ok(None)
[22:22:14.458593] DEBUG:snips_common_cli::cli: conf lookup: snips-common.bind -> Ok(None)
[22:22:14.458876] DEBUG:snips_common_cli::cli: conf lookup: snips-audio-server.bind -> Ok(None)
[22:22:14.459111] DEBUG:snips_common_cli::cli: conf lookup: snips-common.bind -> Ok(None)
[22:22:14.459422] INFO :snips_audio_server_cli: Starting audio server on mqtt for site default
[22:22:14.459850] DEBUG:snips_audio_portaudio : Attempt 0
[22:22:14.459960] DEBUG:snips_audio_portaudio : Mike source capture started
Expression 'alsa_snd_pcm_hw_params_set_period_size_near( pcm, hwParams, &alsaPeriodFrames, &dir )' failed in 'src/hos                                                                                                                        tapi/alsa/pa_linux_alsa.c', line: 923
Expression 'alsa_snd_pcm_hw_params_set_period_size_near( pcm, hwParams, &alsaPeriodFrames, &dir )' failed in 'src/hos                                                                                                                        tapi/alsa/pa_linux_alsa.c', line: 923
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.front
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.rear
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.center_lfe
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.side
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.surround21
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.surround21
[... cut ...]
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.phoneline
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.phoneline
[... 21 lines cut ...]

**/etc/asound.conf**
```
pcm.!default {
    type asym
    playback.pcm {
        type plug
        slave.pcm "hw:0,1"
    }
    capture.pcm "capture"
}

pcm.capture {
    type plug
    slave.pcm "array"
}

pcm.array {
    type dsnoop
    ipc_key 666666
    ipc_key_add_uid yes
    ipc_perm 0666
    slave.pcm "hw:1,0"
}
```

Pillar1989, 2019-10-24:
@Sothymos @cpoisson Has the problem been solved? What can I do for you?

cpoisson, 2019-11-12:
@Pillar1989, Thanks for asking.

This one can be closed. We have a fix of the audio server in house to be deployed soon.

C3TAX, 2019-11-13:
Hey @Sothymos 

your workaround works fine for me, big THX :-)

> Then I tried the other Workarounds but none of the worked for me. The CPU load went down but I didn't get any sound anymore.
> After playing around with the playback configuration, I ended up with the [...]
My Hardware:

Raspi 3 B+
Respeaker 4

```
pi@raspberrypi:~ $ uname -a
Linux raspberrypi 4.19.75-v7+ #1270 SMP Tue Sep 24 18:45:11 BST 2019 armv7l GNU/Linux```

```
pi@raspberrypi:~ $ sam status

Connected to device localhost

OS version ................... Raspbian GNU/Linux 9 (stretch)
Installed assistant .......... alexa
Language ..................... de
Hotword ...................... hey_snips
ASR engine ................... snips
Status ....................... Live

Service status:

snips-asr .................... 0.64.0 (running)
snips-audio-server ........... 0.64.0 (running)
snips-dialogue ............... 0.64.0 (running)
snips-hotword ................ 0.64.0 (running)
snips-nlu .................... 0.64.0 (running)
snips-skill-server ........... 0.64.0 (running)
snips-tts .................... 0.64.0 (running)```

closed 2020-01-15 by Pillar1989