respeaker/seeed-voicecard #194: Mic-3 is muted in 6-mic and 4-mic array

giancarlocp, 2019-10-15:
Testing the 6-Mic Circular and 4-Mic Linear array,
In both cases I have noticed the mic 3 is muted.

Pillar1989, 2019-10-24:
This should be the hardware is broken, please try other @giancarlocp

closed 2019-10-24 by Pillar1989

HinTak, 2020-08-30:
@giancarlocp : this is probably the same issue as https://github.com/respeaker/seeed-voicecard/issues/240 - the seeedvoice card systemd service assumes that the on-board bcm driver is loaded, and set playback audio to the on-board phone jack with a amixer command. Unforturnately, if you don't use the pi's on-board audio (or with a later kernel where they are separate), the amixer command when applied to the wrong device, has the effect of turning the recording volume of the 3rd channel really low!