respeaker/seeed-voicecard #198: Respeaker 2mic hat not recognised by raspbian buster os

jabulonboaz, 2019-10-21:
The Respeaker 2mic hat not recognised by raspbian buster os.
Can you pls look into it?

simonmaurer, 2019-10-22:
which kernel are you using with the install.sh script?
I managed to get (however that is Respeaker 4mic hat) it working with the latest release of Raspbian Lite (Sep 2019) installed on a Raspberry Pi 2B (4.14.66) and Raspberry Pi 4 B (4.14.75)

turmary, 2019-10-23:
Will close this issue, please open a new issue if you have further questions.

closed 2019-10-23 by turmary

simonmaurer, 2019-10-23:
@turmary can you state a reason without knowing whether the issue has been resolved ;) ?

simonmaurer, 2019-10-23:
@turmary did you close this because the issue has been resolved ?

jabulonboaz, 2019-10-23:
The issue is not resolved. Raspian Buster latest build as its available from the raspberry pi foundation official web site does not work with Respeaker 2mic hat.
My set up is on a Raspberry pi 3 (not b+).
Basically the latest release of buster is not compatible with the respeaker 2 mic drivers as the device never gets installed.

I can verify that the respeaker 2mic hat works and gets installed on the same HW setup when i install the drivers on Raspian Jessie.

turmary, 2019-10-24:
We updated the install script (https://github.com/respeaker/seeed-voicecard/commit/873580cfd5b6844107199be1c54bda3a80a05e52), make sure the lastest kernel & headers installed well.

I tested the driver on lastest build with kernel & headers package version 1.20190925+1-1 and RPi3B & RPi3B+. 2Mic works well.

> pi@peter-pi:~ $ arecord -l
> **** List of CAPTURE Hardware Devices ****
> card 1: seeed2micvoicec [seeed-2mic-voicecard], device 0: bcm2835-i2s-wm8960-hifi wm8960-hifi-0 []
>   Subdevices: 1/1
>   Subdevice #0: subdevice #0
> pi@peter-pi:~ $ dpkg -l raspberrypi-kernel
> Desired=Unknown/Install/Remove/Purge/Hold
> | Status=Not/Inst/Conf-files/Unpacked/halF-conf/Half-inst/trig-aWait/Trig-pend
> |/ Err?=(none)/Reinst-required (Status,Err: uppercase=bad)
> ||/ Name               Version        Architecture Description
>+++-==================-==============-============-=================================
> ii  raspberrypi-kernel 1.20190925+1-1 armhf        Raspberry Pi bootloader