respeaker/seeed-voicecard #199: ReSpeaker 4-Mic Linear Array - noise detection problems

julia100009, 2019-11-05:
Hallo,
I have installed the ReSpeaker 4-Mic Linear Array Kit for Raspberry P (4 Model Rev 1.1) as shown on the seeed website (seeed-8mic-voicecard). 
I have programmed an AI that detects noises. Everything works perfectly until I restart the Raspberry. Then the noises are no longer recognized.
The following picture shows the same sound before and after the restart.

Does anyone have any idea what the problem is? Is it the AGC? The recording volume is adjusted somehow, how can i change it?

Thank you in advance. Kind regards,
Julia
[image]

Pillar1989, 2019-12-13:
Have you worked it out?

julia100009, 2019-12-13:
Hi,
I think the problem was AGC. Unfortunately I could not disable AGC on the Model ReSpeaker 4-Mic Linear Array. I bought the model Mic Array v2.0. Now I can adjust the parameters (AGC etc.), the problem is gone.
Thanks for asking.

Pillar1989, 2019-12-14:
AC108 doesn't have  AGC. This is his  specification.
https://cdn.sparkfun.com/assets/6/0/d/c/a/AC108_Datasheet_V1.1.pdf

closed 2020-01-15 by Pillar1989

Atul-Anand-Jha, 2020-03-13:
> Hi,
> I think the problem was AGC. Unfortunately I could not disable AGC on the Model ReSpeaker 4-Mic Linear Array. I bought the model Mic Array v2.0. Now I can adjust the parameters (AGC etc.), the problem is gone.
> Thanks for asking.

@julia100009  ,How did you adjusted the AGC ? I am also unable to activate BF( beam Forming) . How to do these stuffs.