respeaker/seeed-voicecard #202: PGA gain can cause incorrect audio clipping

threebrooks, 2019-12-10:
I have noticed that when you set the "ADC PGA gain" to anything but 0 and the incoming audio happens to clip because it is too loud, the driver no longer produces nicely-clipped audio (i.e. maxing out at the integer min/max values), but instead shows numerical artifacts, similar to integer overflow artifacts. For applications that want to generate user feedback based on that clipping it becomes impossible to detect.

Ways to reproduce:
1. Go into alsamixer, set the "ADC PGA gain" values to a high value
2. Play very loud audio into the microphones
3. Record with arecord, inspect audio visually in waveform editor

Pillar1989, 2019-12-13:
The remote field microphone is always like this, the practical application needs to cooperate with a certain front-end algorithm to deal with the noise

closed 2019-12-13 by Pillar1989

threebrooks, 2019-12-13:
Just to confirm, I am not referring to normal clipping (which is of course normal for any microphone that is overdriven). What I am saying is that if you use the PGA Gain, the values are no longer pegged to the maximum integer value (e.g. 32,767 for a 16-bit recording) but instead start oddly fluctuating, indicating there is some kind of integer overflow happening inside the driver.