respeaker/seeed-voicecard #205: Error! There are no instances of module: seeed-voicecard 0.3 located in the DKMS tree.

jmerkert, 2020-01-14:
Hi,

i have problems installing my ReSpeaker 4 Array. The installation stucks in the make process at the end of the log:

`(Reading database ... 93751 files and directories currently installed.)
Preparing to unpack .../raspberrypi-kernel-headers_1.20190925+1-1_armhf.deb ...
Unpacking raspberrypi-kernel-headers (1.20190925+1-1) over (1.20190925+1-1) ...
Setting up raspberrypi-kernel-headers (1.20190925+1-1) ...
run-parts: executing /etc/kernel/header_postinst.d/dkms 4.19.75+
run-parts: executing /etc/kernel/header_postinst.d/dkms 4.19.75-v7+
run-parts: executing /etc/kernel/header_postinst.d/dkms 4.19.75-v7l+
run-parts: executing /etc/kernel/header_postinst.d/dkms 4.19.75-v8+
Error! There are no instances of module: seeed-voicecard
0.3 located in the DKMS tree.

Creating symlink /var/lib/dkms/seeed-voicecard/0.3/source ->
                 /usr/src/seeed-voicecard-0.3

DKMS: add completed.

Kernel preparation unnecessary for this kernel.  Skipping...

Building module:
cleaning build area...
make -j4 KERNELRELEASE=4.19.75+ -C /lib/modules/4.19.75+/build M=/var/lib/dkms/seeed-voicecard/0.3/build.....
cleaning build area...

DKMS: build completed.

snd-soc-wm8960.ko:
Running module version sanity check.
 - Original module
   - No original module exists within this kernel
 - Installation
   - Installing to /lib/modules/4.19.75+/kernel/sound/soc/codecs/

snd-soc-ac108.ko:
Running module version sanity check.
 - Original module
   - No original module exists within this kernel
 - Installation
   - Installing to /lib/modules/4.19.75+/kernel/sound/soc/codecs/

snd-soc-seeed-voicecard.ko:
Running module version sanity check.
 - Original module
   - No original module exists within this kernel
 - Installation
   - Installing to /lib/modules/4.19.75+/kernel/sound/soc/bcm/

depmod...

DKMS: install completed.

Kernel preparation unnecessary for this kernel.  Skipping...

Building module:
cleaning build area...
make -j4 KERNELRELEASE=4.19.75-v7+ -C /lib/modules/4.19.75-v7+/build M=/var/lib/dkms/seeed-voicecard/0.3/build....`

Any suggestions?

Thanks in advance
Johannes

Pillar1989, 2020-01-15:
There were no errors in your log. You finally tested it. No sound CARDS?

jmerkert, 2020-01-15:
Hi,

the log says `Error! There are no instances of module: seeed-voicecard
0.3 located in the DKMS tree.` Isn't that an error?
When listing the devices, the mic is not listed:

> pi@raspberrypi:~ $ arecord -L
> null
>     Discard all samples (playback) or generate zero samples (capture)
> jack
>     JACK Audio Connection Kit
> pulse
>     PulseAudio Sound Server
> usbstream:CARD=ALSA
>     bcm2835 ALSA
>     USB Stream Output

jmerkert, 2020-01-19:
Installation worked now. My best guess is, that the power supply was not capable.

closed 2020-01-19 by jmerkert