respeaker/seeed-voicecard #207: Not able to install 4mic array- need step by step guidance

Kuldeepwadaskar, 2020-02-05:
Hi i am trying to install 4 mic array as per given instructions but arecord -l does not show any device after installation 
 sudo ./install.sh 
Hit:1 http://ports.ubuntu.com xenial InRelease
Hit:2 http://ports.ubuntu.com xenial-updates InRelease                  
Hit:3 http://packages.ros.org/ros/ubuntu xenial InRelease               
Hit:4 http://ports.ubuntu.com xenial-security InRelease
Hit:5 http://ports.ubuntu.com xenial-backports InRelease
Hit:6 https://packages.ubiquityrobotics.com/ubuntu/ubiquity xenial InRelease
Hit:7 http://ppa.launchpad.net/ubuntu-pi-flavour-makers/ppa/ubuntu xenial InRelease
Reading package lists... Done
Building dependency tree       
Reading state information... Done
2 packages can be upgraded. Run 'apt list --upgradable' to see them.
Reading package lists... Done
Building dependency tree       
Reading state information... Done
libasound2-plugins is already the newest version (1.1.0-0ubuntu1).
libasound2-plugins set to manually installed.
git is already the newest version (1:2.7.4-0ubuntu1.7).
git set to manually installed.
i2c-tools is already the newest version (3.1.2-3~xenial1.0).
i2c-tools set to manually installed.
The following NEW packages will be installed:
  dkms
0 upgraded, 1 newly installed, 0 to remove and 2 not upgraded.
Need to get 66.4 kB of archives.
After this operation, 267 kB of additional disk space will be used.
Get:1 http://ports.ubuntu.com xenial-updates/main armhf dkms all 2.2.0.3-2ubuntu11.8 [66.4 kB]
Fetched 66.4 kB in 1s (64.2 kB/s)
Selecting previously unselected package dkms.
(Reading database ... 180984 files and directories currently installed.)
Preparing to unpack .../dkms_2.2.0.3-2ubuntu11.8_all.deb ...
Unpacking dkms (2.2.0.3-2ubuntu11.8) ...
Processing triggers for man-db (2.7.5-1) ...
Setting up dkms (2.2.0.3-2ubuntu11.8) ...
--2020-02-05 21:16:25--  https://packages.ubiquityrobotics.com/ubuntu/ubiquity/pool/pi/r/raspberrypi-firmware/raspberrypi-kernel_1.20190925+1-1_armhf.deb
Resolving packages.ubiquityrobotics.com (packages.ubiquityrobotics.com)... 2604:a880:2:d0::2ce:a001, 138.197.232.216
Connecting to packages.ubiquityrobotics.com (packages.ubiquityrobotics.com)|2604:a880:2:d0::2ce:a001|:443... connected.
HTTP request sent, awaiting response... 404 Not Found
2020-02-05 21:16:27 ERROR 404: Not Found.

--2020-02-05 21:16:27--  https://packages.ubiquityrobotics.com/ubuntu/ubiquity/pool/pi/r/raspberrypi-firmware/raspberrypi-kernel_1.20190925+1-1_armhf.deb
Resolving packages.ubiquityrobotics.com (packages.ubiquityrobotics.com)... 2604:a880:2:d0::2ce:a001, 138.197.232.216
Connecting to packages.ubiquityrobotics.com (packages.ubiquityrobotics.com)|2604:a880:2:d0::2ce:a001|:443... connected.
HTTP request sent, awaiting response... 404 Not Found
2020-02-05 21:16:29 ERROR 404: Not Found.

--2020-02-05 21:16:29--  https://packages.ubiquityrobotics.com/ubuntu/ubiquity/pool/pi/r/raspberrypi-firmware/raspberrypi-kernel_1.20190925+1-1_armhf.deb
Resolving packages.ubiquityrobotics.com (packages.ubiquityrobotics.com)... 2604:a880:2:d0::2ce:a001, 138.197.232.216
Connecting to packages.ubiquityrobotics.com (packages.ubiquityrobotics.com)|2604:a880:2:d0::2ce:a001|:443... connected.
HTTP request sent, awaiting response... 404 Not Found
2020-02-05 21:16:30 ERROR 404: Not Found.

dpkg-deb: error: '/tmp/raspberrypi-kernel_1.20190925+1-1_armhf.deb' is not a debian format archive
dpkg: error processing archive /tmp/raspberrypi-kernel_1.20190925+1-1_armhf.deb (--install):
 subprocess dpkg-deb --control returned error exit status 2
Errors were encountered while processing:
 /tmp/raspberrypi-kernel_1.20190925+1-1_armhf.deb
--2020-02-05 21:16:31--  https://packages.ubiquityrobotics.com/ubuntu/ubiquity/pool/pi/r/raspberrypi-firmware/raspberrypi-kernel-headers_1.20190925+1-1_armhf.deb
Resolving packages.ubiquityrobotics.com (packages.ubiquityrobotics.com)... 2604:a880:2:d0::2ce:a001, 138.197.232.216
Connecting to packages.ubiquityrobotics.com (packages.ubiquityrobotics.com)|2604:a880:2:d0::2ce:a001|:443... connected.
HTTP request sent, awaiting response... 404 Not Found
2020-02-05 21:16:32 ERROR 404: Not Found.

--2020-02-05 21:16:32--  https://packages.ubiquityrobotics.com/ubuntu/ubiquity/pool/pi/r/raspberrypi-firmware/raspberrypi-kernel-headers_1.20190925+1-1_armhf.deb
Resolving packages.ubiquityrobotics.com (packages.ubiquityrobotics.com)... 2604:a880:2:d0::2ce:a001, 138.197.232.216
Connecting to packages.ubiquityrobotics.com (packages.ubiquityrobotics.com)|2604:a880:2:d0::2ce:a001|:443... connected.
HTTP request sent, awaiting response... 404 Not Found
2020-02-05 21:16:34 ERROR 404: Not Found.

[... 454 lines cut ...]
Please install the linux-headers-4.19.50-v7l+ package,
or use the --kernelsourcedir option to tell DKMS where it's located
setup git config
git init
Initialized empty Git repository in /etc/voicecard/.git/
git add --all
git commit -m "origin configures"
[master (root-commit) ac77700] origin configures
 7 files changed, 1476 insertions(+)
 create mode 100644 ac108_6mic.state
 create mode 100644 ac108_asound.state
 create mode 100644 asound_2mic.conf
 create mode 100644 asound_4mic.conf
 create mode 100644 asound_6mic.conf
 create mode 100644 dkms.conf
 create mode 100644 wm8960_asound.state
Created symlink from /etc/systemd/system/sysinit.target.wants/seeed-voicecard.service to /lib/systemd/system/seeed-voicecard.service.
------------------------------------------------------
Please reboot your raspberry pi to apply all settings
Enjoy!
------------------------------------------------------

uname -r  : 4.19.50-v7+

Please help me to solve this

Pillar1989, 2020-02-14:
Look for your own problems. If everyone can't install and use it, it won't sell

closed 2020-02-14 by Pillar1989

Kuldeepwadaskar, 2020-02-14:
Hey installation was successful..

But it does not have good voice capture range when used for speech
recognition.

Have any solution?
I don't know about others but next time I will not buy it

Best Regards,

Kuldeepwadaskar, 2020-08-07:
[image: image.png]
LET'S CAGE THE VIRUS
CREATE A SHIELD OF PROTECTION AROUND YOUR HOME WITH OUR CAGE
>
> *"DEACTIVATE 99% VIRUS ON VEGETABLES, BELONGINGS, JEWELLERY & OTHER ITEMS
> IN HOME INCLUDING SHOES."*

[image: UV CAGE]

> *Get Early Bird Offer* <http://3notrobotics.in/pricing.html>

talksik, 2023-04-22:
how did you fix it @Kuldeepwadaskar?

lxne, 2023-04-23:
> how did you fix it @Kuldeepwadaskar?

I do not know your hardware and system version,
but you probably need to use HinTak's fork to get it working.