respeaker/seeed-voicecard #209: Remove Kernel Downgrade

mxklb, 2020-02-14:
Actually I can't install the 2mic respeaker driver on a vanilla flashed [HypriotOS v1.12.0](https://github.com/hypriot) on a PI4. What happens is, the /boot partition of the OS is too small (~12MB free space) to downgrade the kernel during your install.sh script. So far so good but finally after your script finished, I end up with a bricked OS with read-only file system. 

The install.sh script shall therefore

- [ ] check that enough space is left on the /boot partition, otherwise skip installation,
- [ ] patch the running kernel, instead of downgrading existing newer kernels.

Ok, now I'll get a flashed [HypriotOS](https://github.com/hypriot) with bigger /boot partition.
P.S. With former releases of this OS, 2mic respeaker worked. I'll post an issue there too.

Pillar1989, 2020-02-14:
I want to Downgrade by default, so that to ensure that the software can be installed, perhaps you can submit a PR, there is an option is the use of the latest kernel.

mxklb, 2020-02-14:
https://github.com/hypriot/image-builder-rpi/issues/346

mxklb, 2020-02-14:
I'm not familiar with patching kernels, driver development. So I won't contribute to the second part - avoid downgrading the kernel for driver installation to succeed. Please do it by your own..

Nevertheless, good job, thank you very much for developing respeaker drivers!

solidssss, 2020-02-15:
Oh, so that was why I could not install the 4mic driver on Raspbian... Anyone know of a fix?

HinTak, 2020-03-19:
See also the now closed #192 , I have pushed upstream to fix the kernel header issue - 

https://github.com/raspberrypi/linux/issues/3279

So it is no longer necessary to downgrade the kernel to get matching headers. In fact I have been using the respeaker driver for both up to date stretch and up to date buster for a few months since. Please remove the kernel downgrade. @Pillar1989

malikasinger1, 2020-03-19:
Actually what is happening to me is it is taking forever in kernel downgrade part in ./install.sh

I am using this version of raspbian (as mentioned here: https://www.raspberrypi.org/downloads/raspbian/):

```
Raspbian Buster Lite
Minimal image based on Debian Buster
Version:February 2020
Release date:2020-02-13
Kernel version:4.19
Size:434 MB
```
it is not only taking forever but sometimes it just got disconnect from the ssh, pi is not restarting not script failing but just got unresponsive

@HinTak as you told above that something was wrong was with raspbian and you said it is now fixed, so if I remove `--force-yes` from the script do you think it will still work perfectly, I also noticed they have added  in this line of script recently (end of 2019)

please let me know your opinion

however what I have done for now is started downloading raspbian stretch with release name `2019-04-08` and also removed the force from script and running it now, see what happens

malikasinger1, 2020-03-19:
someone also tell me the kernel which they are forcing came with which version of raspbian, because I cannot find list of all raspbian with their kernel detail on internet, I will be thankful 

for now I don't care about anything but I have to complete one of my prototype asap, so may be I will be ending up downloading all raspbian releases and installing each of them for checking their version in worst case, lolz how dumb of me 🤣

malikasinger1, 2020-03-19:
here is my fork I removed the `--force-yes`

raspbian version is same as before:

```
Raspbian Buster Lite
Minimal image based on Debian Buster
Version:February 2020
Release date:2020-02-13
Kernel version:4.19
Size:434 MB
```

and here is my install log:

```
Using username "pi".
pi@raspberrypi.local's password:
Access denied
pi@raspberrypi.local's password:
Access denied
pi@raspberrypi.local's password:
Linux raspberrypi 4.19.75-v7+ #1270 SMP Tue Sep 24 18:45:11 BST 2019 armv7l

The programs included with the Debian GNU/Linux system are free software;
the exact distribution terms for each program are described in the
individual files in /usr/share/doc/*/copyright.

Debian GNU/Linux comes with ABSOLUTELY NO WARRANTY, to the extent
permitted by applicable law.
Last login: Thu Mar 19 01:01:07 2020 from 192.168.43.6

SSH is enabled and the default password for the 'pi' user has not been changed.
This is a security risk - please login as the 'pi' user and type 'passwd' to set                                                                                                      a new password.

pi@raspberrypi:~ $ cd data/
pi@raspberrypi:~/data $ cd ..
pi@raspberrypi:~ $ git clone https://github.com/malikasinger1/seeed-voicecard.gi                                                                                                     t
Cloning into 'seeed-voicecard'...
remote: Enumerating objects: 16, done.
remote: Counting objects: 100% (16/16), done.
remote: Compressing objects: 100% (14/14), done.
remote: Total 710 (delta 8), reused 3 (delta 2), pack-reused 694
Receiving objects: 100% (710/710), 1.31 MiB | 197.00 KiB/s, done.
Resolving deltas: 100% (442/442), done.
pi@raspberrypi:~ $ cd seeed-voicecard/
pi@raspberrypi:~/seeed-voicecard $ sudo ./install.sh
Hit:1 http://archive.raspberrypi.org/debian buster InRelease
Get:2 http://raspbian.raspberrypi.org/raspbian buster InRelease [15.0 kB]
Fetched 15.0 kB in 2s (6,930 B/s)
Reading package lists... Done
Building dependency tree
Reading state information... Done
5 packages can be upgraded. Run 'apt list --upgradable' to see them.
Reading package lists... Done
Building dependency tree
Reading state information... Done
dkms is already the newest version (2.6.1-4).
git is already the newest version (1:2.20.1-2+deb10u1).
i2c-tools is already the newest version (4.1-1).
libasound2-plugins is already the newest version (1.1.8-1).
0 upgraded, 0 newly installed, 0 to remove and 5 not upgraded.
--2020-03-19 21:58:49--  http://archive.raspberrypi.org/debian/pool/main/r/raspb                                                                                                     errypi-firmware/raspberrypi-kernel_1.20190925+1-1_armhf.deb
Resolving archive.raspberrypi.org (archive.raspberrypi.org)... 46.235.231.151, 4                                                                                                     6.235.231.111, 176.126.240.167, ...
Connecting to archive.raspberrypi.org (archive.raspberrypi.org)|46.235.231.151|:                                                                                                     80... connected.
HTTP request sent, awaiting response... 200 OK
Length: 70656964 (67M) [application/x-debian-package]
Saving to: ‘/tmp/raspberrypi-kernel_1.20190925+1-1_armhf.deb’

/tmp/raspberrypi-ke  40%[=======>            ]  27.30M   195KB/s    in 2m 53s
[... 176 lines cut ...]
 create mode 100644 asound_4mic.conf
 create mode 100644 asound_6mic.conf
 create mode 100644 dkms.conf
 create mode 100644 wm8960_asound.state
Created symlink /etc/systemd/system/sysinit.target.wants/seeed-voicecard.service → /lib/systemd/system/seeed-voicecard.service.
Job for seeed-voicecard.service failed because the control process exited with error code.
See "systemctl status seeed-voicecard.service" and "journalctl -xe" for details.
------------------------------------------------------
Please reboot your raspberry pi to apply all settings
Enjoy!
------------------------------------------------------
pi@raspberrypi:~/seeed-voicecard $
pi@raspberrypi:~/seeed-voicecard $
pi@raspberrypi:~/seeed-voicecard $
pi@raspberrypi:~/seeed-voicecard $
pi@raspberrypi:~/seeed-voicecard $
pi@raspberrypi:~/seeed-voicecard $
pi@raspberrypi:~/seeed-voicecard $
pi@raspberrypi:~/seeed-voicecard $
pi@raspberrypi:~/seeed-voicecard $
pi@raspberrypi:~/seeed-voicecard $

```

lets see what happens after reboot :-D

malikasinger1, 2020-03-19:
worked ;-)

mxklb, 2020-03-20:
I'll wait until the kernel downgrade is removed - as @HinTak mentioned this is not needed anymore and should be removed. Actually the state of this driver installation script is some kind of messy and needs to be refactored in my opinion.

Psychokiller1888, 2020-03-31:
Yeah, that kernel downgrade is a real pain, terrible

Psychokiller1888, 2020-04-01:
Hallo @Pillar1989 Can we have your input? It's terrible to install on a rpi zero, wait 20+ minutes for nothing. There must be a better way to install those drivers no?

HinTak, 2020-04-23:
The available kernel packages are under
http://archive.raspberrypi.org/debian/pool/main/r/raspberrypi-firmware/ . FWIW, 1.20190925+1-1 is somewhat dated now.

HinTak, 2020-04-26:
I have put out a pull for this https://github.com/respeaker/seeed-voicecard/pull/223

KillingJacky, 2020-04-27:
Done with https://github.com/respeaker/seeed-voicecard/pull/225

ismailryt, 2020-05-07:
Hello @KillingJacky, i want to be sure and ask some question.

After i do " sudo ./install.sh ", and reboot, arecord -L didnt show me any seeed voice card device. So this mean i have to follow #225 ??  Because after i plug my  sd card to raspy3b+, arecord -L show me seeed voice card device.  It makes me confused why in raspy4 it can't , in raspy 3b+ it can

Im using raspy4 and latest raspbian buster fyi. 

Thanks,

HinTak, 2020-05-07:
You need to look at `dmesg` to see what happened. You should have a few ac10x lines. I am on up to date buster on a pi 4. Works just fine.

ismailryt, 2020-05-07:
Pardon @HinTak , that mean you just doing normal instalation and success?  Now im formatting sd card and trying to install again. Im gonna try what you say, so i have to put cmnd dmesg after i do "sudo ./install.sh " for deebugging right?

Please correct me if i wrong

ismailryt, 2020-05-07:
Oh 1 more , where i can find this command and change it 

>  W: --force-yes is deprecated, use one of the options starting with --allow instead.

HinTak, 2020-05-08:
No, you watch dress after reboot. Dmesg shows messages from the kernel (whether the driver is loaded, and successful or not)

HinTak, 2020-08-07:
see https://github.com/respeaker/seeed-voicecard/issues/238#issuecomment-670303966 regarding the unfeasibility of downgrading...

Pillar1989, 2020-09-25:
https://github.com/respeaker/seeed-voicecard/issues/249

closed 2020-09-25 by Pillar1989

turmary, 2020-09-28:
hi @mxklb 

> patch the running kernel, instead of downgrading existing newer kernels,
this could be done by ./install argument --keep-kernel

mxklb, 2020-10-13:
I don't use seeed-voicecard products anymore, this one is not production ready. This all here is just DIY/WIP stuff ..

If you like DIY/WIP, you could stick to seeed-voicecard products. I for myself decided to step away from this vendor (in my opinion at least seeed-voicecard is badly designed & hacky from what I saw..) and use better production ready products.

@turmary thaks for this info, this is really unbelievable, I can't get it .. :+1: