respeaker/seeed-voicecard #211: Add support for Ubuntu (aarch64)

mxklb, 2020-02-19:
When trying to install the driver on Ubuntu with Raspberry PI 4
```
root@ubuntu:/# uname -a
Linux ubuntu 5.3.0-1018-raspi2 #20-Ubuntu SMP Mon Feb 3 19:45:46 UTC 2020 aarch64 aarch64 aarch64 GNU/Linux
```
I get `/boot/overlays not found or not a directory` error:
```
root@ubuntu:/seeed-voicecard# ./install.sh 
/boot/overlays not found or not a directory
dtparam not found
dtoverlay not found
Errors found, exiting.
```
It would be nice to be able to install the driver on other Linux distributions commonly used with a PI.

Is there any chance to install/compile the driver with Ubuntu aarch64 on a PI 4?

AKrause4, 2020-02-22:
I had this exact same issue when trying to install the driver on Jetson Nano (Developer Kit edition), did not have issues on Raspberry Pi 4 with Raspbian Buster

AKrause4, 2020-02-22:
The boot directory is not on its own partition, is that required? 

anna@elf:~/seeed-voicecard$ df -h
Filesystem      Size  Used Avail Use% Mounted on
/dev/mmcblk0p1  118G   12G  102G  10% /
none            1.7G     0  1.7G   0% /dev
tmpfs           2.0G  181M  1.8G  10% /dev/shm
tmpfs           2.0G   77M  1.9G   4% /run
tmpfs           5.0M  4.0K  5.0M   1% /run/lock
tmpfs           2.0G     0  2.0G   0% /sys/fs/cgroup
tmpfs           396M  132K  396M   1% /run/user/1000

mxklb, 2020-02-23:
Main difference is that files configured by seeed-voicecard/install.sh are located in _/boot/firmware_ on ubuntu while rasbian has them directly in _/boot_. The seeed-voicecard/install.sh requires files to be in _/boot_, so therefore the install script stops before doing anything. 

@AKrause4 It is commonly known that the driver works on raspbian, its developed for this OS. This Issue is a feature request to support Ubuntu! Special partitioning of boot in generally is not required, see: https://serverfault.com/questions/334663/what-is-the-recommended-size-for-a-linux-boot-partition But are you shure that you've used vanilla flashed PI 4 ubuntu aarch64. The sd-card should be partitioned into boot and root file-systems, mine looks like this:

```
$ df -h
Filesystem      Size  Used Avail Use% Mounted on
udev            1.9G     0  1.9G   0% /dev
tmpfs           380M  4.3M  375M   2% /run
/dev/mmcblk0p2   59G  5.9G   51G  11% /
tmpfs           1.9G     0  1.9G   0% /dev/shm
tmpfs           5.0M  4.0K  5.0M   1% /run/lock
tmpfs           1.9G     0  1.9G   0% /sys/fs/cgroup
/dev/loop0       81M   81M     0 100% /snap/core/8596
/dev/loop1       81M   81M     0 100% /snap/core/8691
/dev/loop3       60M   60M     0 100% /snap/lxd/13477
/dev/mmcblk0p1  253M  127M  127M  51% /boot/firmware
/dev/loop4       60M   60M     0 100% /snap/lxd/13489
tmpfs           380M  4.0K  380M   1% /run/user/1004 
```

AKrause4, 2020-02-23:
Like I said, I am using a Jetson Nano and have the same issue and if Seeed Studio is selling Nanos then the install script should be updated.

hhuysqt, 2020-03-29:
I also have the identical issue on Ubuntu Server 1804 on RPI 4 with 4GB RAM. I prefer Ubuntu because the official 32-bit Raspbian cannot support up to 4GB RAM out of the box. Support for Ubuntu is strongly demanded :)

hhuysqt, 2020-03-30:
See [seeed-voicecard-v5.3.c](https://github.com/hhuysqt/seeed-voicecard/blob/ubuntu/seeed-voicecard-v5.3.c) for my ad-hoc solution.
I made some modification to work on linux kernel v5.3.0 or higher, which has merged some API changes on `simple_card`: [patch1: separate asoc_simple_card_parse_dai()](https://patchwork.kernel.org/patch/10860821/) and [patch2: rename asoc_simple_card_xxx()](https://patchwork.kernel.org/patch/10860823/). 
To install it on my 64-bit RPI4 Ubuntu 1804 with kernel 5.3.0-1019-raspi2, I manually follow the instructions in `install.sh`. Since there is neither `dtparam` nor  `dtoverlay` command on ubuntu, I [manually insert the device tree blob in sysfs](https://www.96boards.org/documentation/consumer/dragonboard/dragonboard410c/guides/dt-overlays.md.html#3-load-overlays-dynamically).

dony71, 2020-04-08:
@hhuysqt : how do you apply patch for kernel 5.5.7 ?
in simple_card_utils.h, there is NOT any function asoc_simple_card_* but instead asoc_simple_* (card omitted)

hhuysqt, 2020-04-08:
@dony71 Well, I don't patch the kernel, but I modify the driver. See [seeed-voicecard-v5.3.c](https://github.com/hhuysqt/seeed-voicecard/blob/ubuntu/seeed-voicecard-v5.3.c), I change asoc_simple_card_* to asoc_simple_*, and commit other changes as well to match the new simple card APIs in the kernel.

dony71, 2020-04-08:
@hhuysqt I try to compile without patching simple_card_utils.h and get this error below

pi64 /var/lib/dkms/seeed-voicecard/0.3/build # vi make.log
  866 |    dev_dbg(dai->dev, "used as slave when AC101 is master\n");
      |    ^~~~~~~
/var/lib/dkms/seeed-voicecard/0.3/build/ac108.c:868:2: note: here
  868 |  case SND_SOC_DAIFMT_CBS_CFS:    /*AC108 Slave*/
      |  ^~~~
/var/lib/dkms/seeed-voicecard/0.3/build/wm8960.c: In function ‘wm8960_hw_params’:
/var/lib/dkms/seeed-voicecard/0.3/build/wm8960.c:752:6: warning: this statement may fall through [-Wimplicit-fallthrough=]
  752 |   if ((iface & 0x3) != 0) {
      |      ^
/var/lib/dkms/seeed-voicecard/0.3/build/wm8960.c:756:2: note: here
  756 |  default:
      |  ^~~~~~~
/var/lib/dkms/seeed-voicecard/0.3/build/seeed-voicecard.c: In function ‘seeed_voice_card_parse_aux_devs’:
/var/lib/dkms/seeed-voicecard/0.3/build/seeed-voicecard.c:513:28: error: ‘struct snd_soc_aux_dev’ has no member named ‘codec_of_node’
  513 |   priv->snd_card.aux_dev[i].codec_of_node = aux_node;
      |                            ^
make[1]: *** [scripts/Makefile.build:266: /var/lib/dkms/seeed-voicecard/0.3/build/seeed-voicecard.o] Error 1
make[1]: *** Waiting for unfinished jobs....
make: *** [Makefile:1693: /var/lib/dkms/seeed-voicecard/0.3/build] Error 2

hhuysqt, 2020-04-09:
@dony71 Oh sorry but I didn't notice the API changes since kernel V5.4... 
https://elixir.bootlin.com/linux/v5.4-rc1/source/include/sound/soc.h#L965
`struct snd_soc_aux_dev::codec_of_node` had been changed to `struct snd_soc_aux_dev::dlc::of_node`
As an ad-hoc solution, you may change it to
``` c
 priv->snd_card.aux_dev[i].dlc.of_node = aux_node;
```
But currently the arm64 Ubuntu mirror holds `linux-image-5.3.0-1022-raspi2` for 18.04, and `linux-image-5.4.0-1006-raspi2` for 20.04, so I'm not going to update my kernel and I'm not sure if there are any other issues...

StuartIanNaylor, 2020-04-23:
> @dony71 Oh sorry but I didn't notice the API changes since kernel V5.4...
> https://elixir.bootlin.com/linux/v5.4-rc1/source/include/sound/soc.h#L965
> `struct snd_soc_aux_dev::codec_of_node` had been [...]

@hhuysqt It was 20.04 day today and just wondering if you fancied having a go at making an easier install for us noobs.

I guess really we just need to take your changes and just apply the bottom part of the script.
```
dkms build -k $_i -m $mod -v $ver && {
      dkms install --force -k $_i -m $mod -v $ver
    }
```

Not very knowledgable but rather than the -k $_i if you omit is it just created for current?

HinTak, 2020-04-24:
The standard kernel header /usr/include/linux/version.h , contains the kernel version and a macro for  `KERNEL_VERSION(a, b,c)` which allows you to do version comparison likes here: 

https://git.kernel.org/pub/scm/linux/kernel/git/backports/backports.git/tree/backport/backport-include/linux/version.h

(the device driver backport project). what you want is the reverse, porting forward, but the idea is similar: you inserts a few "#if LINUX_VERSION_IS_LESS(x1,x2,x3) ... #endif" everywhere to allow the driver to work for two kernel versions, if they turn out to be different sufficiently.

StuartIanNaylor, 2020-04-24:
I just had a go on Archlinux and can not install there :)

Think I will just wait to see if someone will kindly provide something a little more simple for us who don't know much more than the sound card purchased :)

HinTak, 2020-04-24:
Ported the code to build against 5.5 headers -
https://github.com/HinTak/seeed-voicecard/tree/v5.5 - the details are in the changelog, should work for 5.3(?) 5.4+ . 

The install.sh has a few other problems, so it is perhaps best not to use it especially the dkms part, but do "make ; make install" plus copying the dtbo.

HinTak, 2020-04-24:
I added the the kernel version checking macro but then realize the needed changes are spread across 5.1/5.2/5.3/5.4 (again, see changelog) so it is too much work to try to make one tree work for all .

StuartIanNaylor, 2020-04-24:
Just one error on make

```
In file included from /home/ubuntu/seeed-voicecard/seeed-voicecard.c:28:
/home/ubuntu/seeed-voicecard/seeed-voicecard.c: In function ‘seeed_voice_card_dai_link_of’:
./include/sound/simple_card_utils.h:101:2: error: implicit declaration of function ‘asoc_simple_parse_dai’; did you mean ‘asoc_simple_parse_clk’? [-Werror                                                       =implicit-function-declaration]
  101 |  asoc_simple_parse_dai(node, dai_link->cpus, is_single_link)
```

StuartIanNaylor, 2020-04-25:
> Ported the code to build against 5.5 headers -
> https://github.com/HinTak/seeed-voicecard/tree/v5.5 - the details are in the changelog, should work for 5.3(?) 5.4+ .
>
> The install.sh has a few other [...]

Hintak any chance you could port against 5.4 as 5.5 is already EOL.
5.4 is LTS so at least will run to 2023

HinTak, 2020-04-26:
There is not further work between 5.4 and 5.5 (i.e. None of the changes so far is 5.5 specific). I called the branch v5.5 mostly because I am cross-developing from a 5.5 host. As I said, see the changelog - each of the changes say what version of kernel it is needed.

StuartIanNaylor, 2020-04-26:
```
In file included from /home/ubuntu/seeed-voicecard/seeed-voicecard.c:28:
/home/ubuntu/seeed-voicecard/seeed-voicecard.c: In function ‘seeed_voice_card_dai_link_of’:
./include/sound/simple_card_utils.h:101:2: error: implicit declaration of function ‘asoc_simple_parse_dai’; did you mean ‘asoc_simple_parse_clk’? [-Werror=implicit-function-declarat                    ion]
  101 |  asoc_simple_parse_dai(node, dai_link->cpus, is_single_link)
      |  ^~~~~~~~~~~~~~~~~~~~~
/home/ubuntu/seeed-voicecard/seeed-voicecard.c:304:8: note: in expansion of macro ‘asoc_simple_parse_cpu’
  304 |  ret = asoc_simple_parse_cpu(cpu, dai_link, &single_cpu);
      |        ^~~~~~~~~~~~~~~~~~~~~
```
This is on 5.4 and it doesn't work that is all I know, apols.

StuartIanNaylor, 2020-04-26:
If you copy the dtbo to /boot/firmware/overlays and set /boot/firmware usrcfg.txt
```
dtparam=i2s=on
dtoverlay=i2s-mmap
dtoverlay=seeed-2mic-voicecard
```
It actually finds a wm8960 alsactl and everything is there just not sound or recording.
Noticed it in the kernel for 5.4 and thought give 20.04 a go but no.

HinTak, 2020-04-26:
I have looked at the log, for some reason `asoc_simple_parse_dai` was made static (private) - I.E. You are supposed to use one of the public functions which wraps it. An ugly way of doing it would be to make another private static copy, which was what @hhuysqt did with one such part. I did look at his work after I finished mine - it is more or less the same, but in one big lump (and also some unrelated/questionable extra). I don't think it is a good idea to do such modification in one lump wholesale, though I might have presented mine that way if I have not anticipated reviews/questions.

HinTak, 2020-04-26:
The device tree overlay tells the kernel what composite sound hardware is present and how it relates to the individual ICs and circuitry. So it is not surprising that just loading the overlay gives alsa the ability to "see" a card. The c part of the respeaker driver code is strictly speaking a collection of driver code for the individual IC/circuitry bits, connected by the dt overlay.

StuartIanNaylor, 2020-04-26:
> I have looked at the log, for some reason `asoc_simple_parse_dai` was made static (private) - I.E. You are supposed to use one of the public functions which wraps it. An ugly way of doing it would [...]

With all honesty the driver situation with these cards is extremely bad. Even on raspbian the lockdown to a previous kernel can cause many problems and restrict use of a very varied distro.
The use of the PI4 has a big interest in 64bit OS due to perf gains and Ubuntu until Raspbian catches up does provide a modern LTS 64 bit kernel and has huge interest.
Problems do go further though as from watching the install in progress and the large amount of libs it installs for a 'simple_card' I have already run into problems where why the cards needs a JackD lib I am unsure but installing a voice AI Mycroft causes a dependency clash and its replaced by another lib and renders no audio.

I have little idea about the api and how kernel changes have made that problematic but all I know is the kernel lockdown in Raspbian is extremely bad and massively restrictive in the use of the Pi.
Also for some reason a linux sound card driver seems to be sensitive to what is installed as audio packages, which is also extremely bad.
The driver installs services and overwrites asound.conf and settings and the question is why.
What that install does, why and reason is a mystery and the only thing blatantly clear is that its extremely bad.

We need core drivers that are not dependent on installable packages that runs on Raspbian and the latest LTS.
So its 4.19 and 5.4 and current whats in between actually matters little as the current support for 4.19 and 5.4 is somewhere between non existent and absolutely woeful.

StuartIanNaylor, 2020-04-26:
> The device tree overlay tells the kernel what composite sound hardware is present and how it relates to the individual ICs and circuitry. So it is not surprising that just loading the overlay gives [...]

I was looking at https://github.molgen.mpg.de/git-mirror/linux/blob/master/sound/soc/codecs/wm8960.c

```
MODULE_DESCRIPTION("ASoC WM8960 driver");
--
  | MODULE_AUTHOR("Liam Girdwood");
```
Yeah I guess all we are missing is codec handling.

HinTak, 2020-04-26:
The respeaker, at least the 6-mic version I have, have 3 on-board ICs, ac101 and ac108 . The c code glues in those.

StuartIanNaylor, 2020-04-26:
That is my point we have 2mic, 4  mic, 6 mic with various chips all sharing a single driver that unless you use a single old dated kernel none work!
The C code does not glue those in as they are sperate hardware but a single driver does.

HinTak, 2020-04-26:
There is already an issue opened about removing the forced kernel downgrade - https://github.com/respeaker/seeed-voicecard/issues/209 - I am all for it. To be honest it was a temporary workaround for an upstream issue that was fixed a few months ago (with my pushing them) so can be reverted. Indeed I have a patch to revert that which I meant to issue a pull for here.

While an against staying at 4.19, I understand the reason - the respeaker card is NOT a simple card, despite its size - it has two buttons on, and 12 LEDs . As I explained earlier: it is a composite device, or perhaps, better described as a daughter add-on extension circuit board.

The main issue is not raspbian vs ubuntu, or 32-bit vs 64-bit, but the amount of changes, re-organization and rewrites in the kernel itself between 4.19 and 5.4 . I have to constantly reference the kernel changelog to see what is changed and why it is changed, and how to adjust to it. I understand why seeed studio does not want to do this work... 

(btw, if you go to my page, under font validator's readme, there is a donation link - if either seeed studio or 3Rd party wants to fund the porting work...)

HinTak, 2020-04-26:
The connectivity describes the ICs, etc are in the dt overlays and they are also platform-independent (I.E. Works on non-linux too!). The individual IC drivers are actually all very modular - and, honestly not where the porting problem to new kernels is. Please stop advocating breaking into individual "card" drivers - you don't know what you are talking about.

The code is clean enough that for most of the work I have done so far, I was actually building it against x86_64 host rather than aarch64, and it is okay that way - despite x86_64 host does not even have an I2S bus for doing ARM-style audio!

HinTak, 2020-04-26:
I updated my v5.5 branch and it should build cleanly on 5.4+ onwards. Also took the chance to add a few fixes and remove the forced kernel downgrade. (you'll still need to correct for the difference in the overlay locations).

If it does not run correctly after install (and putting the overlay in the adjusted place) and reboot, please post the output of "dmesg".

(and perhaps a donation at the HinTak/Font-Validator link to get another sd card, at least, would be nice...)

StuartIanNaylor, 2020-04-27:
> The connectivity describes the ICs, etc are in the dt overlays and they are also platform-independent (I.E. Works on non-linux too!). The individual IC drivers are actually all very modular - and, [...]

I don't think I have ever seen a linux driver as such and when you have a WM8960 why are other cards installed and even services to detect cards that you will never get.
I don't have to know very much to notice the manner of this driver is very strange, that is just very apparent knowledge or not.

> I was actually building it against x86_64 host rather than aarch64, and it is okay that way - despite x86_64 host does not even have an I2S bus for doing ARM-style audio!

That is just bizzare you are admitting you are building a device driver on a platform that can not even run it.
I feel this weird twilight moment.

HinTak, 2020-04-27:
@StuartIanNaylor neither are not as bizarre as you think - if you look at the beginning of your raspbian boot message (output of dmesg) - you will find that the raspbian kernel itself is built from cross-compilation. It makes a lot of sense to cross-compile on a much more powerful host. This is very routine for ARM- development.

As I mentioned a few times, the respeaker card is really a composite device / multi-function daughter add-on board. It is also very common for ARM- based add-on boards to have a lot of variants / sub-models. Look under the asoc/soc directory of the kernel tree, rather than the sound tree. (soc = system on chip). You don't know what you are talking about.

HinTak, 2020-04-27:
@StuartIanNaylor - v4.19 was released in October 2018, and raspbian only switched to 4.19 from 4.14 before this winter. That's not bad going for a small company. And seriously, the respeaker works well enough for raspbian (the official pi distro).

I personally am not switching to ubuntu on my pi any time soon - my more interesting focus is this
https://github.com/HinTak/vdfs-driver/commits/master - it is Samsung's file system driver for reading their shipping smart speakers and TVs. The public driver code was based on v4.1, which was released in June 2015. Despite the kernel being 5 years old, samsung is still shipping their TVs and smart speakers with it, and in their upcoming products AFAIC. I am porting the file system driver code from v4.1 32-bit arm to v5.5/5.6 x86_64, so am doing a lot of kernel "archeology" anyway, so have a look at this.

You'd have thought a big company like Samsung, and for TVs costing thousands US dollars, they could afford to keep their software up to date? and runs on 64-bit arm ubuntu at least? LOL!

If you can afford 3 respeaker devices (plus 3 pi's to go with it...) - you can afford to buy me a small blank SD card.

StuartIanNaylor, 2020-04-28:
I doesn't work well.
It works on vanilla raspbian with a clean install.

On completely different services of other installs seeed.voicecard.service fails.
On pulseaudio install it often fails.

```
Apr 28 02:23:52 raspberrypi pulseaudio[669]: W: [alsa-sink-bcm2835-i2s-wm8960-hifi wm8960-hifi-0] alsa-sink.c: Resume failed, couldn't restore original sample set
Apr 28 02:23:52 raspberrypi pulseaudio[669]: W: [alsa-sink-bcm2835-i2s-wm8960-hifi wm8960-hifi-0] alsa-sink.c: Resume failed, couldn't restore original sample set
Apr 28 02:23:52 raspberrypi pulseaudio[669]: W: [alsa-sink-bcm2835-i2s-wm8960-hifi wm8960-hifi-0] alsa-sink.c: Resume failed, couldn't restore original sample set
Apr 28 02:23:52 raspberrypi pulseaudio[669]: W: [alsa-sink-bcm2835-i2s-wm8960-hifi wm8960-hifi-0] alsa-sink.c: Resume failed, couldn't restore original sample set
Apr 28 02:23:52 raspberrypi pulseaudio[669]: W: [alsa-sink-bcm2835-i2s-wm8960-hifi wm8960-hifi-0] alsa-sink.c: Resume failed, couldn't restore original sample set
```
This is my current journalctl and all I have done is install pulseaudio and enabled the module-echo-cancel

Its a complete fudge so that it reverts to a kernel because all you can make it work is with that and alsa.
I am not spending another penny with respeaker or Seeed again as its been a total waste of my my money presuming at least one will work well.
They don't and my money will be to buy a sound card that does.

Thank you.

HinTak, 2020-04-28:
@StuartIanNaylor - pulseaudio doesn't work is a pulseaudio problem. You should test with aplay / arecord to assure correct functionality. You have not heard that the author of pulseaudio (also being the author of the well-loved-not systemd) often get death threats for making strange and controversial decisions? :-)

If you need my advice on a sound device with a simplier driver that's up to date with current kernel development, have a look at the octo card of audioinjector.net . It offers similar functionalities (just rather not-pretty, as in the actual external physical appearance) with an "up-to-date to current kernel" software stack.

I thought the Samsung illustration is sufficient - Seed Studio builds the respeaker for a "standard raspberry pi setup" (i.e. raspbrian). Anything else, well, it is extra, and I'd be happy they do it but not too upset if they don't. They never claim that it would run on anything else. Until Raspbrian itself moved to 64-bit and a more current kernel, I am okay with anything being DYI. Same with Samsung's TV - okay they charge a lot of money. The TV is for watching TV programmes, and I am not expecting any big help from them on reading their firmware files (my fun activity) any time soon. Expecting Samsung to want to help me to run their firmware file system driver on current desktop linux, is ridiculous.

HinTak, 2020-04-28:
Also, I specifically asked for the output of "dmesg" if it does not work. journalctl is not dmesg, it is not, honestly. "dmesg" gives the kernel ring buffer's messages, and is a lot lower level than jouralctl.
BTW, journalctl (part of systemd) is from the same guy who wrote pulseaudio. You cannot use one of  his tools to find out what's wrong another piece of his software! He does not write buggy software (<sarcasm>).

StuartIanNaylor, 2020-04-28:
@HinTak 

>  (also being the author of the well-loved-not systemd) often get death threats for making strange and controversial decisions? :-)

Its actually unnerving to speak to you, its a tirade of BS.
Many thanks for the waste of my time and purchases.
Goodbye and hopefully get better. 
Best wishes.

HinTak, 2020-04-28:
@StuartIanNaylor : the guy who wrote pulseaudio, also wrote systemd later (in that order, historically). Quite well-known fact. Many people try to avoid using either, also quite a well-known fact.

I am not sure what kind of tech support you expect for what is, in reality about $40-60 US piece of audio equipment, and what can be reasonably expected. I gave my advice - injector.net 's audio device is functionally similar, and driver works with up-to-date kernels (but the board itself is not aesthetically pretty!).

A typical Samsung smart TV is in x10 price range, and yet I am not expecting that Samsung to respond to my request of using it outside of what they claim to support... Just saying.

gradil, 2020-07-06:
In my case, I'm locked to use:
 `4.15.0-37-generic #40~upboard06-Ubuntu SMP Wed Nov 27 11:29:44 UTC 2019 x86_64 x86_64 x86_64 GNU/Linux`

I'm using an UP2 (upsquared) Board from AAEON/intel, with R-PI compatible GPIO (already using other Pi-hat boards successfully ).

Is there any support prior to linux kernel v5.3.0? I've not been able to get a workaround.

HinTak, 2020-07-06:
@gradil the normal official master should just work for such old kernel.( raspbian ships 4.19)

gradil, 2020-07-07:
@HinTak I'm not on raspbian, so I get the errors:

/boot/overlays not found or not a directory
dtparam not found
dtoverlay not found
Errors found, exiting.

So given this I'm trying a solution similar to @hhuysqt solution's.

HinTak, 2020-07-07:
@gradil ubuntu put files in different places compared to debian . I think you need to manually copy them into /boot/firmware instead of /boot , or something like that. But there is no code (compiled code) change - just file locations are different.

gradil, 2020-07-07:
@HinTak Yes it's the approach I'm currently trying. If it really works that linearly I think it should be something to add in the  Readme

HinTak, 2020-07-07:
Where files are located is really installation/packaging...

HinTak, 2020-08-07:
FYI. See #240 , which is likely a 64-bit issue.

HinTak, 2020-08-23:
@cyberh0me here it is:

- clone my repo https://github.com/HinTak/seeed-voicecard , switch to the v5.5 branch (it is now the default branch, so switching may not be needed)

- run the "./ubuntu-prerequisite.sh" script

- then run "./install.sh" like you do with upstream

The prerequisite script was written based on my notes/command history while setting my 6-mic up; and it works fine. There are probably some mistakes/typos. I am okay with Seeed Studio or others wanting to sponsor me on testing etc (just click the donate link in my FontValidator repo) - but I don't work for Seeed Studio, so this is provided as is, upto a point where it helps me to make mine work better in the future.

StuartIanNaylor, 2020-08-24:
There is also this repo that doesn't need the scripts and services that respeaker provide.

https://github.com/pguyot/wm8960

HinTak, 2020-08-24:
@StuartIanNaylor I already explained elsewhere - the 2-mics device uses wm8960 which a number of drivers exists, including built-im to the stock latest linux kernel (i.e. You do need extra drivers if you use the latest kernel). The 4-mics and 6-mics devices uses the ac101/ac108 for which this is currently the only option AFAIK. If you have a 2-mics device, yes, please feel free to use the alternatives, including just using the latest kernel itself without anything else!

HinTak, 2020-08-24:
I don't agree with Seeed Studio's policy of downgrading to kernels which are a few years old either. Hence the "v5.5" branch, and it is the default branch on mine, because latest raspbian and ubuntu are both 5.4 based. There is a "v5.7" branch for more recent kernels than 5.4 too, and I imagine if there are future incompatible changes in the kernel, I'll make new branches and switch the default too.

https://github.com/HinTak/seeed-voicecard

cyberh0me, 2020-08-24:
i absolutely agree
this produces update problem issues in the future and makes devices unsecure

haoyangw, 2020-08-25:
@HinTak sorry for the trouble, I tried installing this driver from your repo, by running ubuntu-prerequisite.sh first followed by install.sh and reboot, on Ubuntu Mate 20.04 kernel 5.4-1015 on Raspberry Pi 4. But I rebooted a couple times and the HAT is intermittently working and not working(when it's not working, it's detected under aplay/arecord -l, but the subdevice count would be 0/1, while when it's working, that same count is 1/1). I have the 2Mic HAT. Do you have any suggestions/fixes for this? I'm happy to donate you a coffee or sdcard for this

HinTak, 2020-08-25:
@haoyangw - could you compare the output of `dmesg` right after boot for a success and failure? Also `lsmod`? Also see if adding `dtoverlay=seeed-2mic-voicecard` to /boot/firmware/usercfg.txt helps. I also wonder about loose connection or a flawed device though.

haoyangw, 2020-08-25:
@HinTak Hmm `dmesg` looks the same, both for success and failure. `lsmod` and `dtoverlay -l` looks the same too, seeed-2mic-voicecard dtoverlay and kernel module were both loaded successfully. The only clue I've noticed is that when the HAT is working, pulseaudio would only detect the HAT but fail to detect the onboard Headphone Jack and HDMI Input as audio devices, leaving only 2 audio devices in Pulseaudio, Dummy Output and seeed-2mic-voicecard. When I do a `alsa force-reload`, and get all 3 audio devices detected by pulseaudio, the HAT would stop working and `arecord -l` would indicate subdevices of seeed-2mic-voicecard as '0/1'. Do you have any ideas as to what's happening here? It doesn't make sense to me

HinTak, 2020-08-25:
@haoyangw try adding `dtparam=audio=on` to the beginning of  `/boot/firmware/usercfg.txt` ?

haoyangw, 2020-08-27:
@HinTak Yeah I did, before I found ur repo, I tried installing drivers from the official repo, both using manual dkms commands and '`make`' followed by '`make install`', and I tried the wm8960 repo linked above when that didn't work. Then I finally tried ur repo. But I've tried my best to delete/uninstall the previous drivers completely by undoing the commands detailed in the installation procedure for each of my installation attempts, I'm not sure if that has left over any code that is conflicting with the currently-installed drivers? Hope there's any way to fix this without re-installing my system from scratch xP

HinTak, 2020-08-27:
@haoyangw argh, that makes life complicated...

- Check the content of `/etc/module*` (both the file and down the directory) . You should have just a single "snd-soc-wm8960" entry. 

- If you do `sudo find /lib/modules -type f -name "*wm8960*"` , you should only have "snd-soc-wm8960.ko" . Remove any other which says "wm8960*", then run `sudo depmod -a`.

- check the content of`/boot/firmware/*.txt` - if you have any `wm8960` lines, remove them. Also if you have a `/boot/firmware/overlays/wm8960.dtbo` file, remove it.

- after boot, `lsmod` should only have one entry referring to `wm8960`, which is `snd_soc_wm8960` (or `snd-soc-wm8960`). If there is more than one, you need find find out where (`/etc/modules` or `/boot/firmware/*.txt` ) refers to it.

haoyangw, 2020-08-27:
@HinTak Thanks so much for taking the time for this, but I've checked all 4 items u mentioned and there are no duplicates/conflicts. But the most curious discovery I just made is that currently: after a reboot and not touching anything(no `alsa force-reload` or anything), I found all my devices detected by Pulseaudio(checked via `pacmd list-sources` and `pacmd list-sinks`), and `arecord -l` shows the seeed-voicecard device having `subdevice: 1/1`, and after testing, audio is recorded just fine, **but** amazingly, `dtoverlay -l` lists only the 2 dtparams(`i2c_arm` and `spi`), and **no** seeed-2mic-voicecard. However, checking `systemctl status seeed-voicecard`, I see the line `true`, probably from the `dtoverlay -d $OVERLAYS $overlay || true`. I'm guessing that means the dtoverlay was already loaded before that command? This is the first time I've managed to get all 3 audio devices detected by pulseaudio **and** recording works(`arecord -l` has subdevice 1/1), and curiously all those times before where the recording **didn't** work, `dtoverlay -l` lists seeed-2mic-voicecard, and `systemctl status seeed-voicecard` does not show `true`, but shows `dtoverlay -d /boot/firmsare/overlays seeed-2mic-voicecard` in place of that line. Do you happen to have any ideas what is going on? I wanna reboot a few times to confirm the stability of this system, but I'm worried it's only temporary. Thanks so much for your help once again

HinTak, 2020-08-27:
@haoyangw so do you have a `dtoverlay=seeed-2mic-voicecard` in `/boot/firmware/usercfg.txt`, and it does not have a `#` before it, right? Leave it at that. Find the line `dtoverlay -d $OVERLAYS $overlay || true` in `/usr/bin/seeed-voicecard`, change it to just `true`. That should improve stability.

@Pillar1989 - why did you do both dynamic and static overlay in 277aeac ?

haoyangw, 2020-08-28:
@HinTak yeah I do, that line without being commented(`#`). Weirdly, just now I checked `pacmd list-sources` again and it's back to Dummy Output despite me not rebooting or touching anything(just left the device alone). I tried replacing the line you mentioned with just `true`, but after a reboot, all 3 of my audio devices are loaded, but seeed-voicecard does not work(`arecord -l` shows `subdevice 0/1` again). After some `alsa force-reload`s, it's back to dummy output + seeed-voicecard working again, but I made one observation: Previously when all 3 of my audio devices worked and `arecord` recorded fine, the `seeed-voicecard` audio device was `card 0` under `aplay -l`. Also, the other scenario when the voicecard works was when it was only dummy output + the seeed-voicecard, which made me feel like `seeed-voicecard` needs to be loaded first/by itself for it to work. Do you have any further ideas? Really confused here other than this observation. If you have any debugging I can help with(logs or testing), I would be glad to help. Sorry for the trouble and thanks so much!

HinTak, 2020-08-28:
@haoyangw This still sounds a lot like multiple drivers trying to bind to the same device. Besides the wm2860 itself (which has multiple drivers), the on-board audio apparently also have multiple drivers for 5.4... Anyway, the way to diagnose problems with multiple drivers is this:

- you need to compare carefully the output of `dmesg` when it works and when it does not. In particular, the order of events - there is a timestamp in front of each line. and watch anything to do with 'i2s', 'snd', 'sound', 'wm8960' and 'bcm28xx' (the on-board sound chip). You can save it to 'dmesg > saved-file' to compare, or even attach/post here.

- also watch carefully what 'lsmod' outputs. Again, save them with 'lsmod > savedoutput' if you need to.

- if a device works after reboot, then not after a while, you should watch what 'dmesg' says, and save as above if possible, for any unusual messages. same with 'journalctl -r' (reverse listing of the system logs).

- you can investigate driver binding order problems by manually loading and unloading drivers in different orders. But first, you must shut down any background process which accesses the device. I am not a big fan of pulseaudio, so I can't offer advice on it - but you must shut it down first.

- you can load/unload drivers by doing 'modprobe -v -r <name>' (-r for unloading), and 'modprobe -v <name>' (for loading, -v is just verbose), where name is as in 'lsmod' . Also carefully watch what dmesg says after each command. 'lsmod' is a dependency graph, and if you find one you cannot unload but the reference count is 0 (not required by other modules), that means a background process is using it, and you need to find out what process is.

Good luck...

mxklb, 2020-08-30:
This is issue handling unsatisfied me. Ok I know it should be possible to install the driver on ubuntu, and it's nice to get some hints on it here but this also looks like seeed-voicecard doesn't want to officially support ubuntu aarch64 in near future. 

I'm Ok with this "you'll have to hack it yourself", but I think this issue could be closed, marked as "won't be resolved".

This issue is alive since over a year now. For me this looks like it's possible but won't be supported out of the box. 

What do you think, is there any chance to get official (production ready end user) support for ubuntu aarch64?

[7 more replies]

closed 2020-09-09 by mxklb