respeaker/seeed-voicecard #212: stupid install.sh

kiorry, 2020-02-26:
I was introduced here by dingdang-robot project. when I write the dingdang-robot img into 16G TF card, I try to install this project, system told me , that is not enough space left (20M) on /boot . when I review the install.sh file.......
unit="${boot_space: -1}"
  if [[ "$unit" = "K" ]]; then
    echo "Error: Not enough space left ($boot_space) on /boot"
    exit 1
  elif [[ "$unit" = "M" ]]; then
    if [ "$free_space" -lt "25" ]; then
      echo "Error: Not enough space left ($boot_space) on /boot"
does any one can tell me , which raspberry img has 'G' unit on boot/?

mxklb, 2020-02-26:
Can you tell me want is the output of: `df -h | grep /boot ` on your device?
Please tell me exact output and I'll try to fix it .. thank you for reporting ..

kiorry, 2020-02-27:
> 
> 
> Can you tell me want is the output of: `df -h | grep /boot ` on your device?
> Please tell me exact output and I'll try to fix it .. thank you for reporting ..

pi@raspberrypi:~ $ df -h
Filesystem      Size  Used Avail Use% Mounted on
/dev/root       6.9G  3.3G  3.4G  50% /
devtmpfs        459M     0  459M   0% /dev
tmpfs           464M     0  464M   0% /dev/shm
tmpfs           464M   18M  446M   4% /run
tmpfs           5.0M  4.0K  5.0M   1% /run/lock
tmpfs           464M     0  464M   0% /sys/fs/cgroup
/dev/mmcblk0p1  253M   40M  214M  16% /boot
tmpfs            93M     0   93M   0% /run/user/1000

mxklb, 2020-02-27:
Thank you very much for you answer but I can't reproduce it. What I did:
- put your given `df -h` output into a file *dfh* (just for testing)
- use `cat dfh` instead of `df -h` at line 34 in `install.sh`:
`boot_line=$(cat dfh | grep /boot | head -n 1)`

Every things runs fine as expected, you have 214M space left, `install.sh` didn't exit 1 because of low boot disk space. Maybe there is something else going wrong here. I can't see any problem in the *Check for enough space on /boot volume* part of the install script.

What kind of shell are you using? The script is written for /bin/bash maybe this is it .. ???

kiorry, 2020-02-27:
I apologize for leave out this line what `if [ "$free_space" -lt "25" ]; then`

closed 2020-02-27 by kiorry