respeaker/seeed-voicecard #213: Will respeaker codec drivers(ac108.c) be upstreamed on Linux Kernel? Is there a plan in this regard?

AshaTalambedu, 2020-03-03:


HinTak, 2020-09-16:
Having looked at and studied the driver code quite a bit lately, (see the other issues , for example #248 ) I think it is unlikely. Speaking with first-hand experience as somebody who had submitted code to the kernel and still listed in the MAINTAINERS file, the linux kernel has a policy of not accepting anonymous submissions - every piece of code must be backed by a real person, with real name and identity. That quite spectacularly fails with the ac10x files in the header copyright section: the "reuumilla.com" is a joke-alias of allwinner.com. Apparently it is because allwinner.com management back in 2013 asks their employees to hide their identity .

Thus the respeaker codec drivers would never be accepted into upstream in its current form. It is also unacceptable to simply remove copyright messages by somebody else also. Thus until the original copyright owner, allwinner.com staff, come out and change the header, the codec driver cannot be upstreamed - it would be rejected based on the fake copyright identity alone.

Pillar1989, 2020-09-25:
https://github.com/respeaker/seeed-voicecard/issues/249

closed 2020-09-25 by Pillar1989

turmary, 2020-09-28:
I konw there are too much things about the codec driver which could not accept by linux community.
We haven't a plan to commit them to upstream in the near future.