respeaker/seeed-voicecard #216: [Feature Request] improve install.sh

Daenara, 2020-03-10:
After I just spent an hour watching my pi install those drivers only to notice I forgot to cd into the git directory I thought there might be a few ways to improve the install.sh:

- check the directory before doing all the stuff
- check if the kernel headers are already there (extracting those over the exact same thing is where it spent most time)
- maybe do the same with the kernel (not sure if it pulls stock kernel or does something custom, my ability to read scripts isn't the best)
- add in a way to detect what has already been done (install kernel and so on) and just do what hasn't worked (so you can resume where you were in case of errors)
- add a console switch to force the progress to restart (if the previous was implemented)
- actually end with an error message if important parts of the install failed (I got told to reboot and enjoy my install even though I had like 20 lines of errors above. If that happens in the middle of the install and not at the end it might be missed)
- there were quite a few rmdir messages during the uninstall where it tried to delete stuff but there were files still in there. No idea if that could mess stuff up or not

Nothing I wrote above has any impact on actually getting the driver to work but it would greatly improve usability.

Daenara

Pillar1989, 2020-09-25:
https://github.com/respeaker/seeed-voicecard/issues/249

closed 2020-09-25 by Pillar1989

turmary, 2020-09-29:
hi @Daenara 
the improving status.
- [ ] check the directory before doing all the stuff
- [ ] check if the kernel headers are already there (extracting those over the exact same thing is where it spent most time)
- [ ] maybe do the same with the kernel (not sure if it pulls stock kernel or does something custom, my ability to read scripts isn't the best)
- [ ] add in a way to detect what has already been done (install kernel and so on) and just do what hasn't worked (so you can resume where you were in case of errors)
- [ ] add a console switch to force the progress to restart (if the previous was implemented)
- [ ] actually end with an error message if important parts of the install failed (I got told to reboot and enjoy my install even though I had like 20 lines of errors above. If that happens in the middle of the install and not at the end it might be missed)
- [ ] there were quite a few rmdir messages during the uninstall where it tried to delete stuff but there were files still in there. No idea if that could mess stuff up or not