respeaker/seeed-voicecard #220: Raspberry PI4 + Respeaker 2mic stuck to black screen after driver installation

Deca, 2020-04-13:
Hi,
As the title say I'vs a Raspberry Pi4 and a respeaker 2mic, I did the driver installation with success then rebooted.
The Pi show briefly a rainbow screen then goes black 
I tried to edit the config.txt to go in safe mode without luck and now I'm stuck, any suggestion?

StuartIanNaylor, 2020-04-25:
The drivers only work with raspbian but Pi4 and respeaker 2mic works for me.
Even though it does force raspberrypi-kernel-headers_1.20190925+1-1_armhf.deb
So stuck with that kernel and header.
```
pi@raspberrypi:~ $ aplay -l
**** List of PLAYBACK Hardware Devices ****
card 0: ALSA [bcm2835 ALSA], device 0: bcm2835 ALSA [bcm2835 ALSA]
  Subdevices: 7/7
  Subdevice #0: subdevice #0
  Subdevice #1: subdevice #1
  Subdevice #2: subdevice #2
  Subdevice #3: subdevice #3
  Subdevice #4: subdevice #4
  Subdevice #5: subdevice #5
  Subdevice #6: subdevice #6
card 0: ALSA [bcm2835 ALSA], device 1: bcm2835 IEC958/HDMI [bcm2835 IEC958/HDMI]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 0: ALSA [bcm2835 ALSA], device 2: bcm2835 IEC958/HDMI1 [bcm2835 IEC958/HDMI1]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 1: seeed2micvoicec [seeed-2mic-voicecard], device 0: bcm2835-i2s-wm8960-hifi wm8960-hifi-0 []
  Subdevices: 1/1
  Subdevice #0: subdevice #0

pi@raspberrypi:~ $ uname -a
Linux raspberrypi 4.19.75-v7l+ #1270 SMP Tue Sep 24 18:51:41 BST 2019 armv7l GNU                                                                /Linux

```

Deca, 2020-05-09:
I managed to test it again on a fresh install
I get a succesful installation but black screen at the reboot
[image]

I guess is something related witht forcing the kernel, but I've no idea how to dig deeper in the issue with the failing boot....do you have any suggestion?

Deca, 2020-05-10:
I installed again the raspbian image and this time did a backup of the boot folder so I can hopefully restore after having installed respeaker driver 

it seems that they force the more recent kernel 4.19.97+ over original raspbian image  4.19.66-v7l+ armv7l that I have.

Unfortunately after reboot I can't do much anymore since I've a completely boot failure and black screen
Do you have any suggestion? 

below here full log of respeaker 2mic installation:
`Setting up dkms (2.6.1-4) ...
Setting up raspberrypi-kernel-headers (1.20200212-1) ...
run-parts: executing /etc/kernel/header_postinst.d/dkms 4.19.97+
run-parts: executing /etc/kernel/header_postinst.d/dkms 4.19.97-v7+
run-parts: executing /etc/kernel/header_postinst.d/dkms 4.19.97-v7l+
run-parts: executing /etc/kernel/header_postinst.d/dkms 4.19.97-v8+
Setting up git (1:2.20.1-2+deb10u3) ...
Processing triggers for man-db (2.8.5-2) ...
Reading package lists... Done
Building dependency tree       
Reading state information... Done
raspberrypi-kernel-headers is already the newest version (1.20200212-1).
raspberrypi-kernel-headers set to manually installed.
The following packages were automatically installed and are no longer required:
  libllvm6.0 rpi.gpio-common
Use 'sudo apt autoremove' to remove them.
The following packages will be upgraded:
  raspberrypi-kernel
1 upgraded, 0 newly installed, 0 to remove and 310 not upgraded.
Need to get 71.4 MB of archives.
After this operation, 92.0 MB of additional disk space will be used.
Get:1 http://archive.raspberrypi.org/debian buster/main armhf raspberrypi-kernel armhf 1.20200212-1 [71.4 MB]
Fetched 71.4 MB in 3min 22s (353 kB/s)                                                              
apt-listchanges: Reading changelogs...
(Reading database ... 171533 files and directories currently installed.)
Preparing to unpack .../raspberrypi-kernel_1.20200212-1_armhf.deb ...
run-parts: executing /etc/kernel/prerm.d/dkms 4.19.66+ /boot/kernel.img
run-parts: executing /etc/kernel/prerm.d/dkms 4.19.66-v7+ /boot/kernel7.img
run-parts: executing /etc/kernel/prerm.d/dkms 4.19.66-v7l+ /boot/kernel7l.img
Adding 'diversion of /boot/kernel.img to /usr/share/rpikernelhack/kernel.img by rpikernelhack'
Adding 'diversion of /boot/kernel7.img to /usr/share/rpikernelhack/kernel7.img by rpikernelhack'
Adding 'diversion of /boot/kernel7l.img to /usr/share/rpikernelhack/kernel7l.img by rpikernelhack'
Adding 'diversion of /boot/kernel8.img to /usr/share/rpikernelhack/kernel8.img by rpikernelhack'
Adding 'diversion of /boot/bcm2708-rpi-b-plus.dtb to /usr/share/rpikernelhack/bcm2708-rpi-b-plus.dtb by rpikernelhack'
Adding 'diversion of /boot/bcm2708-rpi-b.dtb to /usr/share/rpikernelhack/bcm2708-rpi-b.dtb by rpikernelhack'
Adding 'diversion of /boot/bcm2708-rpi-cm.dtb to /usr/share/rpikernelhack/bcm2708-rpi-cm.dtb by rpikernelhack'
Adding 'diversion of /boot/bcm2708-rpi-zero-w.dtb to /usr/share/rpikernelhack/bcm2708-rpi-zero-w.dtb by rpikernelhack'
Adding 'diversion of /boot/bcm2708-rpi-zero.dtb to /usr/share/rpikernelhack/bcm2708-rpi-zero.dtb by rpikernelhack'
Adding 'diversion of /boot/bcm2709-rpi-2-b.dtb to /usr/share/rpikernelhack/bcm2709-rpi-2-b.dtb by rpikernelhack'
Adding 'diversion of /boot/bcm2710-rpi-2-b.dtb to /usr/share/rpikernelhack/bcm2710-rpi-2-b.dtb by rpikernelhack'
Adding 'diversion of /boot/bcm2710-rpi-3-b-plus.dtb to /usr/share/rpikernelhack/bcm2710-rpi-3-b-plus.dtb by rpikernelhack'
Adding 'diversion of /boot/bcm2710-rpi-3-b.dtb to /usr/share/rpikernelhack/bcm2710-rpi-3-b.dtb by rpikernelhack'
Adding 'diversion of /boot/bcm2710-rpi-cm3.dtb to /usr/share/rpikernelhack/bcm2710-rpi-cm3.dtb by rpikernelhack'
Adding 'diversion of /boot/bcm2711-rpi-4-b.dtb to /usr/share/rpikernelhack/bcm2711-rpi-4-b.dtb by rpikernelhack'
Adding 'diversion of /boot/COPYING.linux to /usr/share/rpikernelhack/COPYING.linux by rpikernelhack'
Adding 'diversion of /boot/overlays/README to /usr/share/rpikernelhack/overlays/README by rpikernelhack'
Adding 'diversion of /boot/overlays/act-led.dtbo to /usr/share/rpikernelhack/overlays/act-led.dtbo by rpikernelhack'
Adding 'diversion of /boot/overlays/adau1977-adc.dtbo to /usr/share/rpikernelhack/overlays/adau1977-adc.dtbo by rpikernelhack'
Adding 'diversion of /boot/overlays/adau7002-simple.dtbo to /usr/share/rpikernelhack/overlays/adau7002-simple.dtbo by rpikernelhack'
Adding 'diversion of /boot/overlays/ads1015.dtbo to /usr/share/rpikernelhack/overlays/ads1015.dtbo by rpikernelhack'
Adding 'diversion of /boot/overlays/ads1115.dtbo to /usr/share/rpikernelhack/overlays/ads1115.dtbo by rpikernelhack'
Adding 'diversion of /boot/overlays/ads7846.dtbo to /usr/share/rpikernelhack/overlays/ads7846.dtbo by rpikernelhack'
Adding 'diversion of /boot/overlays/adv7282m.dtbo to /usr/share/rpikernelhack/overlays/adv7282m.dtbo by rpikernelhack'
Adding 'diversion of /boot/overlays/adv728x-m.dtbo to /usr/share/rpikernelhack/overlays/adv728x-m.dtbo by rpikernelhack'
Adding 'diversion of /boot/overlays/akkordion-iqdacplus.dtbo to /usr/share/rpikernelhack/overlays/akkordion-iqdacplus.dtbo by rpikernelhack'
Adding 'diversion of /boot/overlays/allo-boss-dac-pcm512x-audio.dtbo to /usr/share/rpikernelhack/overlays/allo-boss-dac-pcm512x-audio.dtbo by rpikernelhack'
Adding 'diversion of /boot/overlays/allo-digione.dtbo to /usr/share/rpikernelhack/overlays/allo-digione.dtbo by rpikernelhack'
[... cut ...]
 create mode 100644 ac108_6mic.state
 create mode 100644 ac108_asound.state
 create mode 100644 asound_2mic.conf
 create mode 100644 asound_4mic.conf
 create mode 100644 asound_6mic.conf
 create mode 100644 dkms.conf
 create mode 100644 wm8960_asound.state
Created symlink /etc/systemd/system/sysinit.target.wants/seeed-voicecard.service → /lib/systemd/system/seeed-voicecard.service.
------------------------------------------------------
Please reboot your raspberry pi to apply all settings
Enjoy!
`

Pillar1989, 2020-11-17:
It seems that there is a problem with the RPI itself. Maybe you can take another test.

closed 2020-11-17 by Pillar1989