respeaker/seeed-voicecard #221: Ubuntu 64 Driver install Raspberry

StuartIanNaylor, 2020-04-23:
The Pi 4 performs better by quite a bit as ArmV8 vs V7, had a go at installing but DKMS make doesn't like the libs of Ubuntu 20,04 LTS.

Would be really good if we could get Ubuntu added to Arch/Raspbian as that would give a 64bit choice.

Its kernel 5.4 and maybe with the dtbo but not sure what toconfigure.
It would be really great to get Ubuntu 64 added to the list and any news at what version will they become Kernel supported?

HinTak, 2020-04-23:
same/similar to https://github.com/respeaker/seeed-voicecard/issues/211 , and more info there.

StuartIanNaylor, 2020-04-23:
> same/similar to #211 , and more info there.

Yeah apols noticed after post anyone up to the job of forking a working repo for Ubuntu? :)
I hacked out much of instal.sh but then found the DKMS make process was more broken than my comfort zone.

closed 2020-04-23 by StuartIanNaylor