respeaker/seeed-voicecard #222: Pop/click at the beginning - Respeaker 2 mic

hawkeye217, 2020-04-26:
I have a 2 mic attached to a Pi Zero W and some small powered speakers that is working well, but I'm getting a click/pop at the beginning of each audio file as the card plays it.  Within several seconds after the audio stops, I hear another pop.  I'm guessing this has to do with the power state of the card.  

I can hack around this by running the following in another terminal window:

```
$ aplay -f cd /dev/zero
```

And this of course keeps the card "awake".  It works, but it's a bit of a hack.  With other cards, I noticed one could add a parameter to the module to prevent it from sleeping/powering down.  This, for example: https://major.io/2019/03/04/stop-audio-pops-on-intel-hd-audio/

Is there anything similar available for the 2 mic card?

Bougakov, 2020-07-22:
It appears that running `aplay -i` provides the same result, but without an unnecessary load on CPU. One can add it to `/etc/rc.local` (don't forget to add ` &` after the command to make it run in background).

mikethebee, 2020-07-22:
Thx @hawkeye217 I have had the same problem on all my Pi'and got around it by trimming off the first second with sox.

Psychokiller1888, 2020-09-25:
Usually it's due to the resampling in asound.conf

joshuajaharwood, 2020-10-20:
This happened to me with ALSA. The only way I found that got around this was by using Pulseaudio.

HinTak, 2020-10-20:
As I wrote in https://github.com/respeaker/seeed-voicecard/issues/251#issuecomment-708271666 , the Audio-Injector Octo card driver has an option to run the clock continuously, and not having such issue with power up / down .

Pillar1989, 2020-11-17:
Does this problem need to be discussed? You can open it again if you need to.

closed 2020-11-17 by Pillar1989

Psychokiller1888, 2020-11-17:
Sure it needs to be discussed, but more importantly, be adressed.... It's been years like this and to be honest, as @HinTak said, I don't see why I'd buy or promote respeaker products any further

mikethebee, 2020-11-18:
The reason I am persevering with my respeaker2 and keenly following the great work of @HinTak is because I needed several such boards for a project and it has been the only one I found that worked easily at the price, functionality and being readily available in quantity. The LEDs, Grove and button are easy to use with python. The amp gives more output than the onboard jack and the two mems mics allow directionality in a small form factor board. Seeed seem to be more responsive than many opensource hw companies, albeit it took some time.