respeaker/seeed-voicecard #226: Error: Not enough space left (20M) on /boot

d3-worgan, 2020-05-18:
Hi,

I have formatted an SD card, installed raspian stretch lite, and then followed the steps to install the seeed drivers for the 4mic array. Unfortunately I get the error:

```
Error: Not enough space left (20M) on /boot
```
when I run

```
sudo ./install.sh
```

How much space is needed? Why does it need space in boot? Has code in the repo changed or what else would have changed in raspbian stretch do you think?

It used to work following excactly the same steps? Do I need to increase the boot space?

HinTak, 2020-05-18:
Argh, because it tries to downgrade the kernel... There is an option not to downgrade.

Pillar1989, 2020-09-25:
https://github.com/respeaker/seeed-voicecard/issues/249

closed 2020-09-25 by Pillar1989

turmary, 2020-09-28:
Now there is message about lowest /boot space requirement, but not accurate.

marxenegls, 2020-11-20:
sob i have the same issue i was all giggles 5minute ago  thought my picroft was in the making  : (- now i'm gonna have to copy the sd card partitions .. format and recreate the whole thing anew - is that it . ? 
.wait a min,  but i thought boot partition is around 65Mb , it says it needs 25Mb .. somebody has been greedy, i'm gonna find out who and i'll be putting em on an acetic diet ..first i'll have to decouple shutdown and move to the operating room next door.. I'll let yous know yous guts i ripped out

marxenegls, 2020-11-20:
found 2 kernel.img + kernel7.img loosing one did it for me in my case the kernel.img since i'm on a pi3