respeaker/seeed-voicecard #228: Support 5.4 kernel

idreamerhx, 2020-05-22:
Hi,

I update raspberry pi zero  archlinux to 

Linux   5.4.40-1-ARCH #1 SMP PREEMPT Thu May 14 01:50:25 UTC 2020 armv6l GNU/Linux

The code seams not support new kernel.

idreamerhx, 2020-05-22:
This is /var/lib/dkms/seeed-voicecard/0.3/build/make.log

DKMS make.log for seeed-voicecard-0.3 for kernel 5.4.40+ (armv6l)
Fri May 22 19:38:15 CST 2020
make: Entering directory '/usr/lib/modules/5.4.40-1-ARCH/build'
  AR      /var/lib/dkms/seeed-voicecard/0.3/build/built-in.a
  CC [M]  /var/lib/dkms/seeed-voicecard/0.3/build/wm8960.o
  LD [M]  /var/lib/dkms/seeed-voicecard/0.3/build/snd-soc-wm8960.o
  CC [M]  /var/lib/dkms/seeed-voicecard/0.3/build/ac108.o
  CC [M]  /var/lib/dkms/seeed-voicecard/0.3/build/ac101.o
  LD [M]  /var/lib/dkms/seeed-voicecard/0.3/build/snd-soc-ac108.o
  CC [M]  /var/lib/dkms/seeed-voicecard/0.3/build/seeed-voicecard.o
/var/lib/dkms/seeed-voicecard/0.3/build/seeed-voicecard.c: In function 'seeed_voice_card_dai_init':
/var/lib/dkms/seeed-voicecard/0.3/build/seeed-voicecard.c:262:8: error: implicit declaration of function 'asoc_simple_card_init_dai'; did you mean 'asoc_simple_dai_init'? [-Werror=implicit-function-declaration]
  262 |  ret = asoc_simple_card_init_dai(codec, &dai_props->codec_dai);
      |        ^~~~~~~~~~~~~~~~~~~~~~~~~
      |        asoc_simple_dai_init
/var/lib/dkms/seeed-voicecard/0.3/build/seeed-voicecard.c: In function 'seeed_voice_card_dai_link_of':
/var/lib/dkms/seeed-voicecard/0.3/build/seeed-voicecard.c:315:8: error: implicit declaration of function 'asoc_simple_card_parse_daifmt'; did you mean 'asoc_simple_parse_daifmt'? [-Werror=implicit-function-declaration]
  315 |  ret = asoc_simple_card_parse_daifmt(dev, node, codec,
      |        ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~
      |        asoc_simple_parse_daifmt
/var/lib/dkms/seeed-voicecard/0.3/build/seeed-voicecard.c:322:8: error: implicit declaration of function 'asoc_simple_card_parse_cpu'; did you mean 'asoc_simple_parse_cpu'? [-Werror=implicit-function-declaration]
  322 |  ret = asoc_simple_card_parse_cpu(cpu, dai_link,
      |        ^~~~~~~~~~~~~~~~~~~~~~~~~~
      |        asoc_simple_parse_cpu
/var/lib/dkms/seeed-voicecard/0.3/build/seeed-voicecard.c:328:8: error: implicit declaration of function 'asoc_simple_card_parse_codec'; did you mean 'asoc_simple_parse_codec'? [-Werror=implicit-function-declaration]
  328 |  ret = asoc_simple_card_parse_codec(codec, dai_link, DAI, CELL);
      |        ^~~~~~~~~~~~~~~~~~~~~~~~~~~~
      |        asoc_simple_parse_codec
/var/lib/dkms/seeed-voicecard/0.3/build/seeed-voicecard.c:340:8: error: implicit declaration of function 'asoc_simple_card_parse_platform'; did you mean 'asoc_simple_parse_platform'? [-Werror=implicit-function-declaration]
  340 |  ret = asoc_simple_card_parse_platform(plat, dai_link, DAI, CELL);
      |        ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
      |        asoc_simple_parse_platform
/var/lib/dkms/seeed-voicecard/0.3/build/seeed-voicecard.c:365:8: error: implicit declaration of function 'asoc_simple_card_parse_clk_cpu'; did you mean 'asoc_simple_parse_clk_cpu'? [-Werror=implicit-function-declaration]
  365 |  ret = asoc_simple_card_parse_clk_cpu(dev, cpu, dai_link, cpu_dai);
      |        ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
      |        asoc_simple_parse_clk_cpu
/var/lib/dkms/seeed-voicecard/0.3/build/seeed-voicecard.c:373:8: error: implicit declaration of function 'asoc_simple_card_parse_clk_codec'; did you mean 'asoc_simple_parse_clk_codec'? [-Werror=implicit-function-declaration]
  373 |  ret = asoc_simple_card_parse_clk_codec(dev, codec, dai_link, codec_dai);
      |        ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
      |        asoc_simple_parse_clk_codec
/var/lib/dkms/seeed-voicecard/0.3/build/seeed-voicecard.c:379:8: error: implicit declaration of function 'asoc_simple_card_canonicalize_dailink'; did you mean 'asoc_simple_canonicalize_cpu'? [-Werror=implicit-function-declaration]
  379 |  ret = asoc_simple_card_canonicalize_dailink(dai_link);
      |        ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
      |        asoc_simple_canonicalize_cpu
/var/lib/dkms/seeed-voicecard/0.3/build/seeed-voicecard.c:384:8: error: implicit declaration of function 'asoc_simple_card_set_dailink_name'; did you mean 'asoc_simple_set_dailink_name'? [-Werror=implicit-function-declaration]
  384 |  ret = asoc_simple_card_set_dailink_name(dev, dai_link,
      |        ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
      |        asoc_simple_set_dailink_name
/var/lib/dkms/seeed-voicecard/0.3/build/seeed-voicecard.c:386:15: error: 'struct snd_soc_dai_link' has no member named 'cpu_dai_name'
  386 |       dai_link->cpu_dai_name,
      |               ^~
/var/lib/dkms/seeed-voicecard/0.3/build/seeed-voicecard.c:388:15: error: 'struct snd_soc_dai_link' has no member named 'codec_dai_name'
  388 |       dai_link->codec_dai_name
      |               ^~
In file included from /var/lib/dkms/seeed-voicecard/0.3/build/seeed-voicecard.c:19:
/var/lib/dkms/seeed-voicecard/0.3/build/seeed-voicecard.c:402:11: error: 'struct snd_soc_dai_link' has no member named 'cpu_dai_name'
  402 |   dai_link->cpu_dai_name,
      |           ^~
./include/linux/device.h:1759:47: note: in definition of macro 'dev_dbg'
 1759 |   dev_printk(KERN_DEBUG, dev, dev_fmt(fmt), ##__VA_ARGS__); \
      |                                               ^~~~~~~~~~~
/var/lib/dkms/seeed-voicecard/0.3/build/seeed-voicecard.c:406:11: error: 'struct snd_soc_dai_link' has no member named 'codec_dai_name'
  406 |   dai_link->codec_dai_name,
      |           ^~
./include/linux/device.h:1759:47: note: in definition of macro 'dev_dbg'
 1759 |   dev_printk(KERN_DEBUG, dev, dev_fmt(fmt), ##__VA_ARGS__); \
[... cut ...]
/var/lib/dkms/seeed-voicecard/0.3/build/seeed-voicecard.c:616:2: error: implicit declaration of function 'asoc_simple_card_clean_reference'; did you mean 'asoc_simple_clean_reference'? [-Werror=implicit-function-declaration]
  616 |  asoc_simple_card_clean_reference(&priv->snd_card);
      |  ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
      |  asoc_simple_clean_reference
cc1: some warnings being treated as errors
make[1]: *** [scripts/Makefile.build:266: /var/lib/dkms/seeed-voicecard/0.3/build/seeed-voicecard.o] Error 1
make: *** [Makefile:1694: /var/lib/dkms/seeed-voicecard/0.3/build] Error 2
make: Leaving directory '/usr/lib/modules/5.4.40-1-ARCH/build'

HinTak, 2020-06-06:
See https://github.com/respeaker/seeed-voicecard/issues/211and branches therein .

idreamerhx, 2020-06-17:
Thanks. according to this post, https://github.com/HinTak/seeed-voicecard/tree/v5.5 this repo work.

closed 2020-06-17 by idreamerhx