respeaker/seeed-voicecard #230: Support Raspbian-buster-arm64

StuartIanNaylor, 2020-05-28:
https://downloads.raspberrypi.org/raspios_arm64/images/raspios_arm64-2020-05-28/2020-05-27-raspios-buster-arm64.zip

HinTak, 2020-06-06:
The respeaker driver code has no arch-dependent code, so it should just work on 64-bit arm. Assuming raspbian 64-bit is also on the 4.19 kernels like raspbian 32-bit, it should just work.

HinTak, 2020-08-07:
FYI. See #240 , which is likely a 64-bit issue.

StuartIanNaylor, 2020-08-20:
I have found an alternative driver so its ok.

closed 2020-08-20 by StuartIanNaylor

HinTak, 2020-08-20:
FWIW, the 2-mic device is quite different from the two 4-mics and 4-mics ones - the 2-mics device is based on wm8960, while all the other 3 are based on ac108 (single or dual ac108, in the case of dual ac108, plus a ac101). 

The wm8960 code itself is just better supported - it is in fact built into later kernels - I have definitely seen it within 5.7.