respeaker/seeed-voicecard #233: Can't install drivers on raspbian with kernel 5.4

Son-kun, 2020-07-21:
I have raspbian buster on rpi 3B with kernel 5.4.51-v8+.
When I try to install drivers using this repo or HinTak fork I can't do it.
Everytime I get:
```Error! Bad return status for module build on kernel: 5.4.51+ (aarch64)```
In make.log i have information about missing asm/types.h file:
```
DKMS make.log for seeed-voicecard-0.3 for kernel 5.4.51-v7l+ (aarch64)
Tue 21 Jul 22:19:04 BST 2020
make: Entering directory '/usr/src/linux-headers-5.4.51-v7l+'
  AR      /var/lib/dkms/seeed-voicecard/0.3/build/built-in.a
  CC [M]  /var/lib/dkms/seeed-voicecard/0.3/build/wm8960.o
  CC [M]  /var/lib/dkms/seeed-voicecard/0.3/build/ac108.o
  CC [M]  /var/lib/dkms/seeed-voicecard/0.3/build/ac101.o
  CC [M]  /var/lib/dkms/seeed-voicecard/0.3/build/seeed-voicecard.o
In file included from ./include/linux/types.h:6,
                 from ./include/linux/list.h:5,
                 from ./include/linux/module.h:9,
                 from /var/lib/dkms/seeed-voicecard/0.3/build/wm8960.c:13:
./include/uapi/linux/types.h:5:10: fatal error: asm/types.h: No such file or directory
 #include <asm/types.h>
          ^~~~~~~~~~~~~
compilation terminated.
make[1]: *** [scripts/Makefile.build:266: /var/lib/dkms/seeed-voicecard/0.3/build/wm8960.o] Error 1
make[1]: *** Waiting for unfinished jobs....
In file included from ./include/linux/types.h:6,
                 from ./include/linux/list.h:5,
                 from ./include/linux/module.h:9,
                 from /var/lib/dkms/seeed-voicecard/0.3/build/ac108.c:15:
./include/uapi/linux/types.h:5:10: fatal error: asm/types.h: No such file or directory
 #include <asm/types.h>
          ^~~~~~~~~~~~~
compilation terminated.
In file included from ./include/linux/types.h:6,
                 from ./include/linux/list.h:5,
                 from ./include/linux/module.h:9,
                 from /var/lib/dkms/seeed-voicecard/0.3/build/ac101.c:27:
./include/uapi/linux/types.h:5:10: fatal error: asm/types.h: No such file or directory
 #include <asm/types.h>
          ^~~~~~~~~~~~~
compilation terminated.
make[1]: *** [scripts/Makefile.build:266: /var/lib/dkms/seeed-voicecard/0.3/build/ac108.o] Error 1
In file included from ./include/linux/compiler.h:180,
                 from ./include/linux/err.h:5,
                 from ./include/linux/clk.h:12,
                 from /var/lib/dkms/seeed-voicecard/0.3/build/seeed-voicecard.c:18:
./include/uapi/linux/types.h:5:10: fatal error: asm/types.h: No such file or directory
 #include <asm/types.h>
          ^~~~~~~~~~~~~
compilation terminated.
make[1]: *** [scripts/Makefile.build:266: /var/lib/dkms/seeed-voicecard/0.3/build/ac101.o] Error 1
make[1]: *** [scripts/Makefile.build:266: /var/lib/dkms/seeed-voicecard/0.3/build/seeed-voicecard.o] Error 1
make: *** [Makefile:1709: /var/lib/dkms/seeed-voicecard/0.3/build] Error 2
make: Leaving directory '/usr/src/linux-headers-5.4.51-v7l+'
```

Any idea how to install drivers for ReSpeaker in rpi with that kernel?

HinTak, 2020-07-21:
`asm/types.h` is part of the kernel-header package.

HinTak, 2020-07-23:
And see https://github.com/respeaker/seeed-voicecard/issues/228#issuecomment-645470225

iovsiann, 2020-07-27:
https://github.com/HinTak/seeed-voicecard/tree/v5.5 compiles successfully, but modprobe hangs and sudo ./uninstall.sh also hangs. I'm using Raspberry Pi 4B with kernel upgraded to 5.4.51-v7l+.

Son-kun, 2020-07-28:
So @HinTak fork for v5.5 version installed without any problems, thx 👍

closed 2020-07-28 by Son-kun