respeaker/seeed-voicecard #234: Rpi 4 fresh install kernel error

jeniara, 2020-07-23:
Did a new install with RPi 4 and a fresh SD-card but it seems to be some problem, related to KERNELRELEASE=5.4.51?

pi@raspberrypi:~/seeed-voicecard $ sudo ./install.sh
will compile with the latest kernel...
Hit:1 http://archive.raspberrypi.org/debian buster InRelease
Hit:2 http://raspbian.raspberrypi.org/raspbian buster InRelease
Reading package lists... Done
Building dependency tree       
Reading state information... Done
All packages are up to date.
Reading package lists... Done
Building dependency tree       
Reading state information... Done
git is already the newest version (1:2.20.1-2+deb10u3).
i2c-tools is already the newest version (4.1-1).
i2c-tools set to manually installed.
The following package was automatically installed and is no longer required:
  rpi-eeprom-images
Use 'sudo apt autoremove' to remove it.
The following additional packages will be installed:
  raspberrypi-kernel-headers
Suggested packages:
  python3-apport menu
The following NEW packages will be installed:
  dkms libasound2-plugins raspberrypi-kernel-headers
0 upgraded, 3 newly installed, 0 to remove and 0 not upgraded.
Need to get 26.4 MB of archives.
After this operation, 171 MB of additional disk space will be used.
Get:1 http://archive.raspberrypi.org/debian buster/main armhf raspberrypi-kernel-headers armhf 1.20200723-1 [26.2 MB]
Get:2 http://raspbian.playstar.se/raspbian buster/main armhf dkms all 2.6.1-4 [74.4 kB]
Get:3 http://raspbian.playstar.se/raspbian buster/main armhf libasound2-plugins armhf 1.1.8-1 [64.2 kB]
Fetched 26.4 MB in 11s (2,341 kB/s)                                            
Selecting previously unselected package dkms.
(Reading database ... 93778 files and directories currently installed.)
Preparing to unpack .../archives/dkms_2.6.1-4_all.deb ...
Unpacking dkms (2.6.1-4) ...
Selecting previously unselected package libasound2-plugins:armhf.
Preparing to unpack .../libasound2-plugins_1.1.8-1_armhf.deb ...
Unpacking libasound2-plugins:armhf (1.1.8-1) ...
Selecting previously unselected package raspberrypi-kernel-headers.
Preparing to unpack .../raspberrypi-kernel-headers_1.20200723-1_armhf.deb ...
Unpacking raspberrypi-kernel-headers (1.20200723-1) ...
Setting up libasound2-plugins:armhf (1.1.8-1) ...
Setting up dkms (2.6.1-4) ...
Setting up raspberrypi-kernel-headers (1.20200723-1) ...
run-parts: executing /etc/kernel/header_postinst.d/dkms 5.4.51+
run-parts: executing /etc/kernel/header_postinst.d/dkms 5.4.51-v7+
run-parts: executing /etc/kernel/header_postinst.d/dkms 5.4.51-v7l+
run-parts: executing /etc/kernel/header_postinst.d/dkms 5.4.51-v8+
Processing triggers for man-db (2.8.5-2) ...
Reading package lists... Done
Building dependency tree       
Reading state information... Done
raspberrypi-kernel is already the newest version (1.20200723-1).
raspberrypi-kernel-headers is already the newest version (1.20200723-1).
raspberrypi-kernel-headers set to manually installed.
The following package was automatically installed and is no longer required:
  rpi-eeprom-images
Use 'sudo apt autoremove' to remove it.
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
W: --force-yes is deprecated, use one of the options starting with --allow instead.

Creating symlink /var/lib/dkms/seeed-voicecard/0.3/source ->
                 /usr/src/seeed-voicecard-0.3

DKMS: add completed.

Kernel preparation unnecessary for this kernel.  Skipping...

Building module:
cleaning build area...
make -j4 KERNELRELEASE=5.4.51+ -C /lib/modules/5.4.51+/build M=/var/lib/dkms/seeed-voicecard/0.3/build....(bad exit status: 2)
Error! Bad return status for module build on kernel: 5.4.51+ (armv7l)
Consult /var/lib/dkms/seeed-voicecard/0.3/build/make.log for more information.
can not compile with this kernel, abort
please try compile with the option --compat-kernel

HinTak, 2020-07-23:
See https://github.com/respeaker/seeed-voicecard/issues/228#issuecomment-645470225

jeniara, 2020-07-24:
Thank you!

rm -R seeed-voicecard/
git clone https://github.com/HinTak/seeed-voicecard.git
cd seeed-voicecard
sudo ./install.sh

this did a clean compile!

jeniara, 2020-07-24:
But the driver is not found... see below. How do I fix the alsa setting?

pi@raspberrypi:~ $ arecord -L
null
    Discard all samples (playback) or generate zero samples (capture)
jack
    JACK Audio Connection Kit
pulse
    PulseAudio Sound Server
usbstream:CARD=ALSA
    bcm2835 ALSA
    USB Stream Output

HinTak, 2020-07-24:
Any messages in `dmesg` about ac101 or ac108?

Daenara, 2020-07-25:
I just ran into the same problem, so here is the dmesg output:
[image]

I also have trouble with the service not starting, might be related, might be not. I had the problem with the service before I made a clean install (and tons of recording problems, hence the clean install). 
[image]

HinTak, 2020-07-25:
The memory leak part is harmless, but you missed the important part : there should be a pair of "--- cut here---" and I need to see everything between the two. (in your first screen shot, you have one of them, and you did not include the part where the important info starts, a bit below)

Daenara, 2020-07-25:
Sorry, didn't see the cut here line, here is the whole thing:
[image]
[image]

I am assuming that the end trace line is the end of the cut, below that looks like some network stuff with ethernet and bluetooth.

HinTak, 2020-07-26:
Yes, thanks. That's the full info. I have never seen it that long though :-( . It is usually about 1/2 to 1/3 this size! ...

rotdrop, 2020-07-30:
I think the bug is fixed in my pull-request https://github.com/HinTak/seeed-voicecard/pull/3

Daenara, 2020-07-30:
Okay, I saw the fix merged and decided to give it a try. First of, the service now runs. I don't get an error anymore at the end of the install process when the service is started and it starts fine after a reboot. The device is also listed in `arecord -L` again.

[image]

I also checked `arecord -l` and there I found the first slightly strange thing. Back when my mic was working on my old install I am pretty sure it listed 8 subdevices, not just the one I get now.

[image]

But the amount of listed devices does not matter so I decided to test if the mic input actually works. This is the result:
[image]
The file I get from that is 44 bytes big and doesn't contain anything.

Here is the output of dmesg, I would have only included the relevant part but I could not find the cut here line at all this time. I think the relevant part is in the last screenshot but to be save i included the rest of the log in the spoiler tag.
<details>
  <summary>Complete log here</summary>

[image]
[image]
[image]
[image]
</details>

[image]

I hope this helps in fixing this, my voice assistant seems kinda useless without being able to hear me so I will help however I can to get this fixed soon.

HinTak, 2020-07-30:
I understand. I'll try to find some time to have a look.

Fmstrat, 2020-08-01:
Hi, I just tried https://github.com/HinTak/seeed-voicecard.git but I'm a little unnerved seeing this:
```
Unpacking raspberrypi-kernel (1.20190925+1-1) over (1.20200723-1) ...
```
Why is it installing a nearly year old kernel?

Daenara, 2020-08-01:
Because it is a kernel that is known to work. But the forced install is removed now in most branches, you need to get the v5.5 branch most likely, not the master one because that one is behind a bit. But right now there are problems with the driver, hence this thread, so the one with the old kernel should at least work.

HinTak, 2020-08-07:
@Daenara - argh, you also have a 4-mic card. I wonder if you get all 4 channels on your hardware with your hardware (I updated v5.5 with a fix a couple of days ago). See #240 , which is likely a 64-bit issue.

Daenara, 2020-08-07:
I don't use 64 bit, still on standard raspbian for quite a few reasons. I saw that you did something but I didn't get around to testing it, will do that after I slept and report back.

HinTak, 2020-08-07:
@Daenara That's good - I am quite sure my missing a channel in the middle of 6 is likely a 64-bit issue (the other channels sound correct to my ears - may not be numbered correctly - but they did record the right sound); so would love to hear from your testing, especially as you have different 4-mic hardware.

Daenara, 2020-08-07:
@HinTak I just tested, still not working for me. Everything looks good after install, I can see the device, in alsamixer I have all 4 channels but I still can't record.

Here is what happens still:
[image]
I might have a wrong command here, I always get it from the [respeaker4 wiki](https://wiki.seeedstudio.com/ReSpeaker_4_Mic_Array_for_Raspberry_Pi/) because I can't get my head around how to select a card to record with (my pi has 3 different inputs atm).

I do not think anything is different in the dmesg from last time I posted, but here you go:
[dmesg.txt](https://github.com/respeaker/seeed-voicecard/files/5042653/dmesg.txt)

As always I did my tests on a freshly installes raspbian os from may, which is the current one. If you need any other outputs or can think of things for me to run to help debug this, just tell me, I'd be glad to help. The respeaker hats are such a good idea in theory, but there always is some kind of issue with the driver and it never seems to run smoothly. I am really glad you are taking the time to try and fix this mess because the official support from seeed-voice is lacking most of the time.

Daenara, 2020-08-31:
@HinTak I just did another test with your latest changes, still nothing changed. Whenever I try to record anything I run into the pcm_read error and I get an error in dmesg
[image]

I have no idea why the respeaker4 is so different from the others but something is obviously still broken here. I couldn't see anything different in dmesg but still, here the output from this test:
[dmesg.txt](https://github.com/respeaker/seeed-voicecard/files/5147823/dmesg.txt)

This time I did the test on the new version of the raspberry pi os that came out 11 days ago but it is the same kernel so I don't think using that image instead of an updated one from May should make a difference.

HinTak, 2020-08-31:
@Daenara The I2S error is a red herring - even when my 6-mics device is worrking okay under the older raspian, it happens. I filed a issue with upstream for bcm2835 but did not get anywhere. Anyway, I think the issue is still open but unrelated. I see you have a usb audio device (C-Media USB Audio Device) connected, That's not a issue; but you are also using a USB serial console (Texas Instruments TI CC2531 USB CDC) ? That means you are using the usb power input line as serial console (to your desktop/laptop)? You might not be getting enough power to drive the pi though. Can you see if it works better powering it properly and accessing the pi by an ethernet cable, for example? Also the C-Media device can be unhooked?

Daenara, 2020-08-31:
I can try with everything unhooked but I doubt it will change anything. The C-Media is where my speakers are connected, the serial is my zigbee stick to control my lamps. It used to run fine with an older kernel with everything connected so I doubt it is a power issue. With my broken drivers I can record still, but I only get noise on all channels so the recording error should not be because of extra hardware.

I will make a test without everything later and report back thought.

HinTak, 2020-08-31:
@Daenara I thought you can't record, rather than the recording came out being wrong. I wonder if it is an alsa mis-config issue . Are `/etc/asound.conf` `/var/lib/alsa/asound.state` symlinks to files in `/etc/voicecard/` ?

Daenara, 2020-08-31:
Did my tests without anything outside connected to the pi, same problem.
I saved the output from the install (thought 3 lines might be missing because they outputted on my terminal), arecord -l, arecord -L, aplay -l and aplay -L into a file, as well as the output of dmesg.

[install.txt](https://github.com/respeaker/seeed-voicecard/files/5152185/install.txt)
[arecord-L.txt](https://github.com/respeaker/seeed-voicecard/files/5152186/arecord-L.txt)
[arecord-L-.txt](https://github.com/respeaker/seeed-voicecard/files/5152187/arecord-L-.txt)
[aplay-l.txt](https://github.com/respeaker/seeed-voicecard/files/5152188/aplay-l.txt)
[aplay-L-.txt](https://github.com/respeaker/seeed-voicecard/files/5152189/aplay-L-.txt)
[dmesg.txt](https://github.com/respeaker/seeed-voicecard/files/5152192/dmesg.txt)

And here is the output of me trying to record, did it with a verbose flag this time.
[image]

My pi is powered by a good usb power supply and has no power shortage even when connecting even more usb devices to it. I always use an ethernet cable on it, I don't trust wifi. Like I said earlier, that Texas Serial thing is just an usb dongle to use zigbee (reflashed bluetooth stick) but this test was without any usb ports used and I even unhooked the things from the i2c ports of my respeaker4.

Here is the asound stuff:
[image]
[image]

Anything else you can think of, just tell me and I can test for it

HinTak, 2020-08-31:
I am out of ideas - I had a look at the 6 files and they are all fairly normal...

Daenara, 2020-08-31:
@HinTak I might be grasping at straws here but I did a completely new install with the respeaker driver and the force downgrade flag to check if it might be my hardware. My mic is working fine then but I saw something in dmesg that is not in your version

[image]

Since the driver with the old kernel is working fine, I doubt it is a hardware issue, so it has to be something to do with whatever changed in the newer kernels that is unique to the respeaker 4. I did a web search on the error message I get and i found a few ppl where it was a confirmed hardware issue as well as [this](https://github.com/thesofproject/sof/issues/718) issue that was the same problem on a different hardware that got fixed. Now I myself don't understand why what they changed should fix anything but I normally don't mess around with syslevel drivers and c much, I normally stick to java, c# and python so that is way out of my depth. Maybe you could take a look at both the extra line as well as that github issue and see if it might give you an idea for a fix. 

Here is a full dmesg output of a working driver, as well as an arecord -L. I checked the -v output of the working recording against the one I posted earlier and it is the same, only difference is that I did not get a read error on the working driver.

[arecord_L.txt](https://github.com/respeaker/seeed-voicecard/files/5152949/arecord_L.txt)
[dmesg.txt](https://github.com/respeaker/seeed-voicecard/files/5152950/dmesg.txt)

I am somewhat torn between being happy that it is not a hardware issue and being sad about it, at least with a hardware issue I could replace my mic and be done with it, if it is a driver issue it will happen again even with a new one and my setup right now depends on the respeaker4 hardware.

HinTak, 2020-09-01:
@Daenara That's an interesting finding. I checked my logs. I have identical hardware, justt swapping SD card between 6-month-old raspbian which had the older kernel, vs current 1-month-old ubuntu, which has the new. My 6-mic device on the raspbian had that line, and on Ubuntu hasn't. However, the 6-mic device has a ac101 controlling two ac108 (and the linear-4-mic works the same - while the  square 4-mic you have as a single ac108), so it is entirely possible that anything to do with ac108 takes a secondary role on my device. I'll have a look why it does not show on ubuntu - at least it is an issue I can look at with my 6-mic hardware.

I know It is beating a dead-horse, but it is certainly disappointing that Seeed Studio does not put some resources (whether actual man power or sponsoring /funding external party to look at these things) into supporting **current** platforms.

HinTak, 2020-09-02:
@Daenara I think you are onto something in your finding with `set_sysclk`. That message is outputted by the `pr_info` at line 816 line https://github.com/HinTak/seeed-voicecard/blob/329c8090ea886243bd3d5ee03ced96550e7851ee/ac108.c#L816 - this means the routine is not being called. As I mentioned earlier, the 6-mics / linear devices have a ac101 controlling two ac108 so wrong/missing settings are probably auto-corrected / don't matter but the square 4-mics devices have only a single ac108 so a missing setting most certainly does. It is supposed to be called by `snd_soc_dai_set_sysclk` within the main `seeed-voicecard.c` calling into sound core calling back . Somehow sound core has changed enough between 4.19 and 5.4 and it does not happened anymore.

younes-professor, 2020-09-02:
@HinTak ,maybe this will help : 
when i execute the commande for recording with 3 channels ,i don't get the error 
arecord -Dac108 -f S32_LE -r 16000 -c 3 hello.wav -v

Recording WAVE 'hello.wav' : Signed 32 bit Little Endian, Rate 16000 Hz, Channels 3
Plug PCM: Route conversion PCM (sformat=S32_LE)
  Transformation table:
    0 <- 0
    1 <- 1
    2 <- 2
Its setup is:
  stream       : CAPTURE
  access       : RW_INTERLEAVED
  format       : S32_LE
  subformat    : STD
  channels     : 3
  rate         : 16000
  exact rate   : 16000 (16000/1)
  msbits       : 32
  buffer_size  : 8000
  period_size  : 2000
  period_time  : 125000
  tstamp_mode  : NONE
  tstamp_type  : MONOTONIC
  period_step  : 1
  avail_min    : 2000
  period_event : 0
  start_threshold  : 1
  stop_threshold   : 8000
  silence_threshold: 0
  silence_size : 0
  boundary     : 2097152000
Slave: Hardware PCM card 1 'seeed-4mic-voicecard' device 0 subdevice 0
Its setup is:
  stream       : CAPTURE
  access       : MMAP_INTERLEAVED
  format       : S32_LE
  subformat    : STD
  channels     : 4
  rate         : 16000
  exact rate   : 16000 (16000/1)
  msbits       : 32
  buffer_size  : 8000
  period_size  : 2000
  period_time  : 125000
  tstamp_mode  : NONE
  tstamp_type  : MONOTONIC
  period_step  : 1
  avail_min    : 2000
  period_event : 0
  start_threshold  : 1
  stop_threshold   : 8000
  silence_threshold: 0
  silence_size : 0
  boundary     : 2097152000
  appl_ptr     : 0
  hw_ptr       : 0

Pillar1989, 2020-09-25:
https://github.com/respeaker/seeed-voicecard/issues/249

closed 2020-09-25 by Pillar1989

WilliamVJacob, 2021-02-19:
I have added the fix to install Respeaker Driver for 64 bit Raspbian OS
Check my forked  seeed-voicecard  repository  for solution
https://github.com/WilliamVJacob/seeed-voicecad

HinTak, 2021-02-19:
Have had v5.8/v5.9 branches for some months https://github.com/HinTak/seeed-voicecard

BeckyHeath, 2021-02-22:
@HinTak I've just reflashed my OS and tried to install your soundcard. The install was successful but I couldn't view the soundcard after reboot

HinTak, 2021-02-22:
@BeckyHeath I'd ask the standard questions: any unusual message during install? And what dmesg says after boot?

BeckyHeath, 2021-02-22:
@HinTak Thanks for your speedy reply! I've got a clean OS that I'll re-run the install on! What version would be best?

HinTak, 2021-02-22:
There is not much to choose from as the install script updates your os to current, wherever you start from :-). Stay with 32-bit raspbian, unless you want to deal with #251 ...

BeckyHeath, 2021-02-22:
Ah no way! I forgot to save the install so uninstalled to re-install again and it all worked perfectly! Thank you!

HinTak, 2021-02-22:
@BeckyHeath good to hear. May I ask which device (2-/6-/4-square/4-linear -mics) you have? If you are feeling generous, there is always a donate button at the bottom of https://hintak.github.io :-).

BeckyHeath, 2021-02-22:
The 6-Mic! Thanks for your help but I usually use the `-- duration` flag to record for a fixed time but now it'll only stop recording if I interrupt manually (with ` ctrl c`) the scripts I'm running [are from this repo, (senors/Respeaker6Mic.py)](https://github.com/BeckyHeath/multi-channel-rpi-eco-monitoring). The full recording command is in the script as follows: 

` cmd = 'sudo arecord -Dac108 -f S32_LE -r 16000 -c 6 --duration {} {}'`
  `subprocess.call(cmd.format(self.record_length, wfile), shell=True`

Do you know why the duration flag won't work any more and is there a way I can fix that?

BeckyHeath, 2021-02-22:
@HinTak I just tried `sudo arecord -Dac108 -f S32_LE -r 1600 --duration 1 test.wav` straight into the terminal and I had the same problem when it'll only end once I interrupt with `ctrl c`

stantonious, 2021-02-22:
Is your sample rate right?  perhaps it should be 16000?

BeckyHeath, 2021-02-22:
Hey,

Great spot, yep that was a typo when writing on the forum I’ll fix it now – cheers

BeckyHeath, 2021-02-22:
@stantonious, yep! great spot that was a typo when I was writing the comment!

HinTak, 2021-02-23:
@BeckyHeath there is no need to "sudo" - the default user is part of the audio group, and have enough rights to record.

If you run the process in background ( append "&" to your command), can you see if the output file continues to grow beyond the specified time? The difference is whether it ignores the duration part, or the command hangs and goes into limbo after it finishes its job. The fact it responds to ctrl-c does not sound like it hangs though.

BeckyHeath, 2021-02-23:
so I just ran `arecord -Dac108 -f S32_LE -r 16000 --duration 1 test.wav` in the terminal, it said:

`Recording WAVE 'test.wav' : Sined 32 bit Littele Endian, Rate 16000, Mono` for a minute or so then I used ctrl-c and it said: 

`^CAborted by signal interrupt...`
`arecord: pcm_read:2145: read error: Interrupeted system call`

a file called "test.wav" appeared but it was an empty file

I tried running the background process `arecord -Dac108 -f S32_LE -r 16000 --duration 1 test.wav &` and it said recording but when I interrupted it said the same message as above, except no file generated at all this time. 

I tried just running `arecord -d 10 test_norm.wav` 

that said: 
`Recording WAVE 'test_norm.wav' : Unsigned 8 bit, Rate 8000 Hz, Mono`

then after ctrl-c: 
`^CAborted by signal Interrupt...`
`arecord: pcm_read:2145: read error: Interrupted system call`
and similarly, a file was generated but it was empty

Install and arecord data: 
[arecord_-L.txt](https://github.com/respeaker/seeed-voicecard/files/6029007/arecord_-L.txt)
[install (1).txt](https://github.com/respeaker/seeed-voicecard/files/6029008/install.1.txt)

I just tried to record in audacity and I can select 6 channels but nothing recorded, so I think something is up with either the soundcard or my hardware, I'll try with a different Pi and Array now.

BeckyHeath, 2021-02-23:
@HinTak I tried on a new pi, fresh sd. `arecord -d 10 test.wav` returned a 10s audio file, (but obviously, the signal was just a flat line at zero). 

I then installed the seeed-voicecard (your one HinTak) and I had the same issue where it doesn't end after 10s and the file is generated but it's empty

BeckyHeath, 2021-02-23:
@HinTak I just tried recording with the 6 channels and the scripts I've been working on and similarly a 6 channel file was generated but all the individual channels were empty (no flat line)

HinTak, 2021-02-24:
@BeckyHeath I forgot wav files probably has a size header and hence only written at end, so size while running has no meaning... Just a thought: have you checked the recording volume? It is one of the settings in amixer.

Pittermaennchen, 2021-03-20:
@HinTak I tried to install Respeaker 4-mic array drivers with
git clone https://github.com/HinTak/seeed-voicecard
changing to seeed-voicecard folder and running
sudo ./install
After reboot
arecord -l
still does not show the soundcard. What did I do wrong?

HinTak, 2021-03-20:
@Pittermaennchen the edit looks fine...

People, you should try filing new bugs instead of commenting on old ones...