respeaker/seeed-voicecard #235: Support Kernel 5.7

dony71, 2020-07-25:
I try this repo below which support kernel v5.5, but error compile for kernel v5.7
https://github.com/HinTak/seeed-voicecard/tree/v5.5
Anybody can help?

error message :

/var/lib/dkms/seeed-voicecard/5.5/build/ac108.c:667:59: error: ‘struct snd_soc_dai’ has no member named ‘playback_active’
  667 |  if ((substream->stream == SNDRV_PCM_STREAM_CAPTURE && dai->playback_active)
      |                                                           ^~
/var/lib/dkms/seeed-voicecard/5.5/build/ac108.c:668:60: error: ‘struct snd_soc_dai’ has no member named ‘capture_active’
  668 |   || (substream->stream == SNDRV_PCM_STREAM_PLAYBACK && dai->capture_active)) {
      |                                                            ^~
/var/lib/dkms/seeed-voicecard/5.5/build/ac101.c: In function ‘ac101_hw_params’:
/var/lib/dkms/seeed-voicecard/5.5/build/ac101.c:1083:17: error: ‘struct snd_soc_dai’ has no member named ‘capture_active’
 1083 |    if (codec_dai->capture_active && dmic_used && codec_aif1_fs[i].samp_rate == 44100) {
      |                 ^~
make[2]: *** [/usr/src/linux-headers-5.7.0-1-common/scripts/Makefile.build:272: /var/lib/dkms/seeed-voicecard/5.5/build/ac108.o] Error 1
make[2]: *** [/usr/src/linux-headers-5.7.0-1-common/scripts/Makefile.build:272: /var/lib/dkms/seeed-voicecard/5.5/build/ac101.o] Error 1
make[1]: *** [/usr/src/linux-headers-5.7.0-1-common/Makefile:1745: /var/lib/dkms/seeed-voicecard/5.5/build] Error 2

HinTak, 2020-07-25:
See https://github.com/HinTak/linux/commit/0f6011fd79a2fb92cb80177fd6bdc8aac3a3cd93#diff-4f1df2a4da465ae5be294130b3d705db - capture_active / playback_active became stream_active[...] lately.

What distro in shipping 5.7?

HinTak, 2020-07-25:
Please use the 5.7 branch I just made, adjusted for that recent change I mentioned. https://github.com/HinTak/seeed-voicecard/tree/v5.7

dony71, 2020-07-28:
@HinTak It's working ... you're the man !!

closed 2020-07-28 by dony71