respeaker/seeed-voicecard #236: install error

DerT94, 2020-07-29:
Hi,

I have the ReSpeaker 4-Mic Array and cannot install the driver.
The driver was already installed for the 2-Mic-Array

I use this fork
[https://github.com/HinTak/seeed-voicecard/tree/v5.5](https://github.com/HinTak/seeed-voicecard/tree/v5.5)

If i run "sudo ./install.sh" it stucks at the following point.

```
DKMS: install completed.
git init
Leeres Git-Repository in /etc/voicecard/.git/ initialisiert
git add --all
git commit -m "origin configures"
[master (Basis-Commit) 339c3b5] origin configures
 7 files changed, 1476 insertions(+)
 create mode 100644 ac108_6mic.state
 create mode 100644 ac108_asound.state
 create mode 100644 asound_2mic.conf
 create mode 100644 asound_4mic.conf
 create mode 100644 asound_6mic.conf
 create mode 100644 dkms.conf
 create mode 100644 wm8960_asound.state
```

HinTak, 2020-07-29:
And reboot and?

DerT94, 2020-07-30:
Nothing.
If I try "arecord -Dac108 -f S32_LE -r 16000 -c 4 hello.wav"
```
pi@testpi:~ $ arecord -D ac108 -f S32_LE -r 16000 -c 4 hello.wav
Aufnahme: WAVE 'hello.wav' : Signed 32 bit Little Endian, Rate: 16000 Hz, 4 Kanäle
arecord: pcm_read:2145: Read error: Input/output error
```

DerT94, 2020-07-30:
It is working with the master branch
https://github.com/HinTak/seeed-voicecard/tree/master

But with an update, the kernel is also updated and then it no longer works

HinTak, 2020-07-30:
What is your current kernel version?

baileyji, 2020-08-02:
@HinTak I'm running into this issue as well. I'm on Raspbian Buster with kernel 5.4.51-v7l+. Looking at the journal for seeed-voicecard it is exiting with a failure right these messages on you v5.5 branch

```
Aug 01 22:13:30 raspberrypi seeed-voicecard[1797]: + echo 'create seeed-4mic-voicecard asound configure file'
Aug 01 22:13:30 raspberrypi seeed-voicecard[1797]: create seeed-4mic-voicecard asound configure file
Aug 01 22:13:30 raspberrypi seeed-voicecard[1797]: + ln -s /etc/voicecard/asound_4mic.conf /etc/asound.conf
Aug 01 22:13:30 raspberrypi seeed-voicecard[1797]: + echo 'create seeed-4mic-voicecard asound status file'
Aug 01 22:13:30 raspberrypi seeed-voicecard[1797]: create seeed-4mic-voicecard asound status file
Aug 01 22:13:30 raspberrypi seeed-voicecard[1797]: + ln -s /etc/voicecard/ac108_asound.state /var/lib/alsa/asound.state
Aug 01 22:13:30 raspberrypi seeed-voicecard[1797]: + alsactl restore
Aug 01 22:13:30 raspberrypi seeed-voicecard[1797]: No state is present for card b1
Aug 01 22:13:30 raspberrypi seeed-voicecard[1797]: Found hardware: "bcm2835_hdmi" "Broadcom Mixer" "" "" ""
Aug 01 22:13:30 raspberrypi seeed-voicecard[1797]: Hardware is initialized using a generic method
Aug 01 22:13:30 raspberrypi seeed-voicecard[1797]: No state is present for card b1
Aug 01 22:13:30 raspberrypi seeed-voicecard[1797]: No state is present for card Headphones
Aug 01 22:13:30 raspberrypi seeed-voicecard[1797]: Found hardware: "bcm2835_headpho" "Broadcom Mixer" "" "" ""
Aug 01 22:13:30 raspberrypi seeed-voicecard[1797]: Hardware is initialized using a generic method
Aug 01 22:13:30 raspberrypi seeed-voicecard[1797]: No state is present for card Headphones
Aug 01 22:13:30 raspberrypi seeed-voicecard[1797]: + amixer -c 1 cset numid=3 1
Aug 01 22:13:30 raspberrypi seeed-voicecard[1797]: amixer: Cannot find the given element from control hw:1
```

arecord -l and arecord -L both look good to me.

Switching to master forced a kernel downgrade to 4.19 but things work. Let me know if you want additional debugging info.

HinTak, 2020-08-06:
@DerT94 I updated my v5.5 branch yesterday with I think a fix. Give it a try?

HinTak, 2020-08-07:
@baileyji  I'd be interested to know if you get all 4 channels on your hardware with your hardware (I updated v5.5 with a fix). See #240 , likely a 64-bit issue.

DerT94, 2020-08-07:
@HinTak 
I have now tested some things.
Hope not too many :)

git clone command:
`git clone -b v5.5 https://github.com/HinTak/seeed-voicecard.git ~/programs/seeed-voicecard_5.5`

### Rapsberry Pi 3 + 2-MIC HAT

Same as in #239 
Failed to start Seeed Voicecard service but the microphones work
Tested on two diffrent RPis

**Installation output**
```
pi@rhasspy1:~/programs/seeed-voicecard_5.5 $ sudo ./install.sh
Hit:1 http://archive.raspberrypi.org/debian buster InRelease
Hit:2 http://raspbian.raspberrypi.org/raspbian buster InRelease
Reading package lists... Done
Building dependency tree
Reading state information... Done
All packages are up to date.
Reading package lists... Done
Building dependency tree
Reading state information... Done
raspberrypi-kernel is already the newest version (1.20200723-1).
raspberrypi-kernel-headers is already the newest version (1.20200723-1).
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
Reading package lists... Done
Building dependency tree
Reading state information... Done
dkms is already the newest version (2.6.1-4).
git is already the newest version (1:2.20.1-2+deb10u3).
i2c-tools is already the newest version (4.1-1).
libasound2-plugins is already the newest version (1.1.8-1).
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
Error! There are no instances of module: seeed-voicecard
0.3 located in the DKMS tree.

Creating symlink /var/lib/dkms/seeed-voicecard/0.3/source ->
                 /usr/src/seeed-voicecard-0.3

DKMS: add completed.

Kernel preparation unnecessary for this kernel.  Skipping...

Building module:
cleaning build area....
make -j4 KERNELRELEASE=5.4.51+ -C /lib/modules/5.4.51+/build M=/var/lib/dkms/seeed-voicecard/0.3/build.....
cleaning build area...

DKMS: build completed.

snd-soc-wm8960.ko:
Running module version sanity check.
 - Original module
   - No original module exists within this kernel
 - Installation
   - Installing to /lib/modules/5.4.51+/kernel/sound/soc/codecs/

snd-soc-ac108.ko:
Running module version sanity check.
 - Original module
   - No original module exists within this kernel
 - Installation
   - Installing to /lib/modules/5.4.51+/kernel/sound/soc/codecs/

snd-soc-seeed-voicecard.ko:
Running module version sanity check.
 - Original module
[... 566 lines cut ...]
Error! Could not locate dkms.conf file.
File: /usr/src/seeed-voicecard-0.3/dkms.conf does not exist.
cp: cannot stat 'seeed-2mic-voicecard.dtbo': No such file or directory
cp: cannot stat 'seeed-4mic-voicecard.dtbo': No such file or directory
cp: cannot stat 'seeed-8mic-voicecard.dtbo': No such file or directory
cp: cannot stat '*.conf': No such file or directory
cp: cannot stat '*.state': No such file or directory
git init
Initialized empty Git repository in /etc/voicecard/.git/
git add --all
git commit -m "origin configures"
On branch master

Initial commit

nothing to commit
cp: cannot stat 'seeed-voicecard': No such file or directory
cp: cannot stat 'seeed-voicecard.service': No such file or directory
Failed to enable unit: Unit file seeed-voicecard.service does not exist.
Failed to start seeed-voicecard.service: Unit seeed-voicecard.service not found.
------------------------------------------------------
Please reboot your raspberry pi to apply all settings
Enjoy!
------------------------------------------------------
```

HinTak, 2020-08-09:
@DerT94 your install procedure for the last entry, Rapsberry Pi 4 + 4-MIC HAT, was wrong - you muat run inside the source directory, not the elsewhere.

As for the service start failure, that seems to be just from the amixer line - I picked up the "-c 1" change from somebody else. Need revisit.

HinTak, 2020-08-09:
@DerT94 I pushed a fix to v5.5 for #239 - I think it is the mixer line - you have a 3nd audio device connected, besides the built-in pi on-board Broadcom audio and the seeed studio, or did you disable the on-board audio?

Anyway, it looks like your 2-mics device is working okay, but one of your two 4-mic setup was a user error. So if you could post `dmesg` output after install/reboot for the 4-mic case?

lxne, 2020-08-12:
I just tested v5.5 with my 4mic linear array and it works. No issues so far.
Before installing v5.5 I updated raspbian buster and uninstalled old drivers.

HinTak, 2020-08-13:
@lxne thanks for trying. It looks like 2-mics and linear 4-mics and the 6-mics are okay with the v5.5 driver, but the square 4-mics isn't, and there is a 64-bit issue (#240 ) somewhere.

Pillar1989, 2020-09-25:
https://github.com/respeaker/seeed-voicecard/issues/249

closed 2020-09-25 by Pillar1989

Alex-Beh, 2021-03-10:
@HinTak Hi, I tried your fork branch v5.5 and it work. I can see the correct output when I `arecord -L`. But it cannot boot up after I reboot my raspberry pi. May I know why? Is the kernel die?

Alex-Beh, 2021-03-10:
@HinTak 

This is the output of `uname -r` :
Linux ubuntu 5.4.0-1029-raspi #32-Ubuntu SMP PREEMPT Thu Feb 4 23:00:55 UTC 2021 aarch64 aarch64 aarch64 GNU/Linux

(previously it only show  5.4.0-1029-raspi)

Output of dpkg --print-architecture:
`arm64`

HinTak, 2021-03-10:
@Alex-Beh It is most definitely #251 then. You can downgrade slightly with https://github.com/HinTak/RaspberryPi-Dev/releases/tag/Ubuntu-raspi-5.4.0-1028.31 . After downgrading and "apt-mark hold ..."; reboot, reinstall the driver as unusual. You do need to mark hold

HinTak, 2021-03-10:
I'll probably make a more recent build at some point.

Alex-Beh, 2021-03-10:
Sorry I am a bit confused on the instruction. 

So I need to 

1.  Download all the 5 .deb file.
2. apt-mark hold the file .deb file
3. sudo dpkg -i the 5 .deb file
4. "sudo tar -xzpvf ..." the addons.tgz file
5. cd /usr/src/linux-headers-5.4.0-1028-raspi
6. reboot

Then "sudo ./install.sh" to install the driver as usual.

Am I correct?

HinTak, 2021-03-10:
> Sorry I am a bit confused on the instruction.
> 
> So I need to
> 
> 1. Download all the 5 .deb file.
> 2. sudo dpkg -i the 5 .deb file
> 3. "sudo tar -xzpvf ..." the addons.tgz file
> 4. cd /usr/src/linux-headers-5.4.0-1028-raspi
> 5. sudo apt-mark ...
> 6. reboot
> 
> Then "sudo ./install.sh" to install the driver as usual.
> 
> Am I correct?

Only 4 deb files (no need to do buildinfo), and step 4 before step 3. Step 5 is 
`apt-mark hold linux-image-5.4.0-1028-raspi` (without the version and extension part) etc. If you make a mistake etc you can verify with `apt-mark showhold` .

Also Raspbian has moved to v5.10 already, so my repo now defaults to v5.9. Just stay with your existing clone - if you are doing a fresh new clone, you need to do `git checkout -b v5.5 origin/v5.5` to switch to the v5.5 branch before ./install.sh .

HinTak, 2021-03-10:
> > I'll probably make a more recent build at some point.
> 
> I'd love the ability to use a current build of Raspbian, I saw your v5.6-v5.9 and was hoping that one of them would have made all this go away. I'm inferring from your comment it does not?

32-bit raspbian is not (yet) affected by #251 .

baileyji, 2021-03-10:
> > > I'll probably make a more recent build at some point.
> > 
> > 
> > I'd love the ability to use a current build of Raspbian, I saw your v5.6-v5.9 and was hoping that one of them would have made all this go away. I'm inferring from your comment it does not?
> 
> 32-bit raspbian is not (yet) affected by #251 .

I don't follow. I switched to your fork as the official repo didn't work with the current release of Raspbian as of ~late-mid last year. I had to roll back to Linux raspberrypi 4.19.75-v7l+ #1270 SMP Tue Sep 24 18:51:41 BST 2019 armv7l GNU/Linux and would really love to update and use the current  5.4.x+ kernel in the more recent official releases.

HinTak, 2021-03-10:
@baileyji current raspbian is 5.10 (since Feb 2021, 6 weeks ago). Ubuntu groovy is 5.8 based and ubuntu focal is 5.4 based. Seeed studio's repo defaults to 5.4 since autumn 2020; my repo have branches for v4.19/v5.5/v5.8/v5.9 (v5.5 branch works for 5.4 kernels, while v5.9 branch works for 5.10 kernels). The default is v5.9 since a few weeks ago, switched over from v5.5.

Alex-Beh, 2021-03-11:
@HinTak I tried the steps, git checkout v5.5. And the OS still crash after I `sudo ./install.sh`. The error is  `kernel:[  260.431895] Kernel panic - not syncing: Aiee, killing interrupt handler!` and I think I didn't install your **RaspberryPi-Dev** fix correctly.

After I completed the step 1 to step 5, the kernel version supposed downgrade to `5.4.0-1028-raspi`? Because it still shown the `5.4.0-1029-raspi` version.

HinTak, 2021-03-11:
@Alex-Beh yup, you are seeing bug #251 ...

Alex-Beh, 2021-03-11:
Hi @HinTak, 

I tried to add --force-downgrade on the 4 deb file. 

- `sudo tar -xzpvf ...` in **/usr/src/linux-headers-5.4.0-1028-raspi** directory. 

- `sudo apt-mark hold linux-image-5.4.0-1028-raspi`

After reboot, `uname -r` still shown the old kernel. Looking forwards to further help. 感恩感谢

HinTak, 2021-03-11:
@Alex-Beh okay, thanks for trying the downgrade route. The ubuntu raspberrypi boot process is different from both ubuntu desktop/x86_64 and raspbian so it is not clear how one can boot an older kernel on ubuntu pi. 

Give me a couple of days to make another set of those debs.

Alex-Beh, 2021-03-12:
Thanks for your help. Let me know if anything I can help.

HinTak, 2021-03-12:
@Alex-Beh I made 5.4.0-1030 - 
https://github.com/HinTak/RaspberryPi-Dev/releases/tag/Ubuntu-raspi-5.4.0-1030.33 - and also updated the release notes to hopefully make the steps clearer. I'd like to have a look at how downgrading could be done on ubuntu raspberrypi (raspbian and ubuntu x86/desktop are well-documented) as it is somewhat tedious to make a new build when a new kernel package from ubuntu comes out. It takes about 5 hours to build, although, after having done the process a 4th time, I only need to attend to the first 30 minutes or so. If it works okay for you, please consider clicking donate at https://hintak.github.io .