respeaker/seeed-voicecard #237: 4 micro linear speaker, I2S sync error, sound output sampling rate seemingly wrong.

rotdrop, 2020-07-30:
Hi there,

I'm trying to use the 4 micro linear speaker array with Ubuntu 20.04 "focal" on RPI 3b+. This is kernel 5.4.0. I was able to comile the fork

https://github.com/HinTak/seeed-voicecard

and together with

https://github.com/HinTak/seeed-voicecard/pull/3

the thing actually loads ok:

```
[   28.241036]  ac10x i2c_id number: 0
[   28.241047]  ac10x data protocol: 0
[   28.347404]  ac10x codec count  : 1
[   28.347924]  ac10x i2c_id number: 1
[   28.347931]  ac10x data protocol: 0
[   28.457386]  ac10x codec count  : 2
[   28.458493] linking cpus, codecs and platforms
[   28.464952] input: seed-voicecard-headset as /devices/platform/soc/3f804000.i2c/i2c-1/1-0035/input/input0
[   28.479861] AC101 as Master
[   28.483667] seeed-voicecard soc:sound: ac10x-codec.1-0035 <-> 3f203000.i2s mapping ok
[   28.483710] seeed-voicecard soc:sound: ASoC: no DMI vendor name!
```

However, trying to follow the examples from the Seed-Wiki https://wiki.seeedstudio.com/ReSpeaker_4-Mic_Linear_Array_Kit_for_Raspberry_Pi/ the thing is not able to record:

```
rhasspy@rpi-sub:~$ arecord -Dac108 -f S32_LE -r 16000 -c 8 a.wav
Recording WAVE 'a.wav' : Signed 32 bit Little Endian, Rate 16000 Hz, Channels 8
overrun!!! (at least 0.150 ms long)
```
At this point the kernel log shows:

```
Jul 30 08:56:41 rpi-sub kernel: [ 1270.577674] bcm2835-i2s 3f203000.i2s: I2S SYNC error!
```

Trying to play sound samples:

```
rhasspy@rpi-sub:~$ aplay -Dac101 master-rhasspy-31b6a23d-fa4a-483e-8a4a-93c93226058b.wav 
Playing WAVE 'master-rhasspy-31b6a23d-fa4a-483e-8a4a-93c93226058b.wav' : Signed 16 bit Little Endian, Rate 16000 Hz, Mono
```

results in some sort of sound output, but seemingly at a rate which is much too low (takes too long and sound is too deep and slow by a factor of 2, 3 or 4, did not measure yet.

Using plughw directly does not change anything.

In another issue it is suggested that the recording issue is caused by a slow SD-Card, however, recording to ``/dev/null`` does not change anything, so I suppose this is not the case.

As the testing is done with the fork by Hintak: should issues be reported there? OTOH, your upstream version does not compile with recent kernels.

Pillar1989, 2020-07-30:
We are checking to see if the latest Raspberry PI kernel I2S and DMA are suitable for multi-channel data transmission.At present, users can also use the 4.19 kernel for testing. @rotdrop 
@HinTak  Will more than 2 channel sound cards in the Raspberry pi community work on the latest kernel? You know what?

rotdrop, 2020-07-30:
@Pillar1989 at least the two-channel sound card works (record and playback at the same time). However, that one does not use the seeed-voicecard driver. 4.19 is not an option for me as then I would have to choose another distro. The current Ubuntu uses 5.4.

HinTak, 2020-07-30:
@Pillar1989 : I think @rotdrop 's change is a step in the right direction but not quite complete. On the whole, I think Debian /Raspbian is too conservative, and using an out-dated Raspbian even more so. I am trying to stay with Raspbian mostly because it is "official", but I personally don't use Debian on the desktop and really don't like it. Ubuntu is bearable, and I really want things a bit more up to date than Ubuntu, so my desktop choice is fedora (and occasionally grab things off the next/pre-release fedora).

Pillar1989, 2020-07-30:
Where is the kernel source for RPI Ubuntu? Does it support more than two channel alsa card?  @HinTak

HinTak, 2020-07-30:
I think it is a variant of upstream raspberry pi one , https://packages.ubuntu.com/focal/linux-raspi2

HinTak, 2020-08-05:
@rotdrop I just updated v5.5 . Give it a try?

rotdrop, 2020-08-05:
@HinTak I certainly will. However, I'm on holiday until the 16th of August. I can't try it before.

HinTak, 2020-08-06:
@rotdrop no problem. Have a good holiday!

HinTak, 2020-08-07:
@rotdrop - are you on 32-bit or 64-bit ubuntu ? I'd be interested to know if you get all 4 channels on your hardware. See https://github.com/respeaker/seeed-voicecard/issues/240 .

HinTak, 2020-08-09:
@rotdrop interesting your linear 4-mic device uses the 8-mics driver like the circular 6-mics device I have, and different from the square 4-mics device in https://github.com/respeaker/seeed-voicecard/issues/236 . I'd definitely like to know if you can hear the individual channels in your 4-mics.

rotdrop, 2020-08-27:
@HinTak I can confirm that your current version of the driver also works with the 4-mic linear array. However, I just did a very short test, I'll let you know  later in the evening whether the 4 channels can record individual sound signals.

HinTak, 2020-08-27:
@rotdrop thanks for the confirmation! I found the issue with empty channels - it is amixer setting changed elsewhere with alsa. Just check your recording volume is not at the bottom.

Pillar1989, 2020-09-25:
https://github.com/respeaker/seeed-voicecard/issues/249

closed 2020-09-25 by Pillar1989