respeaker/seeed-voicecard #238: seeed_voicecard.service is failed using FORCE_KERNEL="1.20200601-1", but downgraded kernel works.

azamiya, 2020-08-05:
## Environment
Rpi4

```
Linux 4.19.118-v7l+ #1311 SMP Mon Apr 27 14:26:42 BST 2020 armv7l GNU/Linux
```

## Current behavior

Follow the download instruction

```
git clone https://github.com/respeaker/seeed-voicecard
cd seeed-voicecard
#sudo ./install.sh
sudo ./install.sh  --compat-kernel
sudo reboot
```

then after reboot, `seeed-voicecard.service` is failed.

```
$ sudo systemctl status seeed-voicecard.service
● seeed-voicecard.service - Seeed Voicecard service
   Loaded: loaded (/lib/systemd/system/seeed-voicecard.service; enabled; vendor preset: enabled)
   Active: failed (Result: exit-code) since Wed 2020-08-05 16:04:51 JST; 5min ago
  Process: 15604 ExecStart=/usr/bin/seeed-voicecard (code=exited, status=1/FAILURE)
 Main PID: 15604 (code=exited, status=1/FAILURE)
```

also no `seeed2micvoicec` on `aplay -l` or `arecord -l`

But,

```
$ sudo /usr/bin/seeed-voicecard
```

this works fine.

## Temporary solution

Downgrade `FORCE_KERNEL` version to `1.20200212-1`

`install.sh`
```
FORCE_KERNEL="1.20200212-1"
#FORCE_KERNEL="1.20200601-1"
```

then do `sudo ./install.sh --compat-kernel`, reboot.
finally I got
```
$ sudo systemctl status seeed-voicecard.service
● seeed-voicecard.service - Seeed Voicecard service
   Loaded: loaded (/lib/systemd/system/seeed-voicecard.service; enabled; vendor preset: enabled)
   Active: active (exited) since Wed 2020-08-05 16:11:04 JST; 8min ago
  Process: 428 ExecStart=/usr/bin/seeed-voicecard (code=exited, status=0/SUCCESS)
 Main PID: 428 (code=exited, status=0/SUCCESS)
```

HinTak, 2020-08-07:
cc @KillingJacky The available tags in the source code repos are:

```
raspberrypi-kernel_1.20200212-1
raspberrypi-kernel_1.20200512-2
raspberrypi-kernel_1.20200527-1
raspberrypi-kernel_1.20200601+arm64-1
raspberrypi-kernel_1.20200601-1
raspberrypi-kernel_1.20200717-1
raspberrypi-kernel_1.20200723-1
```

```
raspberrypi-kernel_1.20200212-1                 4.19.97
raspberrypi-kernel_1.20200512-2                 4.19.97
raspberrypi-kernel_1.20200527-1                 4.19.118
raspberrypi-kernel_1.20200601+arm64-1     5.4.42
raspberrypi-kernel_1.20200601-1                 4.19.118
raspberrypi-kernel_1.20200717-1                 5.4.51
raspberrypi-kernel_1.20200723-1                 5.4.51
```

As I mentioned in #223 , my buster box only gives me the option for raspberrypi-kernel_1.20200723-1 , which is 5.4.51 . I suspect you can only downgrade to what was in your cache, which was the previous kernel you were using, whichever version it was. So pinning on specific version is simply unmaintainable : there are at least 6 different stretch and 6-7 different buster.

Psychokiller1888, 2020-08-09:
Pfff, again a downgrade..... And this kills my apps on hardware using respeakers, because it ends up with `0 upgraded, 0 newly installed, 0 to remove and 2 not upgraded` which forbids any new installation....

HinTak, 2020-08-09:
@Psychokiller1888 wanna gives my v5.5 / v5.7 fork a try? That's the whole point why the branches were created.

Psychokiller1888, 2020-08-09:
Yeah, that's a branch in a fork :-) Found about it a while after posting here and I'm already using it, branch 5.5 for latest raspianOS image, with a respeaker 2. I'm releasing alpha version by tomorrow, got a nice pool of users with a wide variety of hardware, I'll report back on your fork if needed

HinTak, 2020-09-07:
I have updated the earlier kernel version table and put it up as https://github.com/HinTak/RaspberryPi-Dev/blob/master/Raspbian-Kernel-Releases.md - you can see that the 4 most recent releases are 5.4.x based. They are going that way with a brief reverse in  May.

Pillar1989, 2020-09-25:
https://github.com/respeaker/seeed-voicecard/issues/249

closed 2020-09-25 by Pillar1989

mohammad508, 2024-11-06:
> ## Environment
> Rpi4
>
> ```
> Linux 4.19.118-v7l+ #1311 SMP Mon Apr 27 14:26:42 BST 2020 armv7l GNU/Linux
> ```
>
> ## Current behavior
> Follow the download instruction
>
> ```
> git clone [...]

mohammad508, 2024-11-06:
Hi, I'm using a respeaker 4 mic array in Raspberry Pi 3 and Kernal 20230503, following the same instructions in the wiki, but when I am doing a record -L doesn't show up on the voice card. Can you help me, please?

mohammad508, 2024-11-06:
@HinTak can you help me

HinTak, 2024-11-06:
If you are using something as old as a 4.19 kernel, use the 4.19 branch - https://github.com/HinTak/seeed-voicecard - and make sure to NOT upgrade your kernel. Do a `apt mask...` to protect it, first. Good luck.

mohammad508, 2024-11-07:
am I using system 64 or 32? Which is Kernal for Raspbian os recommended to
install?

HinTak, 2024-11-07:
If you were using 4.19, likely it was 32-bit.