respeaker/seeed-voicecard #240: 64-bit driver on 6-mics missed 3rd channel

HinTak, 2020-08-07:
I managed to test on ubuntu 20.04.1 64-bit. 20.04 was no good for me, due to a lot of networking bugs and won't do headless - I spent most of a day just fighting the headless network configuration, then I see that 20.04.1 has been out since the time I downloaded 20.04 , and 20.04.1 came with substantial changes and bug fixes over 20.04 (LTS is a joke...) . It works more or less working correctly, on the 6-mics against kernel 5.4.x which ubuntu shipped - the individual channels of the 8-channels recording plays back correctly, except channel 3 comes out empty.

I went back to raspbian just to be sure it is not a broken microphone.

My first suspicion is that one of the `long` 's in the code should be uint32_t etc. Driver/hardware data are usually fixed size, so using variables which have different sizes on different platforms are definitely wrong.

Alternatively it could be the 419 to 5.4 changes. I suppose I could find out one way or the other if I switch to ubuntu 32-bit. But I don't feel like doing it - spending more money on another SD card. (Having two, one for main use, and another for throw-away tries, is alright - but really this needs a 3-way comparison...). However, if seeedstudio want to sponsor me to do the work, or others, please feel free to get in touch or just click the link  https://sourceforge.net/p/hp-pxl-jetready/donate/ (it is unrelated software but go to my paypal acct).

HinTak, 2020-08-09:
Hmm, almost *all* the usage of `long` in the code is wrong! Irq values, channels, various alsa parameters, are 32-bit.

HinTak, 2020-08-18:
cc @rotdrop  (where the `-c 1` came from). The amixer command when applied to newer kernel, does not enable headphone jack, but turning channel 3's recording volume really low ( 1 in the range of 0 to 255).

changlog for this fix:

    remove "-c 1" and also protect with test for older kernel / alsa
    
    The "-c 1" was simply wrong. The default (-c 0) sound device of
    older kernel / alsa has a numid=3 selector for auto vs headphone vs hdmi .
    Newer kernel simply has multiple devices. Also, headphone is -c 0 .
    
    $ amixer contents
    numid=2,iface=MIXER,name='Headphone Playback Switch'
      ; type=BOOLEAN,access=rw------,values=1
      : values=on
    numid=1,iface=MIXER,name='Headphone Playback Volume'
      ; type=INTEGER,access=rw---R--,values=1,min=-10239,max=400,step=0
      : values=-2000
      | dBscale-min=-102.39dB,step=0.01dB,mute=1
    
    TODO: For now, broadly dividing by kernel v4.x vs v5.x is good enough, but
    it would be nice to find out when / which kernel /alsa this is relevant.

closed 2020-08-18 by HinTak

HinTak, 2020-08-18:
it works perfectly for me on the 6-mic device with ubuntu 64-bit now. I'll add th ubuntu userland stuff in my branch soon.

turmary, 2020-09-29:
#249

turmary, 2020-09-29:
> TODO: For now, broadly dividing by kernel v4.x vs v5.x is good enough, but
> it would be nice to find out when / which kernel /alsa this is relevant.

I round a common solution, match by the widget name 'PCM Playback Route',
in kernel >= 5.4, amixer will just fail.

pi@raspberrypi:~ $ uname -r
4.19.75-v7l+
pi@raspberrypi:~ $ amixer contents
numid=3,iface=MIXER,name='PCM Playback Route'
  ; type=INTEGER,access=rw------,values=1,min=0,max=3,step=0
  : values=1
numid=2,iface=MIXER,name='PCM Playback Switch'
  ; type=BOOLEAN,access=rw------,values=1
  : values=on
numid=1,iface=MIXER,name='PCM Playback Volume'
  ; type=INTEGER,access=rw---R--,values=1,min=-10239,max=400,step=0
  : values=-2206
  | dBscale-min=-102.39dB,step=0.01dB,mute=1
numid=5,iface=PCM,name='IEC958 Playback Con Mask'
  ; type=IEC958,access=r-------,values=1
  : values=[AES0=0x02 AES1=0x00 AES2=0x00 AES3=0x00]
numid=4,iface=PCM,name='IEC958 Playback Default'
  ; type=IEC958,access=rw------,values=1
  : values=[AES0=0x00 AES1=0x00 AES2=0x00 AES3=0x00]

HinTak, 2020-09-29:
You did not see my latest code - it was fixed to the proper condition later.

turmary, 2020-09-29:
OK, I just saw it.