respeaker/seeed-voicecard #245: Conflict between corbosman/pigpiod install in docker-container and re-speaker 2

hugocoolens, 2020-08-28:
After installing corbosman/pigpiod in a docker-container re-speaker sound is slowed down and shifted in frequency in the low frequencies. See also the discussion on thr Rhasspy-forum: [](https://community.rhasspy.org/t/controlling-gpio-12-and-13-of-respeaker-2/1469/13?u=hugocoolens)

hugocoolens, 2020-08-29:
The conflict can be avoided by adding an extra command line in the docker-compose.yml to choose PWM instead of PCM 
This is the pigpiod-part:
pigpiod:
  image: corbosman/pigpiod
  container_name: pigpiod
  restart: always
  privileged: true
  command: -t 0

Pillar1989, 2020-11-17:
Is there anything we can do about it?

closed 2020-11-17 by Pillar1989

hugocoolens, 2020-11-18:
It would be nice if the workaround I gave would not be necessary.

kind regards,
Hugo