respeaker/seeed-voicecard #246: ReSpeaker 4-Mic with Raspberry Pi OS (previously called Raspbian) does not Work

younes-professor, 2020-09-01:
hello
I followed the instruction at https://wiki.seeedstudio.com/ReSpeaker_4_Mic_Array_for_Raspberry_Pi/. to install the driver
but it does not work ,the ReSpeaker 4-Mic is not detected 
-Raspberry Pi OS 2020-08-20 
-raspberry 3b
-ReSpeaker 4-Mic
please can you help

HinTak, 2020-09-01:
Same problem as https://github.com/respeaker/seeed-voicecard/issues/234

younes-professor, 2020-09-01:
@HinTak ,what is the latest version of  Raspbian in which the official instructions worked fine ?

HinTak, 2020-09-01:
@younes-professor none . See comment https://github.com/respeaker/seeed-voicecard/issues/238#issuecomment-670303966 - because the first step of install is to fetch a matching kernel header package (and/or downgrade), and raspbian is simply not keeping the older packages around. You might have some luck with old archived and complete images from raspbian (not `lite`).

younes-professor, 2020-09-01:
@HinTak thank you for you response ,i have successfully  installed the driver using
git clone https://github.com/HinTak/seeed-voicecard.git 
but now when i try to record ,it does not work and give me this error

Recording WAVE 'hello.wav' : Signed 32 bit Little Endian, Rate 16000 Hz, Channels 4
arecord: pcm_read:2145: read error: Input/output error

HinTak, 2020-09-01:
As I said it is same problem as #234 - I only have the 6-mics device (which also is very similar to the linear 4-mics device, but somewhat different from the square 4-mics device)so it is going to be a while if you are looking in my direction for help with the square 4-mics device. It is unfortunate that Seeed Studio staff don't seem to understand the rate of linux kernel development (very fast) and style of raspbian development (very slow and very under-resourced), and that raspbian does not keep old packages around (because they cannot afford the space).

younes-professor, 2020-09-02:
@HinTak ,yes i understand

j1nx, 2020-09-02:
@HinTak Can you do anything if you receive a full dmesg tracelog of the 4-MIC square HAT? There is something not right in the ac108 driver if it comes to 64bit I believe.

Like you said, the others do work fine now. It is just that 4mic currently that still needs a little change.

HinTak, 2020-09-02:
@j1nx @younes-professor  @Daenara the latest thought about the square 4-mic device is at https://github.com/respeaker/seeed-voicecard/issues/234#issuecomment-685257469 - it appears that my update of the driver code to current kernels is okay for the 6-mics (which I have) and the linear 4-mics (which is structurally very similar to the 6-mics) and I think I read somebody saying 2-mics (which has an entirely different chip) is okay too. The square 4-mics is like half a 6-mics but wired differently. I might be able to get ahead a bit by looking at and comparing how my 6-mics works under 6-month-old raspbian (which has the older kernel) and current ubuntu - for which I just swap two SD cards - but what would really help is actually having an actual square 4-mic device myself... 

While I have other projects that I could do the having a variety of platforms - 32-bit arm on raspbian/debian and 64-bit arm on ubuntu, in addition to my normal x86_64 - so I have two SD cards for the two OSes, I don't feel like getting the square 4-mics device myself. I mean, it is really Seeed Studio's staff's job to make it work for current OSes. However, if any of you 3 uses or plans to use the square 4-mics for work, please feel free to click the donate link in my FontVal repo.

HinTak, 2020-09-04:
@j1nx @younes-professor @Daenara Okay, in `seeed-voicecard.c` in the routine `seeed_voice_card_dai_link_of()`, there is a block like this(note the `codec`, rather than `cpu` earlier - there are two very similar blocks):

```
         #if LINUX_VERSION_CODE <= KERNEL_VERSION(4,10,0)
        ret = asoc_simple_card_parse_clk_codec(codec, dai_link, codec_dai);
        #else
       ret = asoc_simple_parse_clk_codec(dev, codec, dai_link, codec_dai);
       #endif 
```

if you add this line below:
```
       codec_dai->sysclk = 24000000;
```

It should give you back the `ac108_set_sysclk  :24000000` line. It does not make any difference to my 6-mics device (which has two ac108 as slaves to ac101), but probably would make your square 4-mic device (which has a single ac108) works better.

Give it a try?

HinTak, 2020-09-04:
`asoc_simple_parse_clk_codec()` has changed a lot, and before it too. That 2400000 value is supposed to result from `asoc_simple_parse_clk_codec()`, but didn't in 5.4 .

j1nx, 2020-09-04:
@HinTak, I will give it a shot this weekend. Strugling with the 2mic speaker at the moment as well, but that is not a driver issue, more a OS issue.

Will get back to you asap.

HinTak, 2020-09-04:
@younes-professor the recording volume is adjustable : use amixer to tune it up. I have learned it the hard way ( #240 ). See if that helps.

Great to hear that the outcome is okay. Now one just to find out how to read and get that `2400000` number in properly. There is just one value, it is read from in the dts file. Somehow between 4.19 and 5.4, the clk value of dts-parsing code has gotten into a pickle. (strange it has no effect on the 6-mics) . Great job @Daenara noticing it.

iovsiann, 2020-09-04:
@younes-professor would you be so kind to recommend alternative mic array vendors? I am having sound quality issues as well. Besides low mic sensitivity, EMI from Raspberry Pi feeds strongly into 4th mic channel in Respeaker 4-mic board. This feed-through is easy to see and hear in Audacity.

Daenara, 2020-09-04:
@HinTak that fixed it. Would be great if you could add it to your code so we don't have to manually add in the line each time we install the drivers. I am really happy to have my mic back in working order. Playing around with a voice assistant that can't listen to you is kinda hard.

HinTak, 2020-09-04:
@iovsiann the injector.net cards are very well-supported software-wise, far better than seeed studio's respeaker range... Before the whole virus thing, I worked for an audio-tech company and had experience with both that and the Seeed Studio ones, and some other non-pi multi-channel gears. So this is first-hand experience.

HinTak, 2020-09-04:
@Daenara okay, I'll put that in as a temporary work-around . It would do for now, except the 2-mics device, since everything else has a ac108 and it is not an adjustable value. The proper way, though, is for it to be read from the dts and passed along. I'll put what I know in the commit message, for a later revisit.

younes-professor, 2020-09-04:
> @younes-professor the recording volume is adjustable : use amixer to tune it up. I have learned it the hard way ( #240 ). See if that helps.
>
> Great to hear that the outcome is okay. Now one just to [...]

> @younes-professor would you be so kind to recommend alternative mic array vendors? I am having sound quality issues as well. Besides low mic sensitivity, EMI from Raspberry Pi feeds strongly into 4th mic channel in Respeaker 4-mic board. This feed-through is easy to see and hear in Audacity.

younes-professor, 2020-09-04:
> @younes-professor would you be so kind to recommend alternative mic array vendors? I am having sound quality issues as well. Besides low mic sensitivity, EMI from Raspberry Pi feeds strongly into 4th mic channel in Respeaker 4-mic board. This feed-through is easy to see and hear in Audacity.

@iovsiann ,to be honest ,i am in the same situation as you ,i have done some research and did not found many options besides this  https://www.matrix.one/products/voice ,it seems worth to be consider but from my experience with Seeed studio ,my concern is more about the software support than the quality of the hardware (i lost 3 precious days struggling with Seeed devices,and they not respond  )
i am a web and mobile developer ,i don't know to much about sound field and kernel programming ,so don't relay to much on my opinion ,consider the option given by @HinTak he knows better

younes-professor, 2020-09-04:
> @younes-professor the recording volume is adjustable : use amixer to tune it up. I have learned it the hard way ( #240 ). See if that helps.
>
> Great to hear that the outcome is okay. Now one just to [...]
[image]

by adjusting ADC1 PGA gain ,the outcome is much better but the other option has no effect ,i don't know what is 'ADC1 PGA gain '

HinTak, 2020-09-04:
I have put the two-line work-around in, https://github.com/HinTak/seeed-voicecard/commit/138d22226ed2b36a94d74873f8f1ac3193ee944f with a lengthy message about my current findings. It is two-line, as I added a check to apply it only for the ac10x-based devices.

@younes-professor what the value means are probably explained in the ac108 data sheet. (it is a pdf, and on github under SeeedDocuments somewhere)

I am okay if you guys want to buy me a coffee (there is a donate link in my Font-Validator repo). While it is fun learning new things - I was an x86/Intel kernel guy until two years ago, and audio is a new field for me too - I do wish that Seeed Studio staffer are more "present"...

HinTak, 2020-09-05:
> On Friday, 4 September 2020, 12:49:18 BST, Peter Steenbergen <notifications@github.com> wrote:

> @HinTak, I will give it a shot this weekend. Strugling with the 2mic speaker at the moment as well, but that is not a driver issue, more a OS issue. Will get back to you asap.

Argh, just in case it is relevant: another user in a different issue (the ubuntu aarch64 one) mentioned that the 2-mic driver seems to be sensitive to driver load order and/or co-operation with the pi's built-in head-phone jack, and the device can stop/start working across reboot. I suspect that this is a different kind of tricky issue - recent kernel has its own built-in driver for the wm8960 chip used on the 2-mic device. However, I only have the 6-mic device, and I cannot look at issues I don't have hardware of (besides the fact that I don't work for Seeed Studio!). The 4-mic is more similar to the 6-mics hardware-wise, but it is already stretching it a bit.

j1nx, 2020-09-05:
@HinTak Just to confirm. the quickfix does indeed work, however far from clean. Here the dmesg log of it;
https://gist.github.com/j1nx/6ec48b3547302528da36d7e8b5bd342e

Thx for the headsup about the 2mic. This might give the right pointers to figure out what monkey business is going on. Will also report back my finding there.

HinTak, 2020-09-05:
@j1nx That message happens quite early - what sound-related application you are running at boot time? I'll see if I can make it happen with the 6-mics.

j1nx, 2020-09-05:
@HinTak It is a minimalistic OS based on Buildroot, which only runs basically PulseAudio as systemwide service and Mycroft A.I. voice assistance.

I think you are after PulseAudio.

HinTak, 2020-09-05:
@j1nx not a big fan of pulseaudio (nor systemd from the same guy...) but it is for playback, and ac108 is the recording/capture chip so Mycroft.ai is the one.

I have no experience with it - in my last job before the pandemic, we were working with/for a few of its commercial/proprietary competitors. While that does not stop us from looking at it or even learn / get some idea from it, we didn't... Anyway, is there any quick and short intro /install instruction for it? I think there are some on seeed studio's; but as I said, we basically skip that whole thing as we had something else...

HinTak, 2020-09-06:
@iovsiann @younes-professor I looked up the support level of the matrix.one - their own engineer responded and addressed and close the equivalent issue in 12 days
 https://github.com/matrix-io/matrixio-kernel-modules/issues/48 .

younes-professor, 2020-09-06:
> @iovsiann @younes-professor I looked up the support level of the matrix.one - their own engineer responded and addressed and close the equivalent issue in 12 days
> [matrix-io/matrixio-kernel-modules#48](https://github.com/matrix-io/matrixio-kernel-modules/issues/48) .

Yes ,they seem more serious ,will try their hardware soon ,and give you my feedback

HinTak, 2020-09-08:
I have created a new repo https://github.com/HinTak/RaspberryPi-Dev , which at the moment is just tips for kernel driver development and troubleshooting. The call stack is in https://github.com/HinTak/seeed-voicecard/commit/138d22226ed2b36a94d74873f8f1ac3193ee944f , and if you have two identical setup, one with 4.19 and the other 5.4 , you can find out the difference between the two and fix what's wrong with 5.4, if only by inserting a lot of `pr_info()`. It is a bit tedious and slow, the way I am doing it - swapping two SD cards - so it will take a while, before I get to the bottom of the issue with recent kernels. But Seeed Studio staff should be able to afford two separate setup! 

@Daenara thanks for the donation :-)

HinTak, 2020-09-14:
After staring and hacking at the code some more, I decided that I have reached my limits and needed to seek "higher authority"... so I filed an issue with the Raspberry Pi Foundation, and specifically asked the guy who last wrote that part of documentation 4-years ago, what he meant. And I got my answer :) ! Anyway, the correct fix to clock issue is in https://github.com/HinTak/seeed-voicecard/commit/47d97e7bd570e3d9cb2187fd041d7bbfda9d14b0 , but you need the commit after it too - the fix is in the *.dtbo files which get copied to `/boot/overlays` , so be sure that's the case.

@j1nx if you still see the stack dump after updating to the latest from my branch, I need to see the actual `*.ko` files as built and installed on your OS, plus the stack dump itself to match them up.

So the official docs at the official pi web site is getting updated / expanded too - the pi 3 was only out two months in mid 2016, and a lot has changed since then - , and I have learned a few things on the way. I think the donate link in my FontVal repo asks for credit card details - this one might work better for pp to pp : https://paypal.me/HinTakLeung?locale.x=en_GB . Buy me a beer if any of you feel like it.

j1nx, 2020-09-15:
Haven't had much time to look at the 4mic stuff as of yet as I was still puzzling with the 2mic. Anyhow, can't figure out why the 2mic get's picked up normally, just not want to play or record anything. Will let it go for now. Ussually the DOH! moment comes when doing something else.

Will rebuild the driver and test the 4mic. Will let you know soon.

ps. Only had a few bucks left on PayPal, but they are yours. Will do another donation later when more money is available on paypal because I think you deserve it when you do somebody else his work ;)

HinTak, 2020-09-16:
@j1nx I hope the current code fixes the messy dmesg - besides the actual *.ko files, if you see `BUG: scheduling while atomic`, it would be good to know how long / and frequent) that goes on (I.e. The time stamps before it, and the first one and last one, and roughly how often) - and also what process is being run ie output of `ps -elf` or `ps -alx`. 

The message is as it says: the system is too busy / a part of the driver code is taking unexpectedly long to run. "scheduling while atomic" means the kernel is pausing the driver code because it is taking too long, where the code itself says "this part is real quick and also should be done in one go, so don't pause/interrupt this". The actual spinlock is actually quite long, so it could be just poorly written code :-).

j1nx, 2020-09-25:
@HinTak  Managed to build new driver with your latest code. A quick spin up of the OS showed the same atomic stuff though.

Will see if I can dig into it this weekend.

HinTak, 2020-09-25:
@j1nx yes, somehow I am seeing it too on 64-bit ubuntu - and it is triggered by ssh'ing into the box itself. (basically just loading it up). I wonder if that's the reason seeed studio recommend uninstalling pulseaudio.

j1nx, 2020-09-25:
That is strange! Anyhow, seeed recomments that type of stuff because they basically don't know what they are doing. (Hence otherwise you would not have to fix their stuff).

Same with downgrading the kernel version ???

If it is PusleAudio that might be triggering it, I guess it has to do with the udev detect module of PulseAudio, which is actually one of the reasons I use Pulse for MycroftOS. Will see if I diasable the udev detect and just mannually use the ALSA device as sinks/source, thing will get better or not.

Pillar1989, 2020-09-25:
https://github.com/respeaker/seeed-voicecard/issues/249

closed 2020-09-25 by Pillar1989

HinTak, 2020-09-25:
@j1nx the "scheduling while atomic" message is triggered by multiple parties simultaneously calling the `*_startup` routine, and is accompanied by an I2S error message half of the time. As I am running a version with some additional debug message's, I can see that, quite often startup haven't reached the end before being called. I.e. I am seeing a pattern of :

- startup begin

- startup begin

- startup end

- startup end

Instead of the correct pattern of begin/end/begin/end . Some of the internal states need to be protected against re-entrance calling like this - the kernel multi-tasks and can do so and often does (e.g. playing music while adjusting volume, etc),  it is up to the driver to protect its guts. Using a spinlock is quite wrong.

j1nx, 2020-09-25:
Anything i can / you want me to test? Like I said, running the drivers within a very minimalistic OS based on Buildroot.

HinTak, 2020-09-25:
@j1nx thanks for the offer - I am not sure there is any easy fix. There is a whole chapter on mutex / spinlock and synchronicity issue in the linux driver development book, if you feel like reading on it. The book is (or was) available in dead-tree form, but is also a free digital download. It is quite well written, but I never got round to read it properly, just reading sections of it from time to time when I need to.

HinTak, 2020-10-31:
@iovsiann @younes-professor @j1nx @Daenara Filed first code-improvement pull for the matrixio card https://github.com/matrix-io/matrixio-kernel-modules/pull/50 , and am interested to see how soon and in what way they will respond to https://github.com/matrix-io/matrixio-kernel-modules/issues/51 https://github.com/matrix-io/matrixio-kernel-modules/issues/52 too. Any of you brought that card (I haven't, just want to have a feel about the driver code and their response first), and/or checking out other mic-arrray devices? I am still having trouble with the respeaker with "BUG: scheduling while atomic" #251, and the issue seems to be worse from having debug statements in the driver code (which slows it down, and therefore more likely to be affected by context switches and contentions).

Daenara, 2020-11-01:
@HinTak I don't have one of those mics but I do hang out in voice assistant forum and there the matrix stuff is used quite often. I have not read anything about any driver problems there, while I read quite a bit about respeaker and ps3eye driver issues, so there seem to be fewer/no driver issues there.