respeaker/seeed-voicecard #247: ReSpeaker 4-Mics Linear Array Kit volume control with red cross in top panel (bar)

lxne, 2020-09-08:
On my Raspberry Pi the volume control symbol in the top panel only shows a red cross if I select the seeed voicecard. Why can't I control the main volume off my 4-Mics Linear Array ReSpeaker with volume control as usual and expected?

[image]
[image]

Pillar1989, 2020-09-25:
https://github.com/respeaker/seeed-voicecard/issues/249

closed 2020-09-25 by Pillar1989

turmary, 2020-09-28:
Solved in commit https://github.com/respeaker/seeed-voicecard/commit/e1110617b2dcb06c0191234b5b643187db2d58e4

lxne, 2020-10-06:
Thanks. It works for me now.