respeaker/seeed-voicecard #248: Ubuntu / Aarch64 / RaspOS / v5.x kernel does not work

HinTak, 2020-09-09:
The other reporter ( https://github.com/respeaker/seeed-voicecard/issues/211 )  gave up eventually. I think it is useful to keep one such item open,  if only to tell other users that the Seeed Studio supplied driver does not work on any current releases of Ubuntu nor Raspbian / RaspOS.

HinTak, 2020-09-09:
Yes I maintain my own "substantial enhancement" fork (  https://github.com/HinTak/seeed-voicecard ) but it is not my job to do so, and I should not have needed to spend time on it.

joshuajaharwood, 2020-09-21:
@HinTak Just wanted to say thanks for maintaining your driver fork, I'm using it to power my Mycroft installation on kernel 5.4. It's disappointing that you've even had to do so in the first place, though :(

twitchyliquid64, 2020-09-21:
Ditto, thank you @HinTak for your work. Just wondering if theres a specific raspbian version youve tested it on that works? I haven't had any luck so far.

HinTak, 2020-09-21:
@twitchyliquid64 what problems are you encountering? You need to use my fork for any of the current releases, not the code here.

I am swapping between current 64-bit ubuntu (20.04.1, released around august) and 6-month old raspbian lite (Feb or March release) on two SD cards. Both are up-to-date userland wise. The ubuntu one uses a heavily debug-enhanced variant of my fork against kernel 5.4.x, while the raspbian one use a heavily debug-enhanced variant of the one here against 4.19. Quite a few others have reported sucess use of my fork against current raspbian. 

Raspbian has been 5.4 based since about June /July. The problem with the code here is that Seeed Studio staff wants you to downgrade your kernel to something quite old, and raspbian no longer keep those old versions. i.e. If you had a 6-month-old raspbian, and you set it up to work **6-months ago** fully, and you configure your system **not to update specific parts **, it continues to work. But you cannot do it **now** with either current or older raspbian.

There is no **fresh** installation **now**, of any past releases or current OS releases that works with Seeed Studio's code.

twitchyliquid64, 2020-09-21:
I've been trying to get your fork to work. On the latest (sept) 64bit Raspbian image there don't appear to be kernel headers, so that was problem 1. I eventually got kernel headers using `rpi-source`. Then I started getting build errors.

I'll reflash my Pi and try again today and get you the exact errors.

twitchyliquid64, 2020-09-21:
Commands run on a Pi 4 flashed with `2020-05-27-raspios-buster-arm64.img`.
```
pi@respeaker:~/seeed-voicecard $ history
    1  sudo apt-get install git
    2  htop
    3  git clone https://github.com/HinTak/seeed-voicecard
    4  cd seeed-voicecard/
    5  sudo ./install.sh 
    6  history
```

Error during `./install.sh`:

```
Error! Your kernel headers for kernel 5.4.42-v8+ cannot be found.
Please install the linux-headers-5.4.42-v8+ package,
or use the --kernelsourcedir option to tell DKMS where it's located
```

I will install the headers and let you know what the error changes to.

twitchyliquid64, 2020-09-21:
I also tried the `v5.7` and `v5.8` branches, but they had substantially more error output, mostly like this:

```
/var/lib/dkms/seeed-voicecard/0.3/build/ac108.c:656:9: error: ‘struct snd_soc_dai’ has no member named ‘stream_active’; did you mean ‘capture_active’?
    dai->stream_active[SNDRV_PCM_STREAM_PLAYBACK], dai->stream_active[SNDRV_PCM_STREAM_CAPTURE]);
         ^~~~~~~~~~~~~
./include/linux/device.h:1759:47: note: in definition of macro ‘dev_dbg’
   dev_printk(KERN_DEBUG, dev, dev_fmt(fmt), ##__VA_ARGS__); \
                                               ^~~~~~~~~~~
```

HinTak, 2020-09-21:
`rpi-source` is a completely different thing. The v5.7/v5.8 branch are future/anticipation branch for when those kernel versions become current.

I'll see what I can find about raspbian kernels and their headers.

HinTak, 2020-09-21:
You should get the 0819 kernel which is 5.4.51-based : https://github.com/HinTak/RaspberryPi-Dev/blob/master/Raspbian-Kernel-Releases.md

HinTak, 2020-09-21:
@twitchyliquid64 interesting - raspbian gives you a 64-bit kernel package but does not give you the accompanying 64-bit kernel header package. (ubuntu does). I'll file an issue with the raspbian people.

HinTak, 2020-09-21:
@twitchyliquid64 filed upstream - https://github.com/raspberrypi/linux/issues/3865 - meanwhile, use current 32-bit raspbian, or current 64-bit ubuntu :-).

HinTak, 2020-09-21:
@twitchyliquid64 I remember something - see https://github.com/respeaker/seeed-voicecard/issues/191 - the raspbian 64-bit headers are in the (32-bit) armhf header packages . It probably will work if you just download the version-matching header armhf deb , but YMMV

Pillar1989, 2020-09-25:
https://github.com/respeaker/seeed-voicecard/issues/249

closed 2020-09-25 by Pillar1989