respeaker/seeed-voicecard #249: The latest kernel will be supported by MID-October at the latest!

Pillar1989, 2020-09-25:
Thank you very much for your support for Seeed products. We've been looking at the ReSpeaker product line, and we're working on the driver update. The kernel will maintain its drivers throughout the product life cycle.

Pillar1989, 2020-09-25:
Thanks a lot to @HinTak  for supporting ReSpeaker and let's put all issues together. We'll solve your problems together in the next update.

HinTak, 2020-09-25:
The easiest for Seeed Studio would be to just pull my v5.5 branch in, and fix what remaining problem there is. (the "scheduling while atomic" issue - I think the spinlock should likely be a mutex).

The donate link in my FontVal repo asks for credit card details - this one might work better for pp to pp : https://paypal.me/HinTakLeung?locale.x=en_GB for donating / sponsoring my work.

HinTak, 2020-09-25:
Here is the other donate link - https://sourceforge.net/p/hp-pxl-jetready/donate/ - both uses PayPal, but one is pp to pp and the other accepts credit cards.

HinTak, 2020-09-25:
https://github.com/respeaker/seeed-voicecard/issues/247 is Afaik a different issue: the driver does not provide a master volume control, so alsa does not find one.

https://github.com/respeaker/seeed-voicecard/issues/213 is yet another issue: the current respeaker driver code cannot be accepted upstream, because the linux kernel has always have a policy of accepting contributions from people using their real / legal names, with valid email addresses. Having fake copyright messages is a straight-forward case of rejection.

Pillar1989, 2020-09-27:
@HinTak  Thank you for collecting questions. What's your E-mail address? We consider other better ways to thank you.

turmary, 2020-09-27:
Hi HinTak,
> 
> 
> The easiest for Seeed Studio would be to just pull my v5.5 branch in, and fix what remaining problem there is. (the "scheduling while atomic" issue - I think the spinlock should likely be a mutex).
> 
I don't think the spinlock should be a mutex, cause XXX_trigger sometimes be in soft-interrupt environment.
mutex will blocked then the flowing will go to a context switch.

HinTak, 2020-09-27:
See my comment regarding #213 - my listed email in the linux kernel tree ( in the MAINTAINERS file), the same one associated with my github account, is genuine and correct.

#213 is yet another issue: the current respeaker driver code cannot be accepted upstream, because the linux kernel has always have a policy of accepting contributions from people using their real / legal names, with valid email addresses. Having fake copyright messages is a straight-forward case of rejection.

HinTak, 2020-09-27:
The spinlock takes way too long - it needs to be something else.

turmary, 2020-09-28:
> See my comment regarding #213 - my listed email in the linux kernel tree ( in the MAINTAINERS file), the same one associated with my github account, is genuine and correct.
>
> #213 is yet another [...]

I konw there are too much things about the codec driver which could not accept by linux community.
We haven't a plan to commit them to upstream in the near future.

> 
> 
> The spinlock takes way too long - it needs to be something else.

The codec need to start codec sysclk by accessing I2C in XXX_trigger, caused by some limitation about hardware.
see https://github.com/respeaker/seeed-voicecard/blob/master/ac101.c#L44-L52

turmary, 2020-09-30:
Hi all,
all those changes merge to branch master from rel-v5.5.

lxne, 2020-09-30:
> Hi all,
> all those changes merge to branch master from rel-v5.5.

So you are saying the master now should work with recent raspbian kernel?

HinTak, 2020-09-30:
@lxne yes, master is now identical to rel-v5.5 . It is functionally identical to my v5.5 branch though, which means it still needs work on the "scheduling while atomic" issue. There is also an upstream issue (I filed https://github.com/raspberrypi/linux/issues/3865 9 days ago) that current 64-bit raspbian does not work. If you want 64-bit, you need to use ubuntu or something non-raspbian.

HinTak, 2020-09-30:
@Pillar1989 -

> @HinTak Thank you for collecting questions. What's your E-mail address? We consider other better ways to thank you.

My email is as in the commit messages - at least two projects I work on - the linux kernel and wine - has a policy of only accepting changes from people with real full legal names and proper email addresses. My email is the same onelisted in the linux kernel source tree's MAINTAINERS file, under rtl8187 (wifi device) which I co-wrote.

HinTak, 2020-10-05:
@Pillar1989 @turmary btw, if you have any documentation or original code samples about the ac101/ac108 codecs and / or materials from allwinner's , that you cannot post publicly, please feel free to send to the same sourceforge email address I use. I suspect some of them might be in Chinese only - I do read /write Chinese natively, if that helps, although I do not usually read, and even less often write, in Chinese for technical matters. (and actual financial commission / sponsorship is useful too...)

turmary, 2020-10-09:
@HinTak we havn't permission to share resoure to you.
But you could download ac108/ac101 datasheet directly from http://www.x-powers.com/en.php/Info/support/article_id/40 without login.
You need logon permission to get other materials.

closed 2020-10-14 by Pillar1989

HinTak, 2020-10-14:
The spinlock issue has not been tackled...

Tom-Lu, 2021-01-21:
Ubuntu still not supported, waste few days got no solution.

```
ubuntu@ubuntu:~$ uname -a
Linux ubuntu 5.4.0-1026-raspi #29-Ubuntu SMP PREEMPT Mon Dec 14 17:01:16 UTC 2020 aarch64 aarch64 aarch64 GNU/Linux
ubuntu@ubuntu:~$ arecord -Dac108 -f S32_LE -r 16000 -d 30 -c 4 hello.wav
Recording WAVE 'hello.wav' : Signed 32 bit Little Endian, Rate 16000 Hz, Channels 4

Message from syslogd@ubuntu at Jan 21 07:39:32 ...
 kernel:[16101.044920] Internal error: Oops: 82000007 [#1] PREEMPT SMP

Message from syslogd@ubuntu at Jan 21 07:39:32 ...
 kernel:[16101.243182] Kernel panic - not syncing: Aiee, killing interrupt handler!
```

HinTak, 2021-01-21:
@Tom-Lu that's issue https://github.com/respeaker/seeed-voicecard/issues/251 .