respeaker/seeed-voicecard #250: Consider using `uname -m` instead

HinTak, 2020-09-28:
https://github.com/respeaker/seeed-voicecard/blob/30ceb546fa1e6d9f3814599f2d0208253e223d85/install.sh#L65

`uname -m` is a much faster, and more distribution-neutral way of doing the same thing as `dpkg --print-architecture`

turmary, 2020-09-29:
$(dpkg --print-architecture)  will return the architecture( or package name postfix ) directly.
But `uname -m` return kernel name, need a transition to architecture.
```
pi@raspberrypi:~ $ uname -m
armv7l
pi@raspberrypi:~ $ dpkg --print-architecture
armhf
```

HinTak, 2020-09-29:
It is more specific (tell u it is a pi 2 or above, plus 32-bit) instead of j just 32-bit. And faster, and works on non-debian/ubuntu systems.

closed 2020-10-14 by Pillar1989