respeaker/seeed-voicecard #252: 4 mic array brands

dinuchiriac, 2020-10-14:
6 Mic array have mic model specified in the specification which is `MSM321A3729H9CP ` but there s nothing about 4 mic array.
Can anyone help me to find the mic brand, model and specs of the 4 mic array microphones?

HinTak, 2020-10-14:
There are at least two 4-mic arrays - one linear and the other square. Which one are you referring to?

- I am just another user.

dinuchiriac, 2020-10-14:
I'm referring to the [square one.]( https://wiki.seeedstudio.com/ReSpeaker_4_Mic_Array_for_Raspberry_Pi/). But in fact, it would be great to have this info for all of the mic arrays.

HinTak, 2020-10-14:
@dinuchiriac came across this a while ago, I think it contains all the info you are after: https://forum.seeedstudio.com/t/microphone-model-on-respeaker/7356/6

... Still just another user

dinuchiriac, 2020-10-14:
Thanks a lot, @HinTak ,
It would be great is anyone from seedstudio could confirm this.

dinuchiriac, 2020-10-14:
Ahh, it looks like @iacheng-lu is actually from seed studio. Thanks again.

closed 2020-10-14 by dinuchiriac

HinTak, 2020-10-14:
Yes, rather different culture on their forum vs github, and also the fact that they are based in China, where github access could be intermittent.
Staffs are more present on the forum and few if any 3rd-party offering answers, while on github, their staff could be absent for months leaving a lot of angry users exchanging tips....