respeaker/seeed-voicecard #253: Output stuttering on 6-mic 3.5mm jack

joshuajaharwood, 2020-10-19:
I'm using a 6-Mic board's onboard 3.5mm jack on a Raspberry Pi 3B+ (kernel 5.4.51) to output to a speaker. I've been using this with librespot's ALSA backend and pulse as ALSA's default sink. 

When outputting to the speaker (using alsa_output.platform-soc_sound.seeed-2ch), the audio takes a long time (about 15 seconds) to start, and regularly stutters for long periods (2-3 seconds). Using a pulse mixer shows that the audio is in fact playing in realtime without stutters, but cannot be heard. 

Switching to the Raspberry Pi's onboard audio (alsa_output.platform-bcm2835_audio.analog-mono) however **displays no such issue**.

I am using the default 6-mic ALSA config as installed by the seeed-voicecard driver, except with default PCM set to 'pulse' in /etc/asound.conf.

Please let me know if you have any insight.

Thank you!

HinTak, 2020-10-19:
Are you seeing any spinlock related error messages in `dmesg` ( see https://github.com/respeaker/seeed-voicecard/issues/251 )? 

-- am just another user

joshuajaharwood, 2020-10-19:
@HinTak Nothing that resembles what's in #251, only the following stand out:
- `bcm2835-i2s 3f203000.i2s: I2S SYNC error!`
- `bcm2835_audio bcm2835_audio: failed to close VCHI service connection (status=1)`. 

I'll include a [dmesg dump](https://pastebin.com/nBzmbNLu) in case there's something I missed.

HinTak, 2020-10-20:
Yes, those two lines need looking at...

HinTak, 2020-10-20:
I filled an issue upstream on the same hardware a while ago, and was told the respeaker driver is basically buggy : 
https://github.com/raspberrypi/linux/issues/3580

HinTak, 2020-11-14:
@joshuajaharwood thanks for the donation! Regarding the stuttering, one direction of approach could be looking at what other heavy-duty processes are running on your system and try to minimize their impact. Have you looked at `ps -lax` and see what processes are busy or taking up cpu time, after the stuttering has been going for a while? If it is not too confidential, post the output of `ps -lax` After the system has been stuttering for a few minutes.

closed 2020-11-17 by Pillar1989