respeaker/seeed-voicecard #255: does not install on latest raspberry pi OS with kernel 5.4

DjDemonD, 2020-10-24:
Unable to install the 2mics hat on zeroW with raspberry pi OS (Aug 2020) with kernel 5.4. Tried older raspian verson (feb 2020) with older 4.x kernel, but the install.sh script upgrades the kernel before installation. Tried modding the script but it didn't install the capture device, only the playback devices.

HinTak, 2020-10-24:
You may find this works slightly better... https://github.com/HinTak/seeed-voicecard

DjDemonD, 2020-10-24:
Thanks, I tried it on my previously installed version of raspbian (feb 2020), it installed the playback device but not the recording device. I will try on fresh install.

DjDemonD, 2020-10-24:
That works brilliantly on the latest version thank you that's amazing. How can I show my appreciation?

closed 2020-10-24 by DjDemonD

HinTak, 2020-10-24:
@DjDemonD there are donation links via PayPal at the bottom of https://hintak.github.io/ - buy me a beer if you feel like it :-) .

HinTak, 2020-10-24:
I saw your post on the respeaker forum (I think it is the same person) and the log there. I think there is a bug in the seeed studio upstream driver . We disagree on basically one thing: I don't care much about older kernels, but whatever is current in Debian/ Ubuntu must work. They want users to stay with a specific older kernel (which is not feasible IMHO, since neither Debian nor Ubuntu let you downgrade that way) - and in doing so they made a mistake of hard-coding to only supporting arm7 - which is 32-bit pi 3 or pi 4 (and maybe pi 2) but zeroW is arm6.

DjDemonD, 2020-10-24:
Yeah that was me. I've posted to say I found the solution. Thanks. Sound card works. Still working now on alexapi but having a sound card that definitely works is a big step in the right direction.

HinTak, 2020-10-24:
@DjDemonD thanks for the donation :-) . The v5.7/5.8 branches on my repo are "future" branches when those becomes mainstream . I think ubuntu 10.10 released a few days ago is based on v5.8