respeaker/seeed-voicecard #256: --compat-kernel

AlanNandy, 2020-10-28:
I use "sudo ./install.sh --compat-kernel" but it still say "Your kernel headers for kernel 5.4.51-v8+ cannot be found at
/lib/modules/5.4.51-v8+/build or /lib/modules/5.4.51-v8+/source.
Can't compile with this kernel, aborting"

HinTak, 2020-10-28:
I am quite sure that you find this works slightly better... https://github.com/HinTak/seeed-voicecard , and just forget the `--compat-kernel` flag in this case. If it works for you, please feel free to make a donation at the links at the bottom of https://hintak.github.io .

HinTak, 2020-10-28:
It is an oversight of the seeed studio folks - that's a bug in theirs on 64-bit kernel support.

closed 2020-11-17 by Pillar1989

WilliamVJacob, 2021-02-19:
I have added the fix to install Respeaker Driver for 64 bit Raspbian OS
Check my forked  seeed-voicecard  repository  for solution
https://github.com/WilliamVJacob/seeed-voicecad