respeaker/seeed-voicecard #257: use an oled display connected via i2c on respeaker 4

Amhehu, 2020-11-02:
Hi!
I try to display some text on an oled display which is connected by a grove cable to my respeaker 4.
I use a raspi 4 with Raspberry OS (buster). i2c is activated in raspi-config.
Directly connected to the raspi the display is detected correctly (address 3c) and shows text.
When I plug in my respeaker i2cdetect find a device with address 3b. After installing the driver (https://wiki.seeedstudio.com/ReSpeaker_4_Mic_Array_for_Raspberry_Pi/) i2cdetect say UU at this address.
I can use the GPIO port on the board, but I don't know how i can use an i2c device.
Can somebody help me?

closed 2020-11-11 by Amhehu

jmrf, 2022-01-27:
@Amhehu I am in a similar situation as yours. Same Raspbian, same Respeaker, same UU at the 3b address.

Did you manage to solved?

HinTak, 2022-01-27:
Actually I like to know too - I bought a few i2c/grove devices and leds to play with before Christmas.

Amhehu, 2022-01-27:
I bought a grove oled display. With that it was no problem.
https://wiki.seeedstudio.com/Grove-OLED_Display_0.96inch/

Here a picture of it
https://paste.pics/1d7d4cbda12d20f85fb0bcc894e05d69