respeaker/seeed-voicecard #258: Raspberry Pi OS 64-bit - Issue with installing on 5.4.72

Arkert, 2020-11-04:
Current Raspberry Pi OS 64-Bit, Linux version 5.4.72-v8+. If I want to install, I get following error message:

`make -j4 KERNELRELEASE=5.4.72-v8+ -C /lib/modules/5.4.72-v8+/build M=/var/lib/dkms/seeed-voicecard/0.3/build.....(bad exit status: 2)
Error! Bad return status for module build on kernel: 5.4.72-v8+ (aarch64)`

And in the logs:

`ERROR: "_mcount" [/var/lib/dkms/seeed-voicecard/0.3/build/snd-soc-seeed-voicecard.ko] undefined!
ERROR: "_mcount" [/var/lib/dkms/seeed-voicecard/0.3/build/snd-soc-ac108.ko] undefined!
ERROR: "cpu_hwcap_keys" [/var/lib/dkms/seeed-voicecard/0.3/build/snd-soc-ac108.ko] undefined!
ERROR: "arm64_const_caps_ready" [/var/lib/dkms/seeed-voicecard/0.3/build/snd-soc-ac108.ko] undefined!
ERROR: "_mcount" [/var/lib/dkms/seeed-voicecard/0.3/build/snd-soc-wm8960.ko] undefined!`

Do I have any chance to install the drivers?

Thanks

HinTak, 2020-11-04:
Same as https://github.com/respeaker/seeed-voicecard/issues/256 - stop using `--compat-kernel` , etc.

Arkert, 2020-11-04:
I'm not using --compat-kernel.

Fresh installation. 
Updating everything to current version. 
clone your (HinTak) repository. 
Let default branch active (v5.5)
sudo ./install.h

Then I get the error message I quoted above.

HinTak, 2020-11-04:
Download the matching kernel-header-*.arm64.deb and install that (`sudo apt-get install ...` ) : http://archive.raspberrypi.org/debian/pool/main/r/raspberrypi-firmware/ ?

Ubuntu 64-bit does work as is...

Arkert, 2020-11-04:
> Download the matching kernel-header-*.arm64.deb and install that (`sudo apt-get install ...` ) : http://archive.raspberrypi.org/debian/pool/main/r/raspberrypi-firmware/ ?

Thanks! I will test it. 

> Ubuntu 64-bit does work as is...

With Ubuntu 64-bit I had crashes (kernel panic) after driver installation. Dunno why. Now I'm testing raspberry pi os

HinTak, 2020-11-04:
Argh, the crash / kernel panic would likely be https://github.com/respeaker/seeed-voicecard/issues/251 .

Arkert, 2020-11-05:
> > Download the matching kernel-header-*.arm64.deb and install that (`sudo apt-get install ...` ) : http://archive.raspberrypi.org/debian/pool/main/r/raspberrypi-firmware/ ?
> 
> Thanks! I will test it.

Still same error:

```
make -j4 KERNELRELEASE=5.4.72-v8+ -C /lib/modules/5.4.72-v8+/build M=/var/lib/dkms/seeed-voicecard/0.3/build.....(bad exit status: 2)
Error! Bad return status for module build on kernel: 5.4.72-v8+ (aarch64)
```

make.log:

> DKMS make.log for seeed-voicecard-0.3 for kernel 5.4.72-v8+ (aarch64)
> make: Entering directory '/usr/src/linux-headers-5.4.72-v8+'
>   AR      /var/lib/dkms/seeed-voicecard/0.3/build/built-in.a
>   CC [M]  /var/lib/dkms/seeed-voicecard/0.3/build/wm8960.o
>   CC [M]  /var/lib/dkms/seeed-voicecard/0.3/build/ac108.o
>   CC [M]  /var/lib/dkms/seeed-voicecard/0.3/build/ac101.o
>   CC [M]  /var/lib/dkms/seeed-voicecard/0.3/build/seeed-voicecard.o
>   LD [M]  /var/lib/dkms/seeed-voicecard/0.3/build/snd-soc-wm8960.o
>   LD [M]  /var/lib/dkms/seeed-voicecard/0.3/build/snd-soc-seeed-voicecard.o
>   LD [M]  /var/lib/dkms/seeed-voicecard/0.3/build/snd-soc-ac108.o
>   Building modules, stage 2.
>   MODPOST 3 modules
> ERROR: "__stack_chk_guard" [/var/lib/dkms/seeed-voicecard/0.3/build/snd-soc-seeed-voicecard.ko] undefined!
> ERROR: "__stack_chk_guard" [/var/lib/dkms/seeed-voicecard/0.3/build/snd-soc-ac108.ko] undefined!
> make[1]: *** [scripts/Makefile.modpost:94: __modpost] Error 1
> make: *** [Makefile:1645: modules] Error 2
> make: Leaving directory '/usr/src/linux-headers-5.4.72-v8+'

I'm clueless right now. Can't get it running on any 64 bit operating system. On ubuntu I get kernel panic and on Raspberry Pi OS 64 bit I get this.

xiaovsma, 2020-11-05:
> Still same error:
>
> ```
> make -j4 KERNELRELEASE=5.4.72-v8+ -C /lib/modules/5.4.72-v8+/build M=/var/lib/dkms/seeed-voicecard/0.3/build.....(bad exit status: 2)
> Error! Bad return status for module build [...]

我也有这个相同问题(I have this same problem)

xiaovsma, 2020-11-05:
> 我也有这个相同问题(I have this same problem)

这个git项目(This git project): https://github.com/HinTak/seeed-voicecard/tree/v5.5
An error occurred when executing the command: sudo ./install.h

xevilstar, 2020-11-17:
cat /var/lib/dkms/seeed-voicecard/0.3/build/make.log 
DKMS make.log for seeed-voicecard-0.3 for kernel 5.4.72-v8+ (aarch64)
Tue 17 Nov 2020 06:30:56 PM CET
make: Entering directory '/usr/src/linux-headers-5.4.72-v8+'
  AR      /var/lib/dkms/seeed-voicecard/0.3/build/built-in.a
  CC [M]  /var/lib/dkms/seeed-voicecard/0.3/build/ac108.o
  CC [M]  /var/lib/dkms/seeed-voicecard/0.3/build/wm8960.o
  CC [M]  /var/lib/dkms/seeed-voicecard/0.3/build/ac101.o
  CC [M]  /var/lib/dkms/seeed-voicecard/0.3/build/seeed-voicecard.o
  LD [M]  /var/lib/dkms/seeed-voicecard/0.3/build/snd-soc-wm8960.o
  LD [M]  /var/lib/dkms/seeed-voicecard/0.3/build/snd-soc-seeed-voicecard.o
  LD [M]  /var/lib/dkms/seeed-voicecard/0.3/build/snd-soc-ac108.o
  Building modules, stage 2.
  MODPOST 3 modules
ERROR: "__stack_chk_guard" [/var/lib/dkms/seeed-voicecard/0.3/build/snd-soc-seeed-voicecard.ko] undefined!
ERROR: "__stack_chk_guard" [/var/lib/dkms/seeed-voicecard/0.3/build/snd-soc-ac108.ko] undefined!
make[1]: *** [scripts/Makefile.modpost:94: __modpost] Error 1
make: *** [Makefile:1645: modules] Error 2
make: Leaving directory '/usr/src/linux-headers-5.4.72-v8+'

HinTak, 2020-11-17:
That's interesting - `__stack_chk_guard` sounds like it is an internal / compiler-built-in (`__*` are, a convention) routinea not supposed to be seen. Anything unusual about your build environment?

HinTak, 2020-11-17:
@xevilstar @xiaovsma @Arkert could you try adding this like to `Makefile`,  before the `if DEBUG` line?

```
ccflags-y += --no_protect_stack
```

It is not the ideal answer but should get you a working driver.

HinTak, 2020-11-17:
Upstream is aware of the problem with__stack_chk_guard 
. See https://github.com/raspberrypi/Raspberry-Pi-OS-64bit/issues/4#issuecomment-717390798

HinTak, 2020-11-17:
The upstream bug is https://github.com/raspberrypi/firmware/issues/1364

WilliamVJacob, 2021-02-19:
I have added the fix to install Respeaker Driver for 64 bit Raspbian OS
Check my forked  seeed-voicecard  repository  for solution
https://github.com/WilliamVJacob/seeed-voicecad

Pittermaennchen, 2021-03-20:
@WilliamVJacob I tried that, but still I get

> DKMS: add completed.
> Error! echo
> Your kernel headers for kernel 5.4.72-v8+ cannot be found at
> /lib/modules/5.4.72-v8+/build or /lib/modules/5.4.72-v8+/source.
> Can't compile with this kernel, aborting
> Please try to compile with the option --compat-kernel

when running sudo ./install.sh --compat-kernel

AIWintermuteAI, 2021-08-20:
Hi, @Arkert !
I made some changes to install.sh script and tested them on 5.10.52-v8+ Raspberry Pi OS on Raspberry Pi 4 1 Gb with 2-mic reSpeaker. After some consideration, I decided to split installation script in 32-bit version and 64-bit version - it is indicted in README now. So, try 
```
sudo ./install_arm64.sh
```
and see if it works for you.

HinTak, 2021-08-20:
@AIWintermuteAI FWIW, I'd say please don't make another nearly identical copy of the script. You end up having to make nearly identical changes twice in both scripts later, etc.

AIWintermuteAI, 2021-08-20:
Of course, the changes now are very small - but in future they might diverge more. There is no immediate problem in having two scripts now. I'll reconsider this at later date.

HinTak, 2021-08-20:
@AIWintermuteAI there is a useful option `uname -m` to ask the current kernel if it is 32-bit or 64-bit. This should be sufficient to merge any difference you need between 32-bit and 64-bit into a few if/then's.

AIWintermuteAI, 2021-09-02:
Installation issue has been fixed. Installation script might change in the future, pay attention to instruction in the README.
Closing the issue.

closed 2021-09-02 by AIWintermuteAI