respeaker/seeed-voicecard #259: Raspberry Pi OS 64-bit - Issue with installing on 5.4.72-v8+

xiaovsma, 2020-11-05:
Hello 

OS environment information, Two git projects, tried it but not:

1) The drivers of 2 Mic Array Kit for Raspberry Pi.

2) OS info
pi@raspberrypi:~ $ uname -a
Linux raspberrypi 5.4.72-v8+ #1356 SMP PREEMPT Thu Oct 22 13:58:52 BST 2020 aarch64 GNU/Linux
pi@raspberrypi:~ $ cd seeed/hinTak-seeed-voicecard/
pi@raspberrypi:~/seeed/hinTak-seeed-voicecard $ git branch -a
* v5.5
  remotes/origin/HEAD -> origin/v5.5
  remotes/origin/master
  remotes/origin/remove-forced-kernel-downgrade
  remotes/origin/v5.5
  remotes/origin/v5.7
  remotes/origin/v5.8
pi@raspberrypi:~/seeed/hinTak-seeed-voicecard $ cd ../seeed-voicecard/; git branch -a
* aarch64
  remotes/origin/HEAD -> origin/master
  remotes/origin/aarch64
  remotes/origin/artikpi
  remotes/origin/dev-bbg
  remotes/origin/linux-4.19-or-less
  remotes/origin/master
  remotes/origin/rel-v5.5
  remotes/origin/wm8960_debug
pi@raspberrypi:~/seeed/seeed-voicecard $ 

3) OS install drivers Kernel errs

DKMS: add completed.

Kernel preparation unnecessary for this kernel.  Skipping...

Building module:
cleaning build area...
make -j4 KERNELRELEASE=5.4.72+ -C /lib/modules/5.4.72+/build M=/var/lib/dkms/seeed-voicecard/0.3/build...(bad exit status: 2)
Error! Bad return status for module build on kernel: 5.4.72+ (aarch64)
Consult /var/lib/dkms/seeed-voicecard/0.3/build/make.log for more information.

Kernel preparation unnecessary for this kernel.  Skipping...

---------------------------------------------------------------------------

pi@raspberrypi:~ $ cat /var/lib/dkms/seeed-voicecard/0.3/build/make.log
DKMS make.log for seeed-voicecard-0.3 for kernel 5.4.72-v8+ (aarch64)
2020年 11月 05日 星期四 14:49:18 CST
make: 进入目录“/usr/src/linux-headers-5.4.72-v8+”
  AR      /var/lib/dkms/seeed-voicecard/0.3/build/built-in.a
  CC [M]  /var/lib/dkms/seeed-voicecard/0.3/build/wm8960.o
  CC [M]  /var/lib/dkms/seeed-voicecard/0.3/build/ac108.o
  CC [M]  /var/lib/dkms/seeed-voicecard/0.3/build/ac101.o
  CC [M]  /var/lib/dkms/seeed-voicecard/0.3/build/seeed-voicecard.o
/var/lib/dkms/seeed-voicecard/0.3/build/seeed-voicecard.c: In function ‘seeed_voice_card_dai_init’:
/var/lib/dkms/seeed-voicecard/0.3/build/seeed-voicecard.c:262:8: error: implicit declaration of function ‘asoc_simple_card_init_dai’; did you mean ‘asoc_simple_dai_init’? [-Werror=implicit-function-declaration]
  ret = asoc_simple_card_init_dai(codec, &dai_props->codec_dai);
        ^~~~~~~~~~~~~~~~~~~~~~~~~
        asoc_simple_dai_init
/var/lib/dkms/seeed-voicecard/0.3/build/wm8960.c: In function ‘wm8960_hw_params’:
In file included from ./include/linux/acpi.h:15,
                 from ./include/linux/i2c.h:13,
                 from /var/lib/dkms/seeed-voicecard/0.3/build/ac108.c:21:
/var/lib/dkms/seeed-voicecard/0.3/build/ac108.c: In function ‘ac108_set_fmt’:
/var/lib/dkms/seeed-voicecard/0.3/build/wm8960.c:752:6: warning: this statement may fall through [-Wimplicit-fallthrough=]
   if ((iface & 0x3) != 0) {
      ^
/var/lib/dkms/seeed-voicecard/0.3/build/wm8960.c:756:2: note: here
  default:
[... 87 lines cut ...]
             platforms
/var/lib/dkms/seeed-voicecard/0.3/build/seeed-voicecard.c:586:13: error: ‘struct snd_soc_dai_link’ has no member named ‘codec_name’; did you mean ‘stream_name’?
   dai_link->codec_name = cinfo->codec;
             ^~~~~~~~~~
             stream_name
/var/lib/dkms/seeed-voicecard/0.3/build/seeed-voicecard.c:587:13: error: ‘struct snd_soc_dai_link’ has no member named ‘cpu_dai_name’; did you mean ‘stream_name’?
   dai_link->cpu_dai_name = cinfo->cpu_dai.name;
             ^~~~~~~~~~~~
             stream_name
/var/lib/dkms/seeed-voicecard/0.3/build/seeed-voicecard.c:588:13: error: ‘struct snd_soc_dai_link’ has no member named ‘codec_dai_name’; did you mean ‘stream_name’?
   dai_link->codec_dai_name = cinfo->codec_dai.name;
             ^~~~~~~~~~~~~~
             stream_name
/var/lib/dkms/seeed-voicecard/0.3/build/seeed-voicecard.c:610:2: error: implicit declaration of function ‘asoc_simple_card_clean_reference’; did you mean ‘asoc_simple_clean_reference’? [-Werror=implicit-function-declaration]
  asoc_simple_card_clean_reference(&priv->snd_card);
  ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
  asoc_simple_clean_reference
cc1: some warnings being treated as errors
make[1]: *** [scripts/Makefile.build:266：/var/lib/dkms/seeed-voicecard/0.3/build/seeed-voicecard.o] 错误 1
make[1]: *** 正在等待未完成的任务....
make: *** [Makefile:1732：/var/lib/dkms/seeed-voicecard/0.3/build] 错误 2
make: 离开目录“/usr/src/linux-headers-5.4.72-v8+”
pi@raspberrypi:~ $

Thank you

xiaovsma, 2020-11-05:
> Hello
> 
> OS environment information, Two git projects, tried it but not:
> 
> 1. The drivers of 2 Mic Array Kit for Raspberry Pi.
> 2. OS info
>    pi@raspberrypi:~ $ uname -a
>    Linux raspberrypi 5.4.72-v8+ #1356 SMP PREEMPT Thu Oct 22 13:58:52 BST 2020 aarch64 GNU/Linux
>    pi@raspberrypi:~ $ cd seeed/hinTak-seeed-voicecard/
>    pi@raspberrypi:~/seeed/hinTak-seeed-voicecard $ git branch -a
> 
> * v5.5
>   pi@raspberrypi:~/seeed/hinTak-seeed-voicecard $ cd ../seeed-voicecard/; git branch -a
> * aarch64
>   pi@raspberrypi:~/seeed/seeed-voicecard $
> 
> 1. OS install drivers Kernel errs
> 
> DKMS: add completed.
> 
> Kernel preparation unnecessary for this kernel. Skipping...
> 
> Building module:
> cleaning build area...
> make -j4 KERNELRELEASE=5.4.72+ -C /lib/modules/5.4.72+/build M=/var/lib/dkms/seeed-voicecard/0.3/build...(bad exit status: 2)
> Error! Bad return status for module build on kernel: 5.4.72+ (aarch64)
> Consult /var/lib/dkms/seeed-voicecard/0.3/build/make.log for more information.
> 
> Kernel preparation unnecessary for this kernel. Skipping...
> 
> pi@raspberrypi:~ $ cat /var/lib/dkms/seeed-voicecard/0.3/build/make.log
> DKMS make.log for seeed-voicecard-0.3 for kernel 5.4.72-v8+ (aarch64)
> 2020年 11月 05日 星期四 14:49:18 CST
> make: 进入目录“/usr/src/linux-headers-5.4.72-v8+”
> AR /var/lib/dkms/seeed-voicecard/0.3/build/built-in.a
> CC [M] /var/lib/dkms/seeed-voicecard/0.3/build/wm8960.o
> CC [M] /var/lib/dkms/seeed-voicecard/0.3/build/ac108.o
> CC [M] /var/lib/dkms/seeed-voicecard/0.3/build/ac101.o
> CC [M] /var/lib/dkms/seeed-voicecard/0.3/build/seeed-voicecard.o
> /var/lib/dkms/seeed-voicecard/0.3/build/seeed-voicecard.c: In function ‘seeed_voice_card_dai_init’:
> /var/lib/dkms/seeed-voicecard/0.3/build/seeed-voicecard.c:262:8: error: implicit declaration of function ‘asoc_simple_card_init_dai’; did you mean ‘asoc_simple_dai_init’? [-Werror=implicit-function-declaration]
> ret = asoc_simple_card_init_dai(codec, &dai_props->codec_dai);
> ^~~~~~~~~~~~~~~~~~~~~~~~~
> asoc_simple_dai_init
> /var/lib/dkms/seeed-voicecard/0.3/build/wm8960.c: In function ‘wm8960_hw_params’:
> In file included from ./include/linux/acpi.h:15,
> from ./include/linux/i2c.h:13,
> from /var/lib/dkms/seeed-voicecard/0.3/build/ac108.c:21:
> /var/lib/dkms/seeed-voicecard/0.3/build/ac108.c: In function ‘ac108_set_fmt’:
> /var/lib/dkms/seeed-voicecard/0.3/build/wm8960.c:752:6: warning: this statement may fall through [-Wimplicit-fallthrough=]
> if ((iface & 0x3) != 0) {
> ^
> /var/lib/dkms/seeed-voicecard/0.3/build/wm8960.c:756:2: note: here
> default:
> ^~~~~~~
> ./include/linux/device.h:1758:5: warning: this statement may fall through [-Wimplicit-fallthrough=]
> ({ \
> 
> ```
>  if (0)        \
>  ~~~^~~~~~~~~~~~
>   dev_printk(KERN_DEBUG, dev, dev_fmt(fmt), ##__VA_ARGS__); \
> ```
> 
> })
> ~~
> /var/lib/dkms/seeed-voicecard/0.3/build/ac108.c:866:4: note: in expansion of macro ‘dev_dbg’
> dev_dbg(dai->dev, "used as slave when AC101 is master\n");
> ^~~~~~~
> /var/lib/dkms/seeed-voicecard/0.3/build/ac108.c:868:2: note: here
[... 100 lines cut ...]
An error occurred when executing the command: sudo ./install.h
Creating symlink /var/lib/dkms/seeed-voicecard/0.3/source ->
                 /usr/src/seeed-voicecard-0.3

DKMS: add completed.

Kernel preparation unnecessary for this kernel.  Skipping...

Building module:
cleaning build area...
make -j4 KERNELRELEASE=5.4.72-v8+ -C /lib/modules/5.4.72-v8+/build M=/var/lib/dkms/seeed-voicecard/0.3/build....(bad exit status: 2)
Error! Bad return status for module build on kernel: 5.4.72-v8+ (aarch64)
Consult /var/lib/dkms/seeed-voicecard/0.3/build/make.log for more information.
mkdir: 无法创建目录 “/etc/voicecard”: 文件已存在
git init
重新初始化已存在的 Git 仓库于 /etc/voicecard/.git/
git add --all
git commit -m "origin configures"
[master 7a13226] origin configures
 1 file changed, 2 insertions(+), 2 deletions(-)
------------------------------------------------------
Please reboot your raspberry pi to apply all settings
Enjoy!
------------------------------------------------------
pi@raspberrypi:~/seeed/hinTak-seeed-voicecard $

HinTak, 2020-11-05:
`asoc_simple_card_clean_reference` sounds like you are using a really old version of the driver code. When did you clone it?

xiaovsma, 2020-11-05:
> `asoc_simple_card_clean_reference` sounds like you are using a really old version of the driver code. When did you clone it?

我尝试过该项目下面的所有半 貌似都失败了(HinTak/seeed-voicecard)
I have tried all the halves of the project and it seems to have failed (HinTak/seeed-voicecard)

xiaovsma, 2020-11-05:
> `asoc_simple_card_clean_reference` sounds like you are using a really old version of the driver code. When did you clone it?

I have tried all the versions of the project below and it seems to have failed (HinTak/seeed-voicecard)
我尝试过该项目下面的所有版本 貌似都失败了(HinTak/seeed-voicecard)

xiaovsma, 2020-11-05:
> > `asoc_simple_card_clean_reference` sounds like you are using a really old version of the driver code. When did you clone it?
> 
> I have tried all the versions of the project below and it seems to have failed (HinTak/seeed-voicecard)
> 我尝试过该项目下面的所有版本 貌似都失败了(HinTak/seeed-voicecard)

这几天来来回回下载好几次了, 还是失败告终(I have downloaded it back and forth several times these days, but it still fails.)

xiaovsma, 2020-11-05:
> `asoc_simple_card_clean_reference` sounds like you are using a really old version of the driver code. When did you clone it?

麻烦您提供一个稳定适合我的驱动版本, 我可以按照您说的版本 我再次尝试安装一下(Please provide a driver version that is stable and suitable for me, I can follow the version you said, I will try to install it again)

HinTak, 2020-11-05:
In the current version of code, all `*simple_card*` had been changed to `*simple*`, without the `_card` part.

xiaovsma, 2020-11-05:
> *simple*

没有找到相关的文件, 请问在那个git项目的哪个分支下 有这些文件*simple*(Did not find the relevant files, which branch of the git project has these files *simple*)

xiaovsma, 2020-11-05:
> In the current version of code, all `*simple_card*` had been changed to `*simple*`, without the `_card` part.

没有找到相关的文件, 请问在那个git项目的哪个分支下 有这些文件simple(Did not find the relevant files, which branch of the git project has these files simple)

HinTak, 2020-11-05:
E.g. https://github.com/respeaker/seeed-voicecard/blob/8cce4e8ffa77e1e2b89812e5e2ccf6cfbc1086cf/seeed-voicecard.c#L725

xiaovsma, 2020-11-05:
> E.g.
> 
> https://github.com/respeaker/seeed-voicecard/blob/8cce4e8ffa77e1e2b89812e5e2ccf6cfbc1086cf/seeed-voicecard.c#L725
> 
> And
> 
> https://github.com/HinTak/seeed-voicecard/blob/87f42ed64dcf6f218b6f0a3d582e94ddf01f9bbc/seeed-voicecard.c#L725

我找到该文件了, 请问我下一步如何操作?(I found the file, what should I do next?)

```
pi@raspberrypi:~/seeed/seeed-voicecard $ ls seeed-voicecard.c 
seeed-voicecard.c
pi@raspberrypi:~/seeed/seeed-voicecard $ ls ../hinTak-seeed-voicecard/seeed-voicecard.c 
../hinTak-seeed-voicecard/seeed-voicecard.c
pi@raspberrypi:~/seeed/seeed-voicecard $ 
```

xiaovsma, 2020-11-05:
在grep时候没有找到相关内容, 如下(No relevant content was found during grep, as follows)

```
pi@raspberrypi:~/seeed/seeed-voicecard $ cat seeed-voicecard.c  | grep  asoc_simple_clean_reference
pi@raspberrypi:~/seeed/seeed-voicecard $ cat ../hinTak-seeed-voicecard/seeed-voicecard.c  | grep  asoc_simple_clean_reference

and

pi@raspberrypi:~/seeed/hinTak-seeed-voicecard $ cat seeed-voicecard.c  | grep  "priv->snd_card"
        priv->snd_card.aux_dev = devm_kzalloc(dev,
                        n * sizeof(*priv->snd_card.aux_dev), GFP_KERNEL);
        if (!priv->snd_card.aux_dev)
                priv->snd_card.aux_dev[i].codec_of_node = aux_node;
        priv->snd_card.num_aux_devs = n;
                ret = snd_soc_of_parse_audio_simple_widgets(&priv->snd_card,
                ret = snd_soc_of_parse_audio_routing(&priv->snd_card,
        ret = asoc_simple_card_parse_card_name(&priv->snd_card, PREFIX);
        priv->snd_card.owner            = THIS_MODULE;
        priv->snd_card.dev              = dev;
        priv->snd_card.dai_link         = priv->dai_link;
        priv->snd_card.num_links        = num;
                priv->snd_card.name     = (cinfo->card) ? cinfo->card : cinfo->name;
        snd_soc_card_set_drvdata(&priv->snd_card, priv);
        ret = devm_snd_soc_register_card(&pdev->dev, &priv->snd_card);
        asoc_simple_card_clean_reference(&priv->snd_card);
pi@raspberrypi:~/seeed/hinTak-seeed-voicecard $ cat ../seeed-voicecard/seeed-voicecard.c  | grep  "priv->snd_card"
        priv->snd_card.aux_dev = devm_kzalloc(dev,
                        n * sizeof(*priv->snd_card.aux_dev), GFP_KERNEL);
        if (!priv->snd_card.aux_dev)
                priv->snd_card.aux_dev[i].codec_of_node = aux_node;
        priv->snd_card.num_aux_devs = n;
                ret = snd_soc_of_parse_audio_simple_widgets(&priv->snd_card,
                ret = snd_soc_of_parse_audio_routing(&priv->snd_card,
        ret = asoc_simple_card_parse_card_name(&priv->snd_card, PREFIX);
        priv->snd_card.owner            = THIS_MODULE;
        priv->snd_card.dev              = dev;
        priv->snd_card.dai_link         = priv->dai_link;
        priv->snd_card.num_links        = num;
                priv->snd_card.name     = (cinfo->card) ? cinfo->card : cinfo->name;
        snd_soc_card_set_drvdata(&priv->snd_card, priv);
        ret = devm_snd_soc_register_card(&pdev->dev, &priv->snd_card);
        asoc_simple_card_clean_reference(&priv->snd_card);
pi@raspberrypi:~/seeed/hinTak-seeed-voicecard $ 
```

xiaovsma, 2020-11-05:
@HinTak 我现在如何操作呢, 还在困惑中, 麻烦给予帮助, 辛苦你了(How do I do it now, still in confusion, please help me, thank you)

HinTak, 2020-11-05:
How did you end up with the wrong code? I'd like to see the top few lines of `git reflog` output. That should show why you end up with the wrong code.

Anyway, just delete everything and start again...

xiaovsma, 2020-11-06:
> How did you end up with the wrong code? I'd like to see the top few lines of `git reflog` output. That should show why you end up with the wrong code.
> 
> Anyway, just delete everything and start again...

```
pi@raspberrypi:~ $ cd seeed/seeed-voicecard/
pi@raspberrypi:~/seeed/seeed-voicecard $ git reflog 
1d405cf (HEAD -> aarch64, origin/aarch64) HEAD@{0}: clone: from https://github.com/respeaker/seeed-voicecard.git
pi@raspberrypi:~/seeed/seeed-voicecard $ cd ../hinTak-seeed-voicecard/; git reflog
19d1e6e (HEAD -> master, origin/master) HEAD@{0}: checkout: moving from v5.5 to master
87f42ed (origin/v5.5, origin/HEAD, v5.5) HEAD@{1}: clone: from https://github.com/HinTak/seeed-voicecard.git
pi@raspberrypi:~/seeed/hinTak-seeed-voicecard $ 
pi@raspberrypi:~/seeed/hinTak-seeed-voicecard $ 
pi@raspberrypi:~/seeed/hinTak-seeed-voicecard $ 
```

xiaovsma, 2020-11-06:
删除重新下载项目尝试安装再次失败(Delete the re-download item and try to install again and fail)
```
pi@raspberrypi:~/seeed/seeed-voicecard $ git branch -a
* master
  remotes/origin/HEAD -> origin/master
  remotes/origin/aarch64
  remotes/origin/artikpi
  remotes/origin/dev-bbg
  remotes/origin/linux-4.19-or-less
  remotes/origin/master
  remotes/origin/rel-v5.5
  remotes/origin/wm8960_debug
pi@raspberrypi:~/seeed/seeed-voicecard $ git reflog
8cce4e8 (HEAD -> master, origin/rel-v5.5, origin/master, origin/HEAD) HEAD@{0}: clone: from https://github.com/respeaker/seeed-voicecard.git
pi@raspberrypi:~/seeed/seeed-voicecard $ 
```

or ...

```
pi@raspberrypi:~/seeed/seeed-voicecard $ cd ../hinTak-seeed-voicecard/
pi@raspberrypi:~/seeed/hinTak-seeed-voicecard $ git reflog
87f42ed (HEAD -> v5.5, origin/v5.5, origin/HEAD) HEAD@{0}: clone: from https://github.com/HinTak/seeed-voicecard.git
pi@raspberrypi:~/seeed/hinTak-seeed-voicecard $ git branch -a
* v5.5
  remotes/origin/HEAD -> origin/v5.5
  remotes/origin/master
  remotes/origin/remove-forced-kernel-downgrade
  remotes/origin/v5.5
  remotes/origin/v5.7
  remotes/origin/v5.8
pi@raspberrypi:~/seeed/hinTak-seeed-voicecard $ 
```

xiaovsma, 2020-11-06:
删除重新下载项目尝试安装再次失败(Delete the re-download item and try to install again and fail)

```
pi@raspberrypi:~/seeed/seeed-voicecard $ git branch -a
* master
  remotes/origin/HEAD -> origin/master
  remotes/origin/aarch64
  remotes/origin/artikpi
  remotes/origin/dev-bbg
  remotes/origin/linux-4.19-or-less
  remotes/origin/master
  remotes/origin/rel-v5.5
  remotes/origin/wm8960_debug
pi@raspberrypi:~/seeed/seeed-voicecard $ git reflog
8cce4e8 (HEAD -> master, origin/rel-v5.5, origin/master, origin/HEAD) HEAD@{0}: clone: from https://github.com/respeaker/seeed-voicecard.git
pi@raspberrypi:~/seeed/seeed-voicecard $ 
```

or

```
pi@raspberrypi:~/seeed/seeed-voicecard $ git branch -a
* aarch64
  master
  remotes/origin/HEAD -> origin/master
  remotes/origin/aarch64
  remotes/origin/artikpi
  remotes/origin/dev-bbg
  remotes/origin/linux-4.19-or-less
  remotes/origin/master
  remotes/origin/rel-v5.5
  remotes/origin/wm8960_debug
pi@raspberrypi:~/seeed/seeed-voicecard $ git reflog
1d405cf (HEAD -> aarch64, origin/aarch64) HEAD@{0}: checkout: moving from master to aarch64
8cce4e8 (origin/rel-v5.5, origin/master, origin/HEAD, master) HEAD@{1}: clone: from https://github.com/respeaker/seeed-voicecard.git
pi@raspberrypi:~/seeed/seeed-voicecard $ 
```

or ...

```
pi@raspberrypi:~/seeed/seeed-voicecard $ cd ../hinTak-seeed-voicecard/
pi@raspberrypi:~/seeed/hinTak-seeed-voicecard $ git reflog
87f42ed (HEAD -> v5.5, origin/v5.5, origin/HEAD) HEAD@{0}: clone: from https://github.com/HinTak/seeed-voicecard.git
pi@raspberrypi:~/seeed/hinTak-seeed-voicecard $ git branch -a
* v5.5
  remotes/origin/HEAD -> origin/v5.5
  remotes/origin/master
  remotes/origin/remove-forced-kernel-downgrade
  remotes/origin/v5.5
  remotes/origin/v5.7
  remotes/origin/v5.8
pi@raspberrypi:~/seeed/hinTak-seeed-voicecard $ 
```

xiaovsma, 2020-11-10:
今天发现了新问题, 可能是由此导致安装失败的问题; 尝试安装找不到源包(A new problem was discovered today, which may have caused the installation to fail; try to install the source package)
```
正在读取状态信息... 完成       
E: 无法定位软件包 linux-raspi
E: 无法定位软件包 linux-headers-raspi
E: 无法定位软件包 linux-image-raspi
正在读取软件包列表... 完成
```

HinTak, 2020-11-10:
Those are harmless - raspbian/debian and ubuntu have different kernel package names, so one set or the other is always missing. You are missing the ubuntu set, which is expected on raspbian.

xiaovsma, 2020-11-10:
> Those are harmless - raspbian/debian and ubuntu have different kernel package names, so one set or the other is always missing. You are missing the ubuntu set, which is expected on raspbian.

I don’t know what exactly caused the installation to fail, but it still fails after trying many ways

xiaovsma, 2020-11-11:
> E.g.
> 
> https://github.com/respeaker/seeed-voicecard/blob/8cce4e8ffa77e1e2b89812e5e2ccf6cfbc1086cf/seeed-voicecard.c#L725
> 
> And
> 
> https://github.com/HinTak/seeed-voicecard/blob/87f42ed64dcf6f218b6f0a3d582e94ddf01f9bbc/seeed-voicecard.c#L725

closed 2020-11-11 by xiaovsma

WilliamVJacob, 2021-02-19:
I have added the fix to install Respeaker Driver for 64 bit Raspbian OS
Check my forked  seeed-voicecard  repository  for solution
https://github.com/WilliamVJacob/seeed-voicecad