respeaker/seeed-voicecard #261: Noise on output

ervinbalazs, 2020-11-05:
After I power up my raspberry there is a noise on the output of the sound card. The noise is present until I play any sound. After the play stops the noise never comes back. 

If I restart the system the noise comes back until I play a sound file again.

Is the a possible solution in driver to solve this problem?

joshuajaharwood, 2020-11-09:
Are you using ALSA or Pulseaudio + ALSA?

HinTak, 2020-11-09:
This https://github.com/respeaker/seeed-voicecard/issues/251#issuecomment-708271666 might be an answer to this sort of problem...

joshuajaharwood, 2020-11-11:
> This [#251 (comment)](https://github.com/respeaker/seeed-voicecard/issues/251#issuecomment-708271666) might be an answer to this sort of problem...

I'm almost certain it is. Starting and stopping playback causes popping as the voicecard is powered up (can recreate this on my end).

AIWintermuteAI, 2021-08-13:
Hi, @ervinbalazs !
I cannot reproduce the problem with the following setup:

Raspberry Pi 4 1 Gb
5.10.17-v7l+ kernel
b595b95
2-mic Hat for Raspberry Pi

The sound output of both headphone jack and JST2.0 Speaker Out does not contain any noise after reboot.

If the issue still persists for you, can you share more information about your setup?