respeaker/seeed-voicecard #263: install unsuccessful  kernel headers for kernel 5.4.51-v8+ cannot be found

er1k1, 2020-11-10:
Raspberry Pi 3B with Linux 9 Stretch, initial sudo ./install seemed to be successful using v7+ kernel:
...
Building module:
cleaning build area...
make -j4 KERNELRELEASE=5.4.51-v7+ -C /lib/modules/5.4.51-v7+/build M=/var/lib/dkms/seeed-voicecard/0.3/build.....
cleaning build area...

DKMS: build completed.
 ...
### Start service seeed-voicecard
    see /var/log/seeed-voicecard.log for more service information
Created symlink /etc/systemd/system/sysinit.target.wants/seeed-voicecard.service → /lib/systemd/system/seeed-voicecard.service.
------------------------------------------------------
Please reboot your device to apply all settings
Enjoy!

But I couldn't see the hardware with aplay -l or arecord -l and no seeed service
rerunning with or without --compat-kernel leads to the kernel headers error "5.4.51-v8+ cannot be found" and aborting build.

If I try the make command above directly or with cmd "dkms build -k 5.4.51-v7+ -m seeed-voicecard -v 0.3"  I get a compile error in the log:
...
gcc: error: unrecognized command line option ‘-mgeneral-regs-only’

HinTak, 2020-11-10:
Mostly similar problem as https://github.com/respeaker/seeed-voicecard/issues/256

er1k1, 2020-11-10:
It is similar but is there a solution? I have tried cloning from HinTak repository and the result is the same.
seeed-voicecard.service is not shown on > sudo service --status-all
..
DKMS: add completed.
Error! echo
Your kernel headers for kernel 5.4.51-v8+ cannot be found at
/lib/modules/5.4.51-v8+/build or /lib/modules/5.4.51-v8+/source.
mkdir: cannot create directory ‘/etc/voicecard’: File exists
git init
Reinitialized existing Git repository in /etc/voicecard/.git/
git add --all
git commit -m "origin configures"
[master 4086f25] origin configures
 1 file changed, 32 insertions(+), 2 deletions(-)
------------------------------------------------------
Please reboot your raspberry pi to apply all settings
Enjoy!

HinTak, 2020-11-10:
I am waiting for Seeed Studio staff to address https://github.com/respeaker/seeed-voicecard/issues/251 , before I spend any more time on seeed studio code. Don't want to look at issues that they know for months.

@Pillar1989 @turmary

er1k1, 2020-11-16:
@Pillar1989 @turmary - Would you like to comment? This product is unusable without working drivers and it appears the driver doesn't work on later Raspbian versions?
Seeed tech support say " Pillar1989 and turmary are our professional engineers in this aspect".
It seems the similar issue was raised 2 months ago and I don't see any solution. If there is one, could you please point me to it,
Thanks

Pillar1989, 2020-11-17:
I don't think there is a problem with the current software. I hope you can read the documentation more. @er1k1

mikethebee, 2020-11-17:
FYI: I got the "...kernel headers for kernel 5.4.51-v8+ cannot be found" message too on a rpi3b with latest OS install (uname=5,4,72-v7+), but the card seems to work normally after a reboot and selecting "seeed-2-mic-voicecard" on the taskbar Audio Output.

er1k1, 2020-11-19:
> I don't think there is a problem with the current software. I hope you can read the documentation more. @er1k1

Invaluable advice.. Where in "the documentation" does it say you cannot uninstall the driver? and if you get errors tough luck?
Thank you so much for your diligence, I won't forget to rate the product.

Gandalfthegreybeard, 2021-01-12:
I note that there is an aarch64 branch... but that does not work for me with the 5.10.5-v8+ kernel.. I believe this is likely due to the change in struct snd_soc_pcm_runtime specifically with respect to how it associates cpu_dai...

HinTak, 2021-01-12:
@Gandalfthegreybeard I have not looked at 5.10 yet (it is out for only a couple of weeks? After Christmas I think?) but you need the v5.9 branch on my repo at least to have a chance of getting v5.10 to work. If it works for you, please feel free to click the donate link.

There are a few incompatible changes between v5.4 which every raspberrypi distro is using, vs 5.9 . As I said, I have not looked at 5.9 vs 5.10

HinTak, 2021-01-12:
And btw, v5.10 is definitely a separate issue, and you should file a new one, rather than hijacking an old issue...

Gandalfthegreybeard, 2021-01-12:
> And btw, v5.10 is definitely a separate issue, and you should file a new one, rather than hijacking an old issue...

Planning on it... and am considering trying the changes myself.. just want to ensure I've encapsulated the issue correctly first..

HinTak, 2021-01-12:
@Gandalfthegreybeard my v5.9 branch builds okay against 5.10 . Anyway, there are incompatible changes between 5.6<->5.7, 5.7<->5.8, and 5.8<->5.9 , hence separate branches.

If it works for you, please feel free to click on the donate link.

AIWintermuteAI, 2021-08-07:
Hi, @er1k1 !
Try installing latest commit of seeed-voicecard on current latest Raspi kernel OS image - 5.10.17-v7l+ kernel. 
Feel free to re-open the issue if it still persists - do however use the template above to describe the problem.

closed 2021-08-07 by AIWintermuteAI