respeaker/seeed-voicecard #265: Kernel Panic on Ubuntu 32 Bit

tomh05, 2020-11-17:
I have a 6-mic array connected to a Raspberry Pi 3B, with a fresh install of 32 bit Ubuntu 20.04.

The install script works, and I can see the device listed in `arecord -L`. But as soon as I run `aplay`or `arecord` from the [seeed instructions](https://wiki.seeedstudio.com/ReSpeaker_6-Mic_Circular_Array_kit_for_Raspberry_Pi/#software), I get a kernel panic that ends "not syncing: Fatal exception in interrupt". 😢  

I wondered if it was related to  https://github.com/respeaker/seeed-voicecard/issues/258 so I tried `ccflags-y += --no_protect_stack` to the build script  as [suggested in this post](https://github.com/respeaker/seeed-voicecard/issues/258#issuecomment-729120542), but it just broke the install script:

```
Building module:
cleaning build area.........
make -j4 KERNELRELEASE=5.4.0-1022-raspi -C /lib/modules/5.4.0-1022-raspi/build M=/var/lib/dkms/seeed-voicecard/0.3/build.....(bad exit status: 2)
ERROR (dkms apport): binary package for seeed-voicecard: 0.3 not found
Error! Bad return status for module build on kernel: 5.4.0-1022-raspi (armv7l)
Consult /var/lib/dkms/seeed-voicecard/0.3/build/make.log for more information.
Can't compile with this kernel, aborting
Please try to compile with the option --compat-kernel
```

HinTak, 2020-11-17:
Was the kernel panic preceded by "scheduling while atomic"? It is probably https://github.com/respeaker/seeed-voicecard/issues/251

HinTak, 2020-11-17:
The `__stack_chk_guard` is a raspbian issue with them mispackaging the header package, and does not apply to ubuntu.

tomh05, 2020-11-17:
Ah, yes. The text was scrolling too fast to see at first, but I took a video recording, and yes it does say "BUG: scheduling while atomic". Probably a dupe of #251 then.

AIWintermuteAI, 2021-08-07:
Hi @tomh05 !
I'll be marking issue as duplicate. We are aware of the issue described in https://github.com/respeaker/seeed-voicecard/issues/251 and will be investigating it after we're done with backlog.

closed 2021-08-07 by AIWintermuteAI

HinTak, 2021-08-07:
@tomh05 btw, there is a workaround in #251, if you are willing to switch to ubuntu 64-bit. (and similar, but more involved change with ubuntu 32-bit too), not a proper fix but at least get the pi and audio generally usable.