respeaker/seeed-voicecard #266: FAQ Link expired

tomh05, 2020-11-18:
At the bottom of README.md, the link to the raspberry pi image gives this message:

> Support for the company which the current shared item is in has been expired. Please contact the share link owner.

bennoheld, 2020-11-23:
This is actually quite frustrating, because according to the README this should be the first approach to solving problems. Please update as soon as possible!

fraserabrown, 2021-03-16:
+1. please fix this or provide another download link. i have bought a mic array which i cannot use.

AIWintermuteAI, 2021-06-30:
Re-uploaded fresh image.
https://github.com/respeaker/seeed-voicecard#technical-support

**Keep in mind, that the image is made only for hardware testing purposes** - it won't be frequently updated, since recommended way of using reSpeaker is to install drivers from this repository.
Closing the issue.

closed 2021-06-30 by AIWintermuteAI