respeaker/seeed-voicecard #267: Random noise on one channel after a while

Daenara, 2020-11-19:
I know this does not sound like a driver issue but I am pretty sure it is. My respeaker4 (square) works very well with the drivers that HinTak fixed now but only for a time. Before the drivers were fixed I always had random noise on at least one channel and my voice couldn't be heard at all, this vanished after [this was found](https://github.com/respeaker/seeed-voicecard/commit/9722cb01e480c0b5b5e94bc6d9f8862c5733f32e). Now I noticed that I still get random noise on at least one channel (which channel seems to be random) after my pi has been running for around two to three hours. Simply rebooting fixes it.

I have not looked at dmesg yet, because I haven't spend enough time at home that wasn't spend sleeping since I noticed this, but I will take a look at that once I am at my pi the next time and post it. I also still run on HinTaks version of the driver, what I have seen committed after that on the official repro has not looked like it would fix this issue, and every time I try to change the drivers I end up with a completely broken mic and have to do a complete reinstall which is not something I prefer to doing with my voice assistant every time an update comes out.

What I would like to know is, if anyone has noticed this issue, or has an idea how to fix it (which I know the dmesg will be needed for, I will add that this evening or tomorrow). If there is a change that has been done already that might have fixed this that I did not recognize as a potential fix, I would love to be pointed at the relevant commit so that I know that the risk of a complete reinstall of my system would be a risk worth taking.

HinTak, 2020-11-24:
I filed this some time ago https://github.com/raspberrypi/linux/issues/3580 - it is probably a milder version of #251 but it would be interesting to know if there is any correlation of the onset of noise with I2s errors.

And are you seeing channel shifts also? As in the mappings of channels off by 2 - in your case, probably 1234 mapped to 3412.

Daenara, 2020-11-24:
Where do I see the channel shifts?

One thing I noticed was that my power supply for the pi went bad, after replacing that the loudness of the noise got better to the point of actually being able to hear what I said over it, but I still get random noise sometimes

HinTak, 2020-11-24:
A channel shift for the square 4-mics would show up as being wrong by 180 degree from a DOA (detecting direction of sound). Ie software analyzing the sound would say the sound came from the exact opposite direction as the correct one.

AIWintermuteAI, 2021-08-07:
Hi, @Daenara !
I'll be closing the issue as it seems it was resolved, or at least the cause of it was found (hardware defect) as pointed out in https://github.com/respeaker/seeed-voicecard/issues/289#issuecomment-825121465
If you have more feedback about this hardware defect, you can contact Seeed tech support as specified here:
**Please send mail to techsupport@seeed.cc for technical support and product quality issue, the mail should contain information of product SKU No./product name and where**

you ordered the product.

closed 2021-08-07 by AIWintermuteAI

HinTak, 2021-08-07:
@Daenara btw, the box it came with has some date and batch/serial number on - if you still have them, it could be useful to post the batch/serial number too. I fully understand given the amount of trouble the broken unit caused, so you may not want to spend any time looking for the old box/packaging...