respeaker/seeed-voicecard #268: Install on other OS (libreelec/kodi, volumio etc.)

h4de5, 2020-12-02:
Hello,

I am wondering if it was possible to install and run the 4mic-array on any other OS than raspbian. I would like to install them on Kodi (https://github.com/LibreELEC/LibreELEC.tv) or Volumio (https://github.com/volumio). On both systems the install script fails:

```
$ ./install.sh
bash: 0: unknown operand
df: invalid option -- 'B'
BusyBox v1.31.0 (2020-10-24 15:30:09 EDT) multi-call binary.

Usage: df [-PkmhT] [FILESYSTEM]...

Print filesystem usage statistics

        -P      POSIX output format
        -k      1024-byte blocks (default)
        -m      1M-byte blocks
        -h      Human readable (e.g. 1K 243M 2G)
        -T      Print filesystem type
Warning: /boot volume not found ..
/boot/overlays not found or not a directory
Errors found, exiting.
```

Both systems are running on a RPI4.

kodi:
```
$ uname -a
Linux kodi 4.19.127 #1 SMP Sat Oct 24 15:01:22 EDT 2020 armv7l GNU/Linux
$ cat /proc/version
Linux version 4.19.127 (jenkins@prometheus) (gcc version 8.3.0 (GCC)) #1 SMP Sat Oct 24 15:01:22 EDT 2020

```
volumio:
```
$ uname -a
Linux volumio 4.19.49v6v7-aufs #1 SMP Tue Jun 11 15:13:27 CEST 2019 armv7l GNU/Linux
$ cat /proc/version
Linux version 4.19.49v6v7-aufs (max@lynx) (gcc version 8.2.0 (Buildroot 2018.08-gbbba56f1-dirty)) #1 SMP Tue Jun 11 15:13:27 CEST 2019
```

HinTak, 2020-12-02:
The current code has dropped older kernel (4.19) support, I think, and also too much bash-isms. Busybox's emulation of bash isn't up for it, I think. And I have never agreed with the whole check disk space and downgrade kernel thing , so you may have better luck with https://github.com/HinTak/seeed-voicecard . That said, it should be possible to rollback and get 4.19 support back again. (revert commits marked "v5.x" etc) . You probably want to follow some of the raspberrypi dev tips I wrote in the raspberrypi-dev repo. See https://hintak.github.io .

HinTak, 2020-12-02:
I don't agree with the whole downgrade kernel thing which "Warning: /boot volume" is for.

h4de5, 2020-12-02:
@HinTak thank you very much for the fast response and the idea.

I am trying to manually go through your install.sh script on libreelec.

* I copied the dtbo files and added the dtparam and dtoverlay strings to the config.txt 
* dtparam and dtoverlay work - but they both don't show any loaded overlays after reboot.
* I left out anything related to kernel checks and apt-get (because apt-get wont work)
* I don't know what to do about the /etc/modules 
* the commands `i2cdetec` and `arecord` are not available.

how far would I need to go back in the git history? I see the bigger changes are starting at around beginning of this year. and adding support for 4.18 at around march 2019. so somewhere in between?

HinTak, 2020-12-02:
@h4de5 actually 5.4 support (for current raspbian /Ubuntu ) was only merged in summer 2020 (after many months of unhappiness of users being forced to stay in old OSes) - so you only need to roll back a few months. I'll make a v4.19 branch soon - just stay on that branch and you'd be fine, and I'd be interested to know what adjustments you need to make to make it work.

Your first 4 points aren't important, if you are happy with staying with one device and hardcoding that into your config.txt . A lot of the complexity is strictly speaking not necessary and only for dynamically detectino which respeaker it is and load the correct one. So that covers half of your 5th point too. To check that you have manually configure correctly you need to read your boot messages (`dmesg`).

It is a bit painful without `arecord` but maybe you have pyalsa /python binding of libalsa) or equivalent?

h4de5, 2020-12-03:
wow. thanks a lot.
but I am not sure where to start from here. I switched to the new branch and run `seeed-voicecard`

thats the outcome.
```
+ export 'PATH=/usr/bin:/usr/sbin:/storage/.kodi/addons/service.system.docker/bin:/storage/.kodi/addons/virtual.rpi-tools/bin:/storage/.kodi/addons/virtual.system-tools/bin:/opt/vc/bin'
+ OVERLAYS=/boot/overlays
+ '[' -d /boot/firmware/overlays ]
+ dtparam -d /boot/overlays 'i2c_arm=on'
+ modprobe i2c-dev
+ dtparam -d /boot/overlays 'spi=on'
+ _VER_RUN=
+ CONFIG=/boot/config.txt
+ '[' -f /boot/firmware/usercfg.txt ]
+ i2cdetect -y 1 0x1a 0x1a
+ egrep '(1a|UU)'
+ awk '{print $2}'
+ is_1a=
+ i2cdetect -y+ egrep '(35|UU)'
 1 0x35 0x35
+ awk '{print $2}'
+ is_35=
+ i2cdetect -y 1 0x3b 0x3b
+ egrep '(3b|UU)'
+ awk '{print $2}'
+ is_3b=3b
+ RPI_HATS='seeed-2mic-voicecard seeed-4mic-voicecard seeed-8mic-voicecard'
+ overlay=
+ '[' x '!=' x ]
+ '[' x3b '!=' x ]
+ '[' x '==' x ]
+ echo 'install 4mic'
install 4mic
+ overlay=seeed-4mic-voicecard
+ asound_conf=/etc/voicecard/asound_4mic.conf
+ asound_state=/etc/voicecard/ac108_asound.state
+ '[' x3b '!=' x ]
+ '[' x '!=' x ]
+ '[' seeed-4mic-voicecard ]
+ echo Install seeed-4mic-voicecard ...
Install seeed-4mic-voicecard ...
+ rm /etc/asound.conf
rm: can't remove '/etc/asound.conf': No such file or directory
+ rm /var/lib/alsa/asound.state
rm: can't remove '/var/lib/alsa/asound.state': No such file or directory
+ uname -r
+ kernel_ver=4.19.127
+ '[[' 4.19.127 '=~' '^4.19.*$' -o 4.19.127 '=~' '^5.*$' ]]
bash: =~: unknown operand
+ dtoverlay -d /boot/overlays seeed-4mic-voicecard
DTOVERLAY[error]: failed to open '/boot/overlays/seeed-4mic-voicecard.dtbo'
* Failed to read '/boot/overlays/seeed-4mic-voicecard.dtbo'
+ true
+ echo 'create seeed-4mic-voicecard asound configure file'
create seeed-4mic-voicecard asound configure file
+ ln -s /etc/voicecard/asound_4mic.conf /etc/asound.conf
ln: /etc/asound.conf: Read-only file system
+ echo 'create seeed-4mic-voicecard asound status file'
create seeed-4mic-voicecard asound status file
+ ln -s /etc/voicecard/ac108_asound.state /var/lib/alsa/asound.state
ln: /var/lib/alsa/asound.state: No such file or directory
+ alsactl restore
alsactl: load_state:1683: Cannot open /var/lib/alsa/asound.state for reading: No such file or directory
+ aplay -l
+ grep -q 'bcm2835 ALSA'
aplay: device_list:272: no soundcards found...
```

I also found a way to install `i2cdetect -y 1`

```
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:          -- -- -- -- -- -- -- -- -- -- -- -- --
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
30: -- -- -- -- -- -- -- -- -- -- -- 3b -- -- -- --
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
70: -- -- -- -- -- -- -- --
```

there is no `/boot/` folder - overlays are located in: `/flash/overlays/`
and `/etc/` is readonly. `asound.conf` should go into `/storage/.config/` (which I did)
i think `asound.state` should go to: `/var/media/root/var/lib/alsa/asound.state` (which I did)

still - `alsoctl `searches for the wrong path:

```
alsactl restore
alsactl: load_state:1683: Cannot open /var/lib/alsa/asound.state for reading: No such file or directory
```

HinTak, 2020-12-03:
@h4de5 that seems to have gone better than expected. So now we have a 3rd variants:

- raspbian Ubuntu kodi
OVERLAYS /boot/overlays /boot/firmware/overlays /flash/overlays
CONFIG /boot/config.txt /boot/firmware/usercfg.txt. ????

Possibly `/flash/config.txt`?

"bash: =~: unknown operand" is another busybox vs bash issue.

I think you need to have a look at `/flash/config.txt`. Adding a line 
`dtoverlay=seeed-4mic-voicecard` to the config file manually is equivalent to all the i2cdetect and `dtoverlay -d ...`, so if you add that line you can drop most of
seeed-voicecard 's content. As I wrote earlier most of it is just trying to dynamically determine if you have 2-mics 4 mics or 6-mics.

h4de5, 2020-12-03:
mmh. you are making me feel, like we could actually get this thing running.. 

I had put `dtoverlay=seeed-4mic-voicecard.dtbo` - which I fixed that now.

I also added this stuff (that I found on the internet, but actually have no idea about..)

```
dtparam=i2c1=on
dtparam=i2c_arm=on
dtoverlay=i2s-mmap
dtparam=audio=on
```

is any of this needed?

not sure about: `usercfg.txt` is there any way to test it without `seeed-voicecard`?

HinTak, 2020-12-03:
I have downloaded the LibreELEC image and have a look. So I updated the v4.19 branch. The correct place is `/flash/distroconfig.txt` though `/boot/config.txt` includes the former and works too. 

Since you seem to be able to run `i2cdetect` correctly, you don't need to worry about the dtparam stuff.

I think we need to see your `dmesg` output with the config.txt change and the card physically hooked up, that"s all. Curious libreELEC have aplay but no arecord! But you should be able to look at `aplay -L` and verify that the card is installed correctly, and use it to play some sound through the respeaker's phone output jack! (follow the instruction in the README).

h4de5, 2020-12-03:
`aplay -L`:

  
```
null
    Discard all samples (playback) or generate zero samples (capture)
sysdefault:CARD=ALSA
    bcm2835 ALSA, bcm2835 ALSA
    Default Audio Device
iec958:CARD=ALSA,DEV=0
    bcm2835 ALSA, bcm2835 IEC958/HDMI
    IEC958 (S/PDIF) Digital Audio Output
```

<details>
  <summary>dmesg</summary>

```
[    0.000000] Booting Linux on physical CPU 0x0
[    0.000000] Linux version 4.19.127 (jenkins@prometheus) (gcc version 8.3.0 (GCC)) #1 SMP Sat Oct 24 15:01:22 EDT 2020
[    0.000000] CPU: ARMv7 Processor [410fd083] revision 3 (ARMv7), cr=30c5383d
[    0.000000] CPU: div instructions available: patching division code
[    0.000000] CPU: PIPT / VIPT nonaliasing data cache, PIPT instruction cache
[    0.000000] OF: fdt: Machine model: Raspberry Pi 4 Model B Rev 1.1
[    0.000000] Memory policy: Data cache writealloc
[    0.000000] cma: Reserved 256 MiB at 0x000000000fc00000
[    0.000000] On node 0 totalpages: 901120
[    0.000000]   DMA zone: 1152 pages used for memmap
[    0.000000]   DMA zone: 0 pages reserved
[    0.000000]   DMA zone: 131072 pages, LIFO batch:31
[    0.000000]   HighMem zone: 770048 pages, LIFO batch:63
[    0.000000] random: get_random_bytes called from start_kernel+0xc0/0x4d8 with crng_init=0
[    0.000000] percpu: Embedded 17 pages/cpu s37580 r8192 d23860 u69632
[    0.000000] pcpu-alloc: s37580 r8192 d23860 u69632 alloc=17*4096
[    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 899968
[    0.000000] Kernel command line: coherent_pool=1M 8250.nr_uarts=0 cma=64M cma=256M video=HDMI-A-1:1920x1080M@60 smsc95xx.macaddr=DC:A6:32:49:9E:98 vc_mem.mem_base=0x3ec00000 vc_mem.mem_size=0x40000000  boot=LABEL=System disk=LABEL=Storage quiet root=/dev/ram0 rdinit=/init usbcore.autosuspend=-1
[    0.000000] Dentry cache hash table entries: 65536 (order: 6, 262144 bytes)
[    0.000000] Inode-cache hash table entries: 32768 (order: 5, 131072 bytes)
[    0.000000] Memory: 3287480K/3604480K available (10240K kernel code, 760K rwdata, 2256K rodata, 6144K init, 559K bss, 54856K reserved, 262144K cma-reserved, 3080192K highmem)
[    0.000000] Virtual kernel memory layout:
[    0.000000]     vector  : 0xffff0000 - 0xffff1000   (   4 kB)
[    0.000000]     fixmap  : 0xffc00000 - 0xfff00000   (3072 kB)
[    0.000000]     vmalloc : 0xe0800000 - 0xff800000   ( 496 MB)
[    0.000000]     lowmem  : 0xc0000000 - 0xe0000000   ( 512 MB)
[    0.000000]     pkmap   : 0xbfe00000 - 0xc0000000   (   2 MB)
[    0.000000]     modules : 0xbf000000 - 0xbfe00000   (  14 MB)
[    0.000000]       .text : 0x(ptrval) - 0x(ptrval)   (12256 kB)
[    0.000000]       .init : 0x(ptrval) - 0x(ptrval)   (6144 kB)
[    0.000000]       .data : 0x(ptrval) - 0x(ptrval)   ( 761 kB)
[    0.000000]        .bss : 0x(ptrval) - 0x(ptrval)   ( 560 kB)
[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
[    0.000000] ftrace: allocating 32912 entries in 97 pages
[    0.000000] rcu: Hierarchical RCU implementation.
[    0.000000] NR_IRQS: 16, nr_irqs: 16, preallocated irqs: 16
[    0.000000] GIC: Using split EOI/Deactivate mode
[    0.000000] arch_timer: cp15 timer(s) running at 54.00MHz (phys).
[    0.000000] clocksource: arch_sys_counter: mask: 0xffffffffffffff max_cycles: 0xc743ce346, max_idle_ns: 440795203123 ns
[    0.000004] sched_clock: 56 bits at 54MHz, resolution 18ns, wraps every 4398046511102ns
[    0.000015] Switching to timer-based delay loop, resolution 18ns
[    0.000234] Console: colour dummy device 80x30
[    0.000262] console [tty0] enabled
[    0.000311] Calibrating delay loop (skipped), value calculated using timer frequency.. 108.50 BogoMIPS (lpj=180000)
[    0.000329] pid_max: default: 32768 minimum: 301
[    0.000502] Mount-cache hash table entries: 1024 (order: 0, 4096 bytes)
[    0.000522] Mountpoint-cache hash table entries: 1024 (order: 0, 4096 bytes)
[    0.001210] CPU: Testing write buffer coherency: ok
[    0.001680] CPU0: thread -1, cpu 0, socket 0, mpidr 80000000
[    0.002418] Setting up static identity map for 0x200000 - 0x20003c
[    0.002585] rcu: Hierarchical SRCU implementation.
[... 290 lines cut ...]
[    8.930902] br-9b31617d392f: port 1(veth26f39a1) entered disabled state
[    9.730533] bcmgenet fd580000.genet eth0: Link is Up - 1Gbps/Full - flow control off
[    9.730561] IPv6: ADDRCONF(NETDEV_CHANGE): eth0: link becomes ready
[    9.737153] 8021q: 802.1Q VLAN Support v1.8
[    9.954658] eth0: renamed from vethdca80e7
[    9.971729] IPv6: ADDRCONF(NETDEV_CHANGE): veth26f39a1: link becomes ready
[    9.971815] br-9b31617d392f: port 1(veth26f39a1) entered blocking state
[    9.971822] br-9b31617d392f: port 1(veth26f39a1) entered forwarding state
[    9.971935] IPv6: ADDRCONF(NETDEV_CHANGE): br-9b31617d392f: link becomes ready
[   11.711504] Bluetooth: Core ver 2.22
[   11.711542] NET: Registered protocol family 31
[   11.711547] Bluetooth: HCI device and connection manager initialized
[   11.711560] Bluetooth: HCI socket layer initialized
[   11.711568] Bluetooth: L2CAP socket layer initialized
[   11.711588] Bluetooth: SCO socket layer initialized
[   11.713581] Bluetooth: HCI UART driver ver 2.3
[   11.713589] Bluetooth: HCI UART protocol H4 registered
[   11.713634] Bluetooth: HCI UART protocol Three-wire (H5) registered
[   11.714763] Bluetooth: HCI UART protocol Broadcom registered
[  791.433743] OF: overlay: WARNING: memory leak will occur if overlay removed, property: /soc/i2c@7e804000/status
[  791.544905] OF: overlay: WARNING: memory leak will occur if overlay removed, property: /soc/spi@7e204000/status
```
</details>

maybe a stupid question, but should there be any vital signs on the respeaker if its pluged in correctly? like would it light a led or something? I already changed the connector twice because I was not sure - but there is no led on at the moment.

HinTak, 2020-12-03:
@h4de5 libreELEC has `pacmd` so you can use that to see if the recording part of the card is detected with `pacmd list-sources` . If that works, you might be able to use `parecord` instead of `arecord` to record off the 4-mics!

h4de5, 2020-12-03:
<details>
<summary>install.sh</summary>
```
./install.sh
bash: 0: unknown operand
Warning: /boot volume not found ..
grep: invalid option -- 'b'
BusyBox v1.31.0 (2020-10-24 15:30:09 EDT) multi-call binary.

Usage: grep [-HhnlLoqvsriwFE] [-m N] [-A/B/C N] PATTERN/-e PATTERN.../-f FILE [FILE]...

Search for PATTERN in FILEs (or stdin)

        -H      Add 'filename:' prefix
        -h      Do not add 'filename:' prefix
        -n      Add 'line_no:' prefix
        -l      Show only names of files that match
        -L      Show only names of files that don't match
        -c      Show only count of matching lines
        -o      Show only the matching part of line
        -q      Quiet. Return 0 if PATTERN is found, 1 otherwise
        -v      Select non-matching lines
        -s      Suppress open and read errors
        -r      Recurse
        -i      Ignore case
        -w      Match whole words only
        -x      Match whole lines only
        -F      PATTERN is a literal (not regexp)
        -E      PATTERN is an extended regexp
        -m N    Match up to N times per file
        -A N    Print N lines of trailing context
        -B N    Print N lines of leading context
        -C N    Same as '-A N -B N'
        -e PTRN Pattern to match
        -f FILE Read pattern from file
zcat: short read
mkdir: can't create directory '/usr/src/': Read-only file system
cp: can't create '/usr/src/seeed-voicecard-0.3/LICENSE': Path does not exist
cp: can't create '/usr/src/seeed-voicecard-0.3/Makefile': Path does not exist
cp: can't create '/usr/src/seeed-voicecard-0.3/README.md': Path does not exist
cp: can't create '/usr/src/seeed-voicecard-0.3/ac101.c': Path does not exist
cp: can't create '/usr/src/seeed-voicecard-0.3/ac101_regs.h': Path does not exist
cp: can't create '/usr/src/seeed-voicecard-0.3/ac108.c': Path does not exist
cp: can't create '/usr/src/seeed-voicecard-0.3/ac108.h': Path does not exist
cp: can't create '/usr/src/seeed-voicecard-0.3/ac108_6mic.state': Path does not exist
cp: can't create '/usr/src/seeed-voicecard-0.3/ac108_asound.state': Path does not exist
cp: can't create directory '/usr/src/seeed-voicecard-0.3/ac108_plugin': No such file or directory
cp: can't create '/usr/src/seeed-voicecard-0.3/ac10x.h': Path does not exist
cp: can't create '/usr/src/seeed-voicecard-0.3/asound_2mic.conf': Path does not exist
cp: can't create '/usr/src/seeed-voicecard-0.3/asound_4mic.conf': Path does not exist
cp: can't create '/usr/src/seeed-voicecard-0.3/asound_6mic.conf': Path does not exist
cp: can't create '/usr/src/seeed-voicecard-0.3/builddtbo.sh': Path does not exist
cp: can't create '/usr/src/seeed-voicecard-0.3/default.pa': Path does not exist
cp: can't create '/usr/src/seeed-voicecard-0.3/dkms.conf': Path does not exist
cp: can't create '/usr/src/seeed-voicecard-0.3/install.sh': Path does not exist
cp: can't create directory '/usr/src/seeed-voicecard-0.3/pulseaudio': No such file or directory
cp: can't create '/usr/src/seeed-voicecard-0.3/seeed-2mic-voicecard-overlay.dts': Path does not exist
cp: can't create '/usr/src/seeed-voicecard-0.3/seeed-2mic-voicecard.dtbo': Path does not exist
cp: can't create '/usr/src/seeed-voicecard-0.3/seeed-4mic-voicecard-overlay.dts': Path does not exist
cp: can't create '/usr/src/seeed-voicecard-0.3/seeed-4mic-voicecard.dtbo': Path does not exist
cp: can't create '/usr/src/seeed-voicecard-0.3/seeed-8mic-voicecard-overlay.dts': Path does not exist
cp: can't create '/usr/src/seeed-voicecard-0.3/seeed-8mic-voicecard.dtbo': Path does not exist
cp: can't create '/usr/src/seeed-voicecard-0.3/seeed-voicecard': Path does not exist
cp: can't create '/usr/src/seeed-voicecard-0.3/seeed-voicecard.c': Path does not exist
cp: can't create '/usr/src/seeed-voicecard-0.3/seeed-voicecard.service': Path does not exist
cp: can't create '/usr/src/seeed-voicecard-0.3/sound-compatible-4.18.h': Path does not exist
cp: can't create directory '/usr/src/seeed-voicecard-0.3/tools': No such file or directory
cp: can't create '/usr/src/seeed-voicecard-0.3/ubuntu-prerequisite.sh': Path does not exist
cp: can't create '/usr/src/seeed-voicecard-0.3/uninstall.sh': Path does not exist
cp: can't create '/usr/src/seeed-voicecard-0.3/wm8960.c': Path does not exist
[... 161 lines cut ...]
        card: 0 <alsa_card.0>
        module: 6
        properties:
                device.description = "Monitor of Built-in Audio Digital Stereo (IEC958)"
                device.class = "monitor"
                alsa.card = "0"
                alsa.card_name = "bcm2835 ALSA"
                alsa.long_card_name = "bcm2835 ALSA"
                alsa.driver_name = "snd_bcm2835"
                device.bus_path = "/devices/platform/soc/soc:audio/sound/card0"
                sysfs.path = "/devices/platform/soc/soc:audio/sound/card0"
                device.form_factor = "internal"
                device.string = "0"
                module-udev-detect.discovered = "1"
                device.icon_name = "audio-card"
```
</details>

it is not showing up, right?

<details>
<summary></summary>
```
```
</details>

HinTak, 2020-12-03:
@h4de5 okay, your dmesg is half-way to correct - these two lines means your overlay is half-loaded:

```
[  791.433743] OF: overlay: WARNING: memory leak will occur if overlay removed, property: /soc/i2c@7e804000/status
[  791.544905] OF: overlay: WARNING: memory leak will occur if overlay removed, property: /soc/spi@7e204000/status
```

your question about LEDs: no, the leds are programable and you can play with them (I have a script which makes them do a spinning rainbow...), but they don't automatically light up.

@h4de5 since LibreELEC does not have dkms, we need to do things the old-fashioned way: if you do `make`, it should hopefully create a few "*.ko". `make install` should copy them to some place under `/lib/modules/4.19.127` or `/usr/lib/kernel-overlays/base/lib/modules/4.19.127` ?

The part of interesting dmesg on raspbian looks like this:
```
[    2.224602] snd_soc_seeed_voicecard: loading out-of-tree module taints kernel.
...
[    3.452025]  ac10x i2c_id number: 1
[    3.452036]  ac10x data protocol: 0
...
[    3.836116]  ac10x codec count  : 1
[    3.836963]  ac10x i2c_id number: 0
[    3.836973]  ac10x data protocol: 0
...
[    4.165136]  ac10x codec count  : 2
...
[    4.491265] input: seed-voicecard-headset as /devices/platform/soc/fe804000.i2c/i2c-1/1-0035/input/input0
[    4.503653] ac108_set_sysclk  :24000000
[    4.518598] AC101 as Master
[    4.523353] seeed-voicecard soc:sound: ac10x-codec.1-0035 <-> fe203000.i2s mapping ok
...
[    6.542662] OF: overlay: WARNING: memory leak will occur if overlay removed, property: /soc/i2c@7e804000/status
...
[    6.901904] OF: overlay: WARNING: memory leak will occur if overlay removed, property: /soc/spi@7e204000/status
[    7.187898] OF: overlay: WARNING: memory leak will occur if overlay removed, property: /soc/i2s@7e203000/#sound-dai-cells
[    7.187968] OF: overlay: WARNING: memory leak will occur if overlay removed, property: /soc/i2s@7e203000/status
```

This is the 6-mics devices; I believe the 4-mics shows fewer ac10x lines, with a "codec count 1", and does not have the "input: seed-voicecard-headset" lines.

It appears that the libreELEC does not ship any dev tools, so unfortunately you may need to install "make", "gcc", "kernel-dev" and perhaps a few other dev-related packages before you can run "make" correctly to generate the "*.ko" files.

HinTak, 2020-12-03:
@h4de5 I had a look at volumio - it turns out to be just old raspbian, so as far as I see it should just work, if you are careful about kernel header package installation.

HinTak, 2020-12-03:
My notes - 
https://github.com/HinTak/RaspberryPi-Dev/blob/master/LibreELEC-and-Volumio.md

It looks like Volumio is perhaps easier.

If your `apt list raspberrypi-kernel` on Volumio looks similar to this:
Listing... Done
raspberrypi-kernel-headers/testing 1.20201022-1 armhf [upgradable from: 1.20200512-2]
raspberrypi-kernel/testing 1.20201022-1 armhf [upgradable from: 1.20200512-2]
```

if your `apt list -a raspberrypi-kernel` on Volumio looks like this:

```
Listing... Done
raspberrypi-kernel/testing 1.20201022-1 armhf [upgradable from: 1.20200512-2]
raspberrypi-kernel/now 1.20200512-2 armhf [installed,upgradable to: 1.20201022-1]
```

You should be able to download the matching header, 

http://archive.raspberrypi.org/debian/pool/main/r/raspberrypi-firmware/raspberrypi-kernel-headers_1.20200512-2_armhf.deb

`dpkg -i ` the downloaded file, then do `apt-mark hold raspberrypi-kernel raspberrypi-kernel-headers` and verify with `apt-mark showholds` that they are marked to stay not to auto-upgrade, and just follow ./install.sh as normal raspbian.

I'll put these in the Volumio section of the above doc.

h4de5, 2020-12-03:
I had no luck finding a way to get the build tools into kodi. 

lets assume - just for a moment - that I have no idea about building stuff with make .. what happens if I build it in a docker container that is running within libreelec or even on a separate raspbian installation on that rpi and copy them over?

so, you think installation on volumio would be easier? i just checked, and that volumio is running on a rpi3b (not a rpi4).

HinTak, 2020-12-03:
@h4de5 Volumio seems to be just older raspbian (Jesse), it may indeed be a bit too old for pi4? Not sure about this.

HinTak, 2020-12-03:
@h4de5 I think volumio would work on pi4 - while most of the system is older raspbian, the kernel is a fairly current one (20200512 as far as I can see, identical to what I am running on my kernel-version-locked current raspbian).

h4de5, 2020-12-03:
ok. in this case i'll have a look at volumio later today.
not sure - but the last time i tried to get the respeaker run there, I got stuck somewhere else..

btw. I have put `ac108_asound.state` to: `/var/media/root/var/lib/alsa/asound.state`

h4de5, 2020-12-04:
I am on volumio now.
install.sh ran through - but had many errors:

<details>
<summary>sudo ./install.sh</summary>

```
Hit http://archive.volumio.org jessie InRelease
Hit http://archive.volumio.org jessie InRelease
Hit https://download.docker.com jessie InRelease
Get:1 https://download.docker.com jessie/stable armhf Packages [4943 B]
Get:2 https://download.docker.com jessie/stable Translation-en [393 B]
Get:3 http://archive.volumio.org jessie/main Sources [7762 kB]
Get:4 https://download.docker.com jessie/stable Translation-en [393 B]
Get:5 https://download.docker.com jessie/stable Translation-en [393 B]
Get:6 https://download.docker.com jessie/stable Translation-en [393 B]
Get:7 https://download.docker.com jessie/stable Translation-en [393 B]
Ign https://download.docker.com jessie/stable Translation-en
Get:8 http://archive.volumio.org jessie/contrib Sources [64.5 kB]
Get:9 http://archive.volumio.org jessie/non-free Sources [113 kB]
Get:10 http://archive.volumio.org jessie/rpi Sources [1132 B]
Get:11 http://archive.volumio.org jessie/main armhf Packages [9539 kB]
Get:12 http://archive.volumio.org jessie/contrib armhf Packages [43.3 kB]
Get:13 http://archive.volumio.org jessie/non-free armhf Packages [88.1 kB]
Get:14 http://archive.volumio.org jessie/rpi armhf Packages [1356 B]
Get:15 http://archive.volumio.org jessie/main Sources [57.1 kB]
Get:16 http://archive.volumio.org jessie/ui Sources [17.4 kB]
Get:17 http://archive.volumio.org jessie/main armhf Packages [171 kB]
Get:18 http://archive.volumio.org jessie/ui armhf Packages [58.9 kB]
Ign http://archive.volumio.org jessie/contrib Translation-en
Ign http://archive.volumio.org jessie/main Translation-en
Ign http://archive.volumio.org jessie/non-free Translation-en
Ign http://archive.volumio.org jessie/rpi Translation-en
Ign http://archive.volumio.org jessie/main Translation-en
Ign http://archive.volumio.org jessie/ui Translation-en
Fetched 17.9 MB in 22s (787 kB/s)
Reading package lists... Done
Building dependency tree
Reading state information... Done
24 packages can be upgraded. Run 'apt list --upgradable' to see them.
Reading package lists... Done
Building dependency tree
Reading state information... Done
Package raspberrypi-kernel is not available, but is referred to by another package.
This may mean that the package is missing, has been obsoleted, or
is only available from another source

E: Package 'raspberrypi-kernel' has no installation candidate
Reading package lists... Done
Building dependency tree
Reading state information... Done
E: Unable to locate package linux-raspi
E: Unable to locate package linux-headers-raspi
E: Unable to locate package linux-image-raspi
Reading package lists... Done
Building dependency tree
Reading state information... Done
git is already the newest version.
libasound2-plugins is already the newest version.
libasound2-plugins set to manually installed.
i2c-tools is already the newest version.
The following extra packages will be installed:
  binutils cpp cpp-4.6 cpp-4.9 fakeroot gcc gcc-4.6 gcc-4.9 libasan1 libatomic1 libcloog-isl4 libfakeroot libgcc-4.9-dev libisl10 libmpc3 libmpfr4 libubsan0 linux-base
  linux-headers-3.6-trunk-common linux-headers-3.6-trunk-rpi linux-image-3.6-trunk-rpi linux-kbuild-3.6 make patch
Suggested packages:
  binutils-doc cpp-doc gcc-4.6-locales gcc-4.9-locales gcc-multilib autoconf automake libtool flex bison gdb gcc-doc libmudflap0-4.6-dev gcc-4.6-doc libgcc1-dbg libgomp1-dbg
  libquadmath-dbg libmudflap0-dbg libcloog-ppl1 libcloog-ppl0 libppl-c4 libppl12 libppl9 binutils-gold gcc-4.9-doc libitm1-dbg libatomic1-dbg libasan1-dbg liblsan0-dbg libtsan0-dbg
  libubsan0-dbg libcilkrts5-dbg linux-doc-3.6 debian-kernel-handbook fdutils make-doc ed diffutils-doc
Recommended packages:
  linux-headers-686-pae linux-headers-amd64 linux-headers-generic linux-headers linux-image
[... 619 lines cut ...]
[   54.675674] FS-Cache: Duplicate cookie detected
[   54.675682] FS-Cache: O-cookie c=50c2ed4e [p=14b63648 fl=222 nc=0 na=1]
[   54.675686] FS-Cache: O-cookie d=38bd6c2b n=73f76261
[   54.675689] FS-Cache: O-key=[8] '020001bdc0a8000a'
[   54.675720] FS-Cache: N-cookie c=ab2bce66 [p=14b63648 fl=2 nc=0 na=1]
[   54.675724] FS-Cache: N-cookie d=38bd6c2b n=8430c83f
[   54.675727] FS-Cache: N-key=[8] '020001bdc0a8000a'
[   54.675855] FS-Cache: Duplicate cookie detected
[   54.675862] FS-Cache: O-cookie c=50c2ed4e [p=14b63648 fl=222 nc=0 na=1]
[   54.675867] FS-Cache: O-cookie d=38bd6c2b n=73f76261
[   54.675869] FS-Cache: O-key=[8] '020001bdc0a8000a'
[   54.675894] FS-Cache: N-cookie c=ab2bce66 [p=14b63648 fl=2 nc=0 na=1]
[   54.675898] FS-Cache: N-cookie d=38bd6c2b n=df543971
[   54.675901] FS-Cache: N-key=[8] '020001bdc0a8000a'
[   54.679356] FS-Cache: Duplicate cookie detected
[   54.679372] FS-Cache: O-cookie c=50c2ed4e [p=14b63648 fl=222 nc=0 na=1]
[   54.679377] FS-Cache: O-cookie d=38bd6c2b n=73f76261
[   54.679380] FS-Cache: O-key=[8] '020001bdc0a8000a'
[   54.679406] FS-Cache: N-cookie c=ed757e77 [p=14b63648 fl=2 nc=0 na=1]
[   54.679411] FS-Cache: N-cookie d=38bd6c2b n=b75dfe9d
[   54.679417] FS-Cache: N-key=[8] '020001bdc0a8000a'
```
</details>

I am going through them right now.

h4de5, 2020-12-04:
this is was my headers look like:

```
apt list -a raspberrypi-kernel
Listing... Done
raspberrypi-kernel/oldstable 1.20170703-1 armhf
```

HinTak, 2020-12-04:
> this is was my headers look like:
> 
> ```
> apt list -a raspberrypi-kernel
> Listing... Done
> raspberrypi-kernel/oldstable 1.20170703-1 armhf
> ```

How old is your volumio installation? As it says 2017, it is antique...

h4de5, 2020-12-04:
I am wondering myself - its the current volumio (2.853), but I mostly updated it through the service itself - still 2017 seems a little too old.
[image]

also - not sure if this matters - this installation is running on a berryboot bootloader, with the image located on a iSCSI mount.

I tried to download those headers (only after I already tried to run install.sh)

```
$ sudo dpkg -i raspberrypi-kernel-headers_1.20170703-1_armhf.deb
[sudo] password for volumio:
Selecting previously unselected package raspberrypi-kernel-headers.
dpkg: warning: files list file for package 'libgc1c2:armhf' missing; assuming package has no files currently installed
dpkg: warning: files list file for package 'libunistring0:armhf' missing; assuming package has no files currently installed
dpkg: warning: files list file for package 'guile-2.0-libs:armhf' missing; assuming package has no files currently installed
(Reading database ... 34657 files and directories currently installed.)
Preparing to unpack raspberrypi-kernel-headers_1.20170703-1_armhf.deb ...
Unpacking raspberrypi-kernel-headers (1.20170703-1) ...
Setting up raspberrypi-kernel-headers (1.20170703-1) ...
run-parts: executing /etc/kernel/header_postinst.d/dkms 4.9.35+
Error! Bad return status for module build on kernel: 4.9.35+ (armv7l)
Consult /var/lib/dkms/seeed-voicecard/0.3/build/make.log for more information.
run-parts: executing /etc/kernel/header_postinst.d/dkms 4.9.35-v7+
Error! Bad return status for module build on kernel: 4.9.35-v7+ (armv7l)
Consult /var/lib/dkms/seeed-voicecard/0.3/build/make.log for more information.

$ cat /var/lib/dkms/seeed-voicecard/0.3/build/make.log
DKMS make.log for seeed-voicecard-0.3 for kernel 4.9.35-v7+ (armv7l)
Fri Dec  4 21:30:41 UTC 2020
make: Entering directory '/usr/src/linux-headers-4.9.35-v7+'
  LD      /var/lib/dkms/seeed-voicecard/0.3/build/built-in.o
  CC [M]  /var/lib/dkms/seeed-voicecard/0.3/build/wm8960.o
In file included from /var/lib/dkms/seeed-voicecard/0.3/build/wm8960.c:28:0:
/var/lib/dkms/seeed-voicecard/0.3/build/sound-compatible-4.18.h:19:5: warning: "__has_attribute" is not defined [-Wundef]
 #if __has_attribute(__fallthrough__)
     ^
/var/lib/dkms/seeed-voicecard/0.3/build/sound-compatible-4.18.h:19:20: error: missing binary operator before token "("
 #if __has_attribute(__fallthrough__)
                    ^
scripts/Makefile.build:293: recipe for target '/var/lib/dkms/seeed-voicecard/0.3/build/wm8960.o' failed
make[1]: *** [/var/lib/dkms/seeed-voicecard/0.3/build/wm8960.o] Error 1
Makefile:1490: recipe for target '_module_/var/lib/dkms/seeed-voicecard/0.3/build' failed
make: *** [_module_/var/lib/dkms/seeed-voicecard/0.3/build] Error 2
make: Leaving directory '/usr/src/linux-headers-4.9.35-v7+'
```

HinTak, 2020-12-04:
@h4de5 okay, for some strange reason, it is stuck at the older 4.9 kernel. Anyway, the error you saw is because of older compiler (than gcc 5 ... Current is 10, you know...) did not have __has_attribute . I just pushed a fix - please pull and see if you manage to continue compilation?

h4de5, 2020-12-04:
```
$ cat /var/lib/dkms/seeed-voicecard/0.3/build/make.log
DKMS make.log for seeed-voicecard-0.3 for kernel 4.9.35-v7+ (armv7l)
Fri Dec  4 22:22:06 UTC 2020
make: Entering directory '/usr/src/linux-headers-4.9.35-v7+'
  LD      /var/lib/dkms/seeed-voicecard/0.3/build/built-in.o
  CC [M]  /var/lib/dkms/seeed-voicecard/0.3/build/wm8960.o
In file included from /var/lib/dkms/seeed-voicecard/0.3/build/wm8960.c:28:0:
/var/lib/dkms/seeed-voicecard/0.3/build/sound-compatible-4.18.h:19:48: error: missing binary operator before token "("
 #if defined(__has_attribute) && __has_attribute(__fallthrough__)
                                                ^
scripts/Makefile.build:293: recipe for target '/var/lib/dkms/seeed-voicecard/0.3/build/wm8960.o' failed
make[1]: *** [/var/lib/dkms/seeed-voicecard/0.3/build/wm8960.o] Error 1
Makefile:1490: recipe for target '_module_/var/lib/dkms/seeed-voicecard/0.3/build' failed
make: *** [_module_/var/lib/dkms/seeed-voicecard/0.3/build] Error 2
make: Leaving directory '/usr/src/linux-headers-4.9.35-v7+'
```

should I try to update the berryboot installation?

h4de5, 2020-12-04:
did I mentioned, that volumio is running on a rpi3 (not a 4) ? maybe thats the difference?

HinTak, 2020-12-04:
Okay, made a mistake with the fix - pushed a better fix.

Likely volumio has a special pi4 build with a more current kernel? I mean, pi 4 did not exist in 2017.

h4de5, 2020-12-05:
i tried to update some more stuff. getting the latest berryboot version - it seems its running on 64bit now.

```
$ sudo dpkg -i raspberrypi-kernel-headers_1.20170703-1_armhf.deb
[sudo] password for volumio:
dpkg: warning: files list file for package 'libgc1c2:armhf' missing; assuming package has no files currently installed
dpkg: warning: files list file for package 'libunistring0:armhf' missing; assuming package has no files currently installed
dpkg: warning: files list file for package 'guile-2.0-libs:armhf' missing; assuming package has no files currently installed
(Reading database ... 65410 files and directories currently installed.)
Preparing to unpack raspberrypi-kernel-headers_1.20170703-1_armhf.deb ...
Unpacking raspberrypi-kernel-headers (1.20170703-1) over (1.20170703-1) ...
kSetting up raspberrypi-kernel-headers (1.20170703-1) ...
run-parts: executing /etc/kernel/header_postinst.d/dkms 4.9.35+
Error! Bad return status for module build on kernel: 4.9.35+ (aarch64)
Consult /var/lib/dkms/seeed-voicecard/0.3/build/make.log for more information.
run-parts: executing /etc/kernel/header_postinst.d/dkms 4.9.35-v7+
Error! Bad return status for module build on kernel: 4.9.35-v7+ (aarch64)
Consult /var/lib/dkms/seeed-voicecard/0.3/build/make.log for more information.

$ cat /var/lib/dkms/seeed-voicecard/0.3/build/make.log
DKMS make.log for seeed-voicecard-0.3 for kernel 4.9.35-v7+ (aarch64)
Sat Dec  5 00:20:05 UTC 2020
make: Entering directory '/usr/src/linux-headers-4.9.35-v7+'
  LD      /var/lib/dkms/seeed-voicecard/0.3/build/built-in.o
  CC [M]  /var/lib/dkms/seeed-voicecard/0.3/build/wm8960.o
gcc: error: unrecognized command line option ‘-mgeneral-regs-only’
scripts/Makefile.build:293: recipe for target '/var/lib/dkms/seeed-voicecard/0.3/build/wm8960.o' failed
make[1]: *** [/var/lib/dkms/seeed-voicecard/0.3/build/wm8960.o] Error 1
Makefile:1490: recipe for target '_module_/var/lib/dkms/seeed-voicecard/0.3/build' failed
make: *** [_module_/var/lib/dkms/seeed-voicecard/0.3/build] Error 2
make: Leaving directory '/usr/src/linux-headers-4.9.35-v7+'
```

HinTak, 2020-12-05:
According to the berryboot changelog, https://www.berryterminal.com/doku.php/berryboot/changelog , current berryboot is 5.4 based, same as Ubuntu and raspbian...

h4de5, 2020-12-05:
yes. thats what I hoped to get :\
but `apt list -a raspberrypi-kernel` still shows the one from 2017

the kernel only changed a little - from **4.19.49v6v7** to **4.19.127v64**

<details>
<summary>dmesg</summary>

```
[    0.000000] Booting Linux on physical CPU 0x0000000000 [0x410fd034]
[    0.000000] Linux version 4.19.127v64 (max@lynx) (gcc version 7.5.0 (Ubuntu/Linaro 7.5.0-3ubuntu1~18.04)) #1 SMP PREEMPT Fri Jun 12 22:09:51 CEST 2020
[    0.000000] Machine model: Raspberry Pi 3 Model B Rev 1.2
[    0.000000] efi: Getting EFI parameters from FDT:
[    0.000000] efi: UEFI not found.
[    0.000000] cma: Reserved 256 MiB at 0x000000001ec00000
[    0.000000] On node 0 totalpages: 221184
[    0.000000]   DMA32 zone: 3456 pages used for memmap
[    0.000000]   DMA32 zone: 0 pages reserved
[    0.000000]   DMA32 zone: 221184 pages, LIFO batch:63
[    0.000000] random: get_random_bytes called from start_kernel+0xb4/0x484 with crng_init=0
[    0.000000] percpu: Embedded 24 pages/cpu s58840 r8192 d31272 u98304
[    0.000000] pcpu-alloc: s58840 r8192 d31272 u98304 alloc=24*4096
[    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3
[    0.000000] Detected VIPT I-cache on CPU0
[    0.000000] CPU features: enabling workaround for ARM erratum 843419
[    0.000000] CPU features: enabling workaround for ARM erratum 845719
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 217728
[    0.000000] Kernel command line: coherent_pool=1M 8250.nr_uarts=0 cma=256M video=HDMI-A-1:1920x1080M@60 vc_mem.mem_base=0x3ec00000 vc_mem.mem_size=0x40000000  elevator=deadline quiet bootmenutimeout=7 datadev=iscsi qmap=de vncinstall ipv4=192.168.0.73/255.255.240.0/192.168.0.1
[    0.000000] Dentry cache hash table entries: 131072 (order: 8, 1048576 bytes)
[    0.000000] Inode-cache hash table entries: 65536 (order: 7, 524288 bytes)
[    0.000000] Memory: 588060K/884736K available (9596K kernel code, 998K rwdata, 2860K rodata, 1024K init, 1137K bss, 34532K reserved, 262144K cma-reserved)
[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
[    0.000000] ftrace: allocating 31135 entries in 122 pages
[    0.000000] rcu: Preemptible hierarchical RCU implementation.
[    0.000000] rcu:     RCU restricting CPUs from NR_CPUS=64 to nr_cpu_ids=4.
[    0.000000]  Tasks RCU enabled.
[    0.000000] rcu: Adjusting geometry for rcu_fanout_leaf=16, nr_cpu_ids=4
[    0.000000] NR_IRQS: 64, nr_irqs: 64, preallocated irqs: 0
[    0.000000] arch_timer: cp15 timer(s) running at 19.20MHz (phys).
[    0.000000] clocksource: arch_sys_counter: mask: 0xffffffffffffff max_cycles: 0x46d987e47, max_idle_ns: 440795202767 ns
[    0.000007] sched_clock: 56 bits at 19MHz, resolution 52ns, wraps every 4398046511078ns
[    0.000249] Console: colour dummy device 80x25
[    0.000278] console [tty0] enabled
[    0.000330] Calibrating delay loop (skipped), value calculated using timer frequency.. 38.40 BogoMIPS (lpj=76800)
[    0.000351] pid_max: default: 32768 minimum: 301
[    0.000756] Mount-cache hash table entries: 2048 (order: 2, 16384 bytes)
[    0.000790] Mountpoint-cache hash table entries: 2048 (order: 2, 16384 bytes)
[    0.024122] ASID allocator initialised with 32768 entries
[    0.032114] rcu: Hierarchical SRCU implementation.
[    0.040801] EFI services will not be available.
[    0.048190] smp: Bringing up secondary CPUs ...
[    0.080470] Detected VIPT I-cache on CPU1
[    0.080548] CPU1: Booted secondary processor 0x0000000001 [0x410fd034]
[    0.112593] Detected VIPT I-cache on CPU2
[    0.112646] CPU2: Booted secondary processor 0x0000000002 [0x410fd034]
[    0.144750] Detected VIPT I-cache on CPU3
[    0.144795] CPU3: Booted secondary processor 0x0000000003 [0x410fd034]
[    0.144980] smp: Brought up 1 node, 4 CPUs
[    0.144994] SMP: Total of 4 processors activated.
[    0.145010] CPU features: detected: 32-bit EL0 Support
[    0.146009] CPU: All CPU(s) started at EL2
[    0.146064] alternatives: patching kernel code
[    0.147605] devtmpfs: initialized
[    0.164535] Enabled cp15_barrier support
[    0.164572] Enabled setend support
[    0.165176] clocksource: jiffies: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 7645041785100000 ns
[    0.165206] futex hash table entries: 1024 (order: 4, 65536 bytes)
[    0.177536] xor: measuring software checksum speed
[    0.217035]    8regs     :  1144.000 MB/sec
[    0.257133]    8regs_prefetch:  1019.000 MB/sec
[... 289 lines cut ...]
[   34.583144] aufs au_warn_loopback:123:loop1[1076]: you may want to try another patch for loopback file on ext4(0xef53) branch
[   37.359178] bridge: filtering via arp/ip/ip6tables is no longer available by default. Update your scripts to load br_netfilter if you need this.
[   37.367634] Bridge firewalling registered
[   38.041071] brcmfmac: brcmf_cfg80211_set_power_mgmt: power save disabled
[   39.921128] EXT4-fs (dm-1): mounted filesystem with ordered data mode. Opts: (null)
[   40.034913] br-bde535b3b227: port 1(vethe6957a9) entered blocking state
[   40.034927] br-bde535b3b227: port 1(vethe6957a9) entered disabled state
[   40.035211] device vethe6957a9 entered promiscuous mode
[   40.035653] br-bde535b3b227: port 1(vethe6957a9) entered blocking state
[   40.035665] br-bde535b3b227: port 1(vethe6957a9) entered forwarding state
[   40.035908] br-bde535b3b227: port 1(vethe6957a9) entered disabled state
[   40.040382] br-bde535b3b227: port 1(vethe6957a9) entered blocking state
[... cut ...]
[   44.690625] br-bde535b3b227: port 1(vethe6957a9) entered disabled state
[   44.693021] eth0: renamed from veth6f228d6
[   44.716841] br-bde535b3b227: port 1(vethe6957a9) entered blocking state
[   44.716861] br-bde535b3b227: port 1(vethe6957a9) entered forwarding state
```
</details>

$ uname -a
Linux Kallisto 4.19.127v64 #1 SMP PREEMPT Fri Jun 12 22:09:51 CEST 2020 aarch64 GNU/Linux

$ cat /proc/version
Linux version 4.19.127v64 (max@lynx) (gcc version 7.5.0 (Ubuntu/Linaro 7.5.0-3ubuntu1~18.04)) #1 SMP PREEMPT Fri Jun 12 22:09:51 CEST 2020

HinTak, 2020-12-05:
You need version-matched header package - `apt list -a raspberrypi-kernel-headers`. And you should use a `arm64` header package `raspberrypi-kernel-headers_1.20170703-1_arm64.deb`? 

 The "error: unrecognized command line option ‘-mgeneral-regs-only’" comes from armhf/arm64 32-bit/64-bit confusion.

h4de5, 2020-12-05:
so.. while talking about confusion..

```
$ uname -a
Linux Kallisto 5.4.73v64 #2 SMP PREEMPT Tue Nov 3 16:11:05 CET 2020 aarch64 GNU/Linux

$ cat /proc/version
Linux version 5.4.73v64 (max@lynx) (gcc version 7.5.0 (Ubuntu/Linaro 7.5.0-3ubuntu1~18.04)) #2 SMP PREEMPT Tue Nov 3 16:11:05 CET 2020

$ sudo apt list -a raspberrypi-kernel
Listing... Done
raspberrypi-kernel/oldstable 1.20170703-1 armhf
```

HinTak, 2020-12-05:
@h4de5 I think that is berryboot - it uses its own kernel (outside of the volumio/raspbian packaging system) to boot the rest of volumio. It is not a bootloader like the one on the x86 desktop. It is more like customisation of kernel + initramfs .

It is an interesting journey for me - didn't know about any of these other distros, and I have been using Linux since it was only about 5 years old! (it is just over 30 now).

h4de5, 2020-12-05:
I liked berryboot because it would allow me to install, change, add and reuse multiple distros on the same sdcard. I could just add another os and play around, or clone an existing partition before i try out something new. With the iscsi setup backups are even easier.

The downsides seems to surface slowly only now. I often needed special images that allow updates (like volumio and kodi: https://github.com/macmpi/berryboot-scripts)
Also HATs like that often ask for additional efforts.

Good to hear that you to learned something ;)

Any more ideas i could go after?

HinTak, 2020-12-05:
@h4de5 it is a little painful, but since berryboot runs its own kernel, you really needs to have a customized berryboot image to add new kernel modules for hardware drivers such as this. So "HATs ask for additional efforts" is exactly it. For example https://github.com/maxnet/berryboot/issues/631 is somewhat a similar problem.

h4de5, 2020-12-05:
@HinTak Kodi is running no noobs. would this help?

HinTak, 2020-12-05:
This is more directly relevant https://github.com/maxnet/berryboot/issues/576

h4de5, 2020-12-05:
mmh. i think this is part of the upgrade procedure (to manually extract the shared modules)

I did this yesterday as well: https://github.com/maxnet/berryboot/issues/666#issuecomment-739083724

HinTak, 2020-12-05:
@h4de5 stock volumio (without berryboot) is easiest, I think; berryboot (with anything) requires some customization of the berryboot build process and is next. LibreELEC is missing a lot of dev tools so hardest.

Noobs isn't really relevant - it installs the others as they are.

Berryboot seems small enough, I might have a go at inserting the respeaker driver in there...

h4de5, 2020-12-05:
and this would be possible? adding a driver?

my issue with berryboot is, that both - the kodi and the volumio on the rpi3 - have other partitions I am currently using. so in this case i may have to put it on a different place altogether. I have two dedicated volumios, but those are running on a rpi0 and they do have HATs installed already (a PHAT Dac for sound output: https://de.pinout.xyz/pinout/phat_dac).

how do you see the chances that they work well together, after the last two days? 🤣 

if not, I may have to just postbone the usage of this piece for now. I'll add one (propably) last device in the future anyway and I was thinking about having a normal raspian installation and run volumio as app or from a docker image. I mean, this would be the most straight forward solution anyway, right?

HinTak, 2020-12-05:
@h4de5 I suspect that how berryboot works, is booting its own kernel while having a (small) addition to the full OSes it boots by adding a "/lib/modules/<berrybootkernel>" directory to it, plus other small scripts. So it should be possible to add the respeaker modules there (or other out-of-tree driver like exfat in the above report), if you customize the berryboot build. I'll need to take apart the current berryboot image to be sure.

Devices with in-tree drivers (I guess your PHAT DAC is) is much easier, but the respeaker driver will be never be in-tree ... See
https://github.com/respeaker/seeed-voicecard/issues/213

h4de5, 2020-12-05:
The problem with the phat dac is that they may use the same gpio pins.. at least thats what i expect after the last two days ;)

HinTak, 2020-12-05:
In-tree drivers only requires additing a line config.txt to activate (and removing to deactivate). Out-of-tree driver needs to adds a few things to /lib/modules/<kernelversion> , in addition to editing config.txt . Berryboot is tricky because /lib/modules itself comes from the OSes (volumio) but /lib/modules/<kernel version> from berryboot. 

dkms and other stuff we covered earlier tries to build and add the respeaker driver to /lib/modules/<kernelversion>, but it requirea compiler and also other relevant stuff (kernel headers) under `/lib/modules/<version>/build/`.

Dkms is more or less jus copying the respeaker source to /usr/src and "make; make install" automatically for each new kernel. (hence if you are happy to stay with the same kernel forever, you can live without it)

h4de5, 2020-12-05:
Have a look at `shared.img` in the berryboot package. During an upgrade I manually extract this file on the first start using berryboots own tty terminal. It holds the modules and the kernel files. If its enough to place them there before volumio can access it, it should work.

HinTak, 2020-12-05:
Yes, that looks like it - I also just unzipped berryboot, and it looks like what I thought - it is a custom kernel to run the rest of other OSes. So it probably can be rebuilt with the respeaker driver modules included; then it is easier editing config.txt and adding the other scripts on the rest of the OSes.

HinTak, 2020-12-08:
@h4de5 backup your stuff and have a go with this ? https://github.com/HinTak/berryboot/releases . You mostly want the +respeaker.zip (the *.sig are the accompanying gpg signatures - ignore if you don't know what that is; not important) , in fact perhaps even less - you just need "kernel_rpi64.img" and "shared.img" . They need to be a matching pair - I built from berryboot current, so it is more recent that 5.4.73, and it is 5.4.81:

```
Linux version 5.4.81v64 (Hin-Tak@localhost.localdomain) (gcc version 10.2.1 20200826 (Red Hat Cross 10.2.1-3) (GCC)) #1 SMP PREEMPT Tue Dec 8 00:15:44 GMT 2020
```

Please backup your boot folder before you try this (so that you can roll back to your known good berryboot). Then you need a `dtoverlay=seeed-4mic-voicecard` line in config.txt . The rest is in userland and in the rest of the OS (alsa in volumio etc).

h4de5, 2020-12-08:
and its working in volumio the rhasspy assistant.
[image]

thank you for your help! couldn't have done it without that.

shall we try to convince berryboot to include your drivers to their upstream?

HinTak, 2020-12-08:
Cool! I'd be interested to see what you need to do on the OS side (I think it is probably just blindly running install.sh or most of it, and igoring errors...) , and any anomalies when using. And click donate if you feel generous :-).

I have already got a pull on updating the Readme on the 64-bit vanilla build instruction - 
https://github.com/maxnet/berryboot/pull/674

I have collected a few patches for Berryboot at 
https://github.com/HinTak/RaspberryPi-Dev/tree/master/patches - mostly because my desktop (fedora 33, a few weeks old) is a lot newer than the berryboot guy's (it seems he is using Ubuntu 18.04, which is 2018 spring), compiler-wise. I just need to write up some instructions from my notes to new arbitrary out-of-tree drivers to berryboot. It is basically build berryboot (which took about 4 hours) plus some extra steps at the end (making additional content for shared.img and re-generate it).

h4de5, 2020-12-08:
@HinTak I have setup now a completely new volumio on berryboot with your image.
which branch should I try ? the v5.5 or the v4.19?

HinTak, 2020-12-08:
@h4de5 v5.5, but it doesn'y really matter as the difference is in the c code (the kernel modules) which is already in the modified berryboot. In fact you probably don't need to do anything other than the extra line of "dtoverlay=..." in config.txt.

HinTak, 2020-12-08:
@h4de5 I think at most you might want to install the i2c-tools, but you can almost certainly skip all of the beginning of install.sh until the "#set dtoverlays" line. I think you can delete everything after the first line "#!/bin/bash" to the "#set dtoverlays" line. You don't need to do the last 4 lines about seeed-voicecard service ,  as all the service does (in the middle of the "seeed-voicecard") is to set alsa's config and default volume. (asound.conf and asound.state) based on what card is detected - since you know which card you have, you can just do the two / three lines once and for all.

h4de5, 2020-12-08:
@HinTak : it was showing up right after reboot, so no additional config or setup was necessary for that at least. but i could not get it working. I tried to combine it first with a hifiberry dacplus and after that i tried to use it allone on a rpi4, without any luck.

now i am back to the original image, device and overall setup that was working 8hours ago .. but I can't get any sound in.
the device still shows up in i2cdetect - even though I am pretty sure it showed as `3b` instead of `UU` 

```
 sudo i2cdetect -y 1 0x3b 0x3b
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:
10:
20:
30:                                  UU
40:
50:
60:
70:
```

<details>
<summary>dmesg</summary>

```
[    0.000000] Booting Linux on physical CPU 0x0000000000 [0x410fd034]
[    0.000000] Linux version 5.4.81v64 (Hin-Tak@localhost.localdomain) (gcc version 10.2.1 20200826 (Red Hat Cross 10.2.1-3) (GCC)) #1 SMP PREEMPT Tue Dec 8 00:15:44 GMT 2020
[    0.000000] Machine model: Raspberry Pi 3 Model B Rev 1.2
[    0.000000] efi: Getting EFI parameters from FDT:
[    0.000000] efi: UEFI not found.
[    0.000000] Reserved memory: created CMA memory pool at 0x000000001ec00000, size 256 MiB
[    0.000000] OF: reserved mem: initialized node linux,cma, compatible id shared-dma-pool
[    0.000000] On node 0 totalpages: 221184
[    0.000000]   DMA zone: 3456 pages used for memmap
[    0.000000]   DMA zone: 0 pages reserved
[    0.000000]   DMA zone: 221184 pages, LIFO batch:63
[    0.000000] percpu: Embedded 31 pages/cpu s89304 r8192 d29480 u126976
[    0.000000] pcpu-alloc: s89304 r8192 d29480 u126976 alloc=31*4096
[    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3
[    0.000000] Detected VIPT I-cache on CPU0
[    0.000000] CPU features: detected: ARM erratum 845719
[    0.000000] CPU features: detected: ARM erratum 843419
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 217728
[    0.000000] Kernel command line: coherent_pool=1M 8250.nr_uarts=0 snd_bcm2835.enable_compat_alsa=0 snd_bcm2835.enable_hdmi=1 video=Composite-1:720x480@60i vc_mem.mem_base=0x3ec00000 vc_mem.mem_size=0x40000000  elevator=deadline quiet bootmenutimeout=7 datadev=iscsi qmap=de vncinstall ipv4=192.168.0.73/255.255.240.0/192.168.0.1
[    0.000000] Dentry cache hash table entries: 131072 (order: 8, 1048576 bytes, linear)
[    0.000000] Inode-cache hash table entries: 65536 (order: 7, 524288 bytes, linear)
[    0.000000] mem auto-init: stack:off, heap alloc:off, heap free:off
[    0.000000] Memory: 584540K/884736K available (10812K kernel code, 1022K rwdata, 3488K rodata, 960K init, 1224K bss, 38052K reserved, 262144K cma-reserved)
[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
[    0.000000] ftrace: allocating 32483 entries in 127 pages
[    0.000000] rcu: Preemptible hierarchical RCU implementation.
[    0.000000] rcu:     RCU restricting CPUs from NR_CPUS=256 to nr_cpu_ids=4.
[    0.000000]  Tasks RCU enabled.
[    0.000000] rcu: RCU calculated value of scheduler-enlistment delay is 25 jiffies.
[    0.000000] rcu: Adjusting geometry for rcu_fanout_leaf=16, nr_cpu_ids=4
[    0.000000] NR_IRQS: 64, nr_irqs: 64, preallocated irqs: 0
[    0.000000] random: get_random_bytes called from start_kernel+0x334/0x4c8 with crng_init=0
[    0.000000] arch_timer: cp15 timer(s) running at 19.20MHz (phys).
[    0.000000] clocksource: arch_sys_counter: mask: 0xffffffffffffff max_cycles: 0x46d987e47, max_idle_ns: 440795202767 ns
[    0.000006] sched_clock: 56 bits at 19MHz, resolution 52ns, wraps every 4398046511078ns
[    0.000256] Console: colour dummy device 80x25
[    0.000286] printk: console [tty0] enabled
[    0.000340] Calibrating delay loop (skipped), value calculated using timer frequency.. 38.40 BogoMIPS (lpj=76800)
[    0.000362] pid_max: default: 32768 minimum: 301
[    0.000539] LSM: Security Framework initializing
[    0.000800] Mount-cache hash table entries: 2048 (order: 2, 16384 bytes, linear)
[    0.000834] Mountpoint-cache hash table entries: 2048 (order: 2, 16384 bytes, linear)
[    0.002311] Disabling memory control group subsystem
[    0.004419] ASID allocator initialised with 32768 entries
[    0.004698] rcu: Hierarchical SRCU implementation.
[    0.005471] EFI services will not be available.
[    0.006057] smp: Bringing up secondary CPUs ...
[    0.007283] Detected VIPT I-cache on CPU1
[... 434 lines cut ...]
[  107.695098] x5 : 0000000000000000 x4 : 0000000000000000
[  107.696270] x3 : 00000000ffffffff x2 : ffffffc024029000
[  107.697402] x1 : 0000000000000000 x0 : 0000000000000000
[  107.698496] Call trace:
[  107.699546]  preempt_count_sub+0x88/0xd0
[  107.700568]  _raw_spin_unlock_irqrestore+0x44/0x90
[  107.701576]  ac108_trigger+0xe0/0x154 [snd_soc_ac108]
[  107.702595]  snd_soc_dai_trigger+0x4c/0x60 [snd_soc_core]
[  107.703575]  soc_pcm_trigger+0x44/0x110 [snd_soc_core]
[  107.704524]  snd_pcm_do_start+0x4c/0x5c [snd_pcm]
[  107.705438]  snd_pcm_action_single+0x50/0xa4 [snd_pcm]
[  107.706322]  snd_pcm_action+0xb8/0xd0 [snd_pcm]
[  107.707173]  snd_pcm_action_lock_irq+0x90/0xc0 [snd_pcm]
[  107.708004]  snd_pcm_common_ioctl+0x4b0/0x1260 [snd_pcm]
[  107.708811]  snd_pcm_ioctl_compat+0x9f0/0xdcc [snd_pcm]
[  107.709596]  __arm64_compat_sys_ioctl+0x3f8/0x10bc
[  107.710401]  el0_svc_common.constprop.0+0x7c/0x1b4
[  107.711206]  el0_svc_compat_handler+0x30/0x60
[  107.712008]  el0_svc_compat+0x8/0x10
[  107.712798] ---[ end trace 5e7b8e772f50a3b3 ]---

```
</details>

now I am really afraid I might have bricked it while mounting or something ..

HinTak, 2020-12-08:
Oh, you are unfortunately seeing this issue - https://github.com/respeaker/seeed-voicecard/issues/251 - seeed studio staff (the other party in the conversation) thinks it is just  a performance issue, but it is quite serious. For some reason it seems to come up only for some usage...

h4de5, 2020-12-08:
is this something that can be fixed? or how does this happen?

HinTak, 2020-12-09:
@h4de5 it seems to be a flaw that was always there, but gotten more serious with newer kernel and/or 64-bitness, and also only affect some people/usage. Anyway, I think I'd like to try building berryboot in 32-bit mode, that might help. Give me a day. #251 is most certainly fixable, but it will take some quite drastic changes, and much efforts, so I'd rather seeed studio people do it...

h4de5, 2020-12-09:
@HinTak I am not sure if I understand - you say it is a software related bug, right? so its not physically broken? cause I am really worried. I have restartet the rpi yesterday like 50 times - and I could not get it to work again. still - i have left the rpi running with the mic open for a full day now and this i2s sync error come up only three times according to dmesg.

HinTak, 2020-12-09:
@h4de5 it is most certainly a software related bug, but is only triggered by some (unknown) types of access patterns and loads. The general circumstances is that the CPU is running a certain path in the driver when it is asked to put it aside to do something else (as in any modern multitasking multi-CPU systems). So it depends on what other things are running, and how are you using the sound device. The problem seems to affect 32-bit build less; my guess is that scheduling and the nature of how to switch tasks are both different on 32-bit vs 64-bit, even on the same hardware. Besides the fact that the raspbian people still recommend using the 32-bit OSes, while others like Ubuntu and Berryboot had jumped ahead to 64-bit for its better performance - I suspect the raspbian people probably know there are some (not publicized) issues with moving to 64-bit.

Anyway, I  am about half-way into the 32-bit build; will post when I finish.

HinTak, 2020-12-09:
32-bit builds are up - https://github.com/HinTak/berryboot/releases . The usual - you need to add a line to config.txt . Let me know if it works better or not.

HinTak, 2020-12-10:
The new doc is up as https://github.com/HinTak/RaspberryPi-Dev/blob/master/Customizing-Berryboot.md about how to add the respeaker drivers to berryboot .

h4de5, 2020-12-11:
@HinTak I am not sure what I am doing wrong, but I the rpi4 cannot boot with that image. 

I removed all files except the config/cmdline/udev/iscsi from the sdcard and copied your files on it.

boot says:

```
...
mmc-bmc2835 fe300000.mmcnr: mmc-bcm2835: Unable to initialise DMA Channel. Falling back to PIO
...
/dev/mmcblk0p1: Can't open blockdev
/dev/mmcblk0: Can't open blockdev
...
Error finding or mouting boot partition. :-( 
```

HinTak, 2020-12-11:
How does it differ from this https://github.com/maxnet/berryboot/issues/673 ?

[25 more replies]

closed 2020-12-24 by h4de5