respeaker/seeed-voicecard #271: Kernel 5.4.79 breaks udev support

xBelladonna, 2020-12-09:
I'm using a 4-mic array for Raspberry Pi and I recently upgraded my kernel to 5.4.79-v7+. dkms upgraded kernel modules automatically and installation went fine, however I'm using pulseaudio to share the device out over the network and noticed that it was not being detected anymore. I decided to perform a full re-installation of the OS and went through all installation steps from scratch but faced the same problem: pulseaudio does not automatically detect the 4-mic array but it is detected by ALSA just fine.

I tried loading pulseaudio's ALSA source module manually (`pactl load-module module-alsa-source device=ac108`) and it works fine. I then tried switching pulseaudio from using udev detection to static hardware detection by changing pulseaudio default startup script (changed `load-module module-udev-detect` to `load-module module-detect`) and it also discovers the mic array, so it seems this is a problem involving udev somehow. 

I'm currently working around the problem by manually loading the ALSA source module as part of the default pulseaudio startup script. I am inexperienced in this area of development and am happy to perform any debugging steps necessary to provide more information however I'm not quite sure where to start and would appreciate some pointers.

HinTak, 2020-12-09:
Presumably you had another device or older system that works. You can use `udevadm monitor ...` to look at udev events to see how it looks on a working system, compared to one where it does not.

Disclaimer: I don't work for seeed studio....

AIWintermuteAI, 2021-08-13:
Hi @xBelladonna !
I have tested 4mic reSpeaker multiple times recently with the following setup:

Raspberry Pi 4 1 Gb
5.10.17-v7l+ kernel
41a09a3
reSpeaker 4-mic hat for raspberry pi

The recording function worked normally for example recording command specified in wiki article
https://wiki.seeedstudio.com/ReSpeaker_4_Mic_Array_for_Raspberry_Pi/

If the issue still persists for you, could you share more details?

AIWintermuteAI, 2021-09-02:
Closing the issue due to the lack of activity.
Feel free to re-open if issue persists.