respeaker/seeed-voicecard #273: amixer breaks recording

slesinger, 2020-12-18:
Hello,
I have 4 Mic Array and RPI Zero. Install was smooth and arecord working fine until I set volumes with following commands:
```
amixer -c 2 set 'ADC1 PGA gain' capture 61% unmute
amixer -c 2 set 'ADC2 PGA gain' capture 61% unmute
amixer -c 2 set 'ADC3 PGA gain' capture 61% unmute
amixer -c 2 set 'ADC4 PGA gain' capture 61% unmute
amixer -c 2 set 'CH1 digital volume' capture 81% unmute
amixer -c 2 set 'CH2 digital volume' capture 81% unmute
amixer -c 2 set 'CH3 digital volume' capture 81% unmute
amixer -c 2 set 'CH4 digital volume' capture 81% unmute
```
Since then, I always get recording which you can see on this screenshot, regardles how much loud I speak: https://paste.pics/B2XEW
I reinstalled from scratch few times without luck. Only when I bought another RPI 3 device it worked again until I have executed the amixer command above. The sad story repeated.
LEDs are working fine.
Would you have an idea what's wrong?
Thank you in advance

AIWintermuteAI, 2021-08-20:
Hi, @slesinger !

I tried to reproduce the problem with the following setup:
Raspberry Pi Zero W
5.10.52+ kernel
https://github.com/respeaker/seeed-voicecard/commit/d840f77541f9f1ff64097a2aec274d549d6e8a23

Few things that are somewhat wrong:
1) According to ALSA command line utility ```alsamixer```, all 4 channels ADC PGA gain is set to 0 by default for 4-mic reSpeaker. So, there is no need to change that settings.
[image]
 2) If you use Raspberry Pi Zero and 4-mic reSpeaker, under normal circumstances, your sound card number should be 0. Executing commands from your message throws an error for me here:
```
pi@raspberrypi:~ $ amixer -c 2 set 'CH1 digital volume' capture 90% unmute
Invalid card number.
Usage: amixer <options> [command]
```
But changing -c arguments from 2 to 0 works:
```
pi@raspberrypi:~ $ amixer -c 0 set 'CH1 digital volume' capture 90% unmute
Simple mixer control 'CH1 digital volume',0
  Capabilities: volume volume-joined
  Playback channels: Mono
  Capture channels: Mono
  Limits: 0 - 255
  Mono: 230 [90%]
pi@raspberrypi:~ $ amixer -c 0 set 'CH2 digital volume' capture 90% unmute
Simple mixer control 'CH2 digital volume',0
  Capabilities: volume volume-joined
  Playback channels: Mono
  Capture channels: Mono
  Limits: 0 - 255
  Mono: 230 [90%]
pi@raspberrypi:~ $ amixer -c 0 set 'CH3 digital volume' capture 90% unmute
Simple mixer control 'CH3 digital volume',0
  Capabilities: volume volume-joined
  Playback channels: Mono
  Capture channels: Mono
  Limits: 0 - 255
  Mono: 230 [90%]
pi@raspberrypi:~ $ amixer -c 0 set 'CH4 digital volume' capture 90% unmute
Simple mixer control 'CH4 digital volume',0
  Capabilities: volume volume-joined
  Playback channels: Mono
  Capture channels: Mono
  Limits: 0 - 255
  Mono: 230 [90%]
pi@raspberrypi:~ $ alsamixer
pi@raspberrypi:~ $ arecord -Dac108 -f S32_LE -r 16000 -c 4 hello.wav
Recording WAVE 'hello.wav' : Signed 32 bit Little Endian, Rate 16000 Hz, Channels 4
```
I tweaked sound volumes both with alsamixer and using amixer command - all worked great. Just make sure you don't change ADC PDG gain and don't set digital volume too low - 81 was fairly quiet, 50 resulted in no sound at all.

HinTak, 2021-08-20:
The device numbering ie `-c 2` changed between 4.19 and 5.x, or rather depends on a few kernel module options to do with multiseat and multiple hdmi output on newer pi's (see https://www.raspberrypi.org/forums/viewtopic.php?f=28&t=249204) . @slesinger likely read the commands from outdated instructions without understanding what they do.

And the instructions wherever he reads it from needs updating...

AIWintermuteAI, 2021-08-20:
Right, thank you for the explanation, @HinTak .
I checked https://github.com/Seeed-Studio/wiki-documents/search?q=amixer for instructions on that in wiki, it doesn't seem we mention that anywhere there. Probably he has found it somewhere else on the Internet.

closed 2021-09-02 by AIWintermuteAI