respeaker/seeed-voicecard #274: Seeed 4-mic linear memory leak overlay

faaafo, 2020-12-22:
What can I do here?

`journalctl -b -p 3`

    Dec 22 18:45:20 raspberrypi kernel: OF: overlay: WARNING: memory leak will occur if overlay removed, property: /soc/i2c@7e804000/status
    Dec 22 18:45:20 raspberrypi kernel: OF: overlay: WARNING: memory leak will occur if overlay removed, property: /soc/spi@7e204000/status
    Dec 22 18:45:20 raspberrypi kernel: OF: overlay: WARNING: memory leak will occur if overlay removed, property: /soc/i2s@7e203000/#sound-dai-cells
    Dec 22 18:45:20 raspberrypi kernel: OF: overlay: WARNING: memory leak will occur if overlay removed, property: /soc/i2s@7e203000/status
    Dec 22 18:45:20 raspberrypi kernel: OF: overlay: WARNING: memory leak will occur if overlay removed, property: /soc/i2c@7e804000/status
    Dec 22 18:45:20 raspberrypi kernel: OF: overlay: WARNING: memory leak will occur if overlay removed, property: /soc/sound/compatible
    Dec 22 18:45:20 raspberrypi kernel: OF: overlay: WARNING: memory leak will occur if overlay removed, property: /soc/sound/seeed-voice-card,format
    Dec 22 18:45:20 raspberrypi kernel: OF: overlay: WARNING: memory leak will occur if overlay removed, property: /soc/sound/seeed-voice-card,name
    Dec 22 18:45:20 raspberrypi kernel: OF: overlay: WARNING: memory leak will occur if overlay removed, property: /soc/sound/status
    Dec 22 18:45:20 raspberrypi kernel: OF: overlay: WARNING: memory leak will occur if overlay removed, property: /soc/sound/seeed-voice-card,bitclock-ma
    Dec 22 18:45:20 raspberrypi kernel: OF: overlay: WARNING: memory leak will occur if overlay removed, property: /soc/sound/seeed-voice-card,frame-maste
    Dec 22 18:45:20 raspberrypi kernel: OF: overlay: WARNING: memory leak will occur if overlay removed, property: /soc/sound/seeed-voice-card,channels-pl
    Dec 22 18:45:20 raspberrypi kernel: OF: overlay: WARNING: memory leak will occur if overlay removed, property: /soc/sound/seeed-voice-card,channels-ca

---

    Raspberry Pi 3 Model B Plus Rev 1.3\0
    5.4.79-v7+

HinTak, 2020-12-22:
The message is harmless - memory is leaked if and only if you plug and unplug the device repeatedly, but since you cannot and should not unplug it while the os is running, the message is meaningless.

AIWintermuteAI, 2021-07-30:
Closing the issue as it seems it was resolved. Feel free to re-open if you have issues because of these messages.
```
Dec 22 18:45:20 raspberrypi kernel: OF: overlay: WARNING: memory leak will occur if overlay removed, property: /soc/i2c@7e804000/status
Dec 22 18:45:20 raspberrypi kernel: OF: overlay: WARNING: memory leak will occur if overlay removed, property: /soc/spi@7e204000/status
Dec 22 18:45:20 raspberrypi kernel: OF: overlay: WARNING: memory leak will occur if overlay removed, property: /soc/i2s@7e203000/#sound-dai-cells
Dec 22 18:45:20 raspberrypi kernel: OF: overlay: WARNING: memory leak will occur if overlay removed, property: /soc/i2s@7e203000/status
Dec 22 18:45:20 raspberrypi kernel: OF: overlay: WARNING: memory leak will occur if overlay removed, property: /soc/i2c@7e804000/status
Dec 22 18:45:20 raspberrypi kernel: OF: overlay: WARNING: memory leak will occur if overlay removed, property: /soc/sound/compatible
Dec 22 18:45:20 raspberrypi kernel: OF: overlay: WARNING: memory leak will occur if overlay removed, property: /soc/sound/seeed-voice-card,format
Dec 22 18:45:20 raspberrypi kernel: OF: overlay: WARNING: memory leak will occur if overlay removed, property: /soc/sound/seeed-voice-card,name
Dec 22 18:45:20 raspberrypi kernel: OF: overlay: WARNING: memory leak will occur if overlay removed, property: /soc/sound/status
Dec 22 18:45:20 raspberrypi kernel: OF: overlay: WARNING: memory leak will occur if overlay removed, property: /soc/sound/seeed-voice-card,bitclock-ma
Dec 22 18:45:20 raspberrypi kernel: OF: overlay: WARNING: memory leak will occur if overlay removed, property: /soc/sound/seeed-voice-card,frame-maste
Dec 22 18:45:20 raspberrypi kernel: OF: overlay: WARNING: memory leak will occur if overlay removed, property: /soc/sound/seeed-voice-card,channels-pl
Dec 22 18:45:20 raspberrypi kernel: OF: overlay: WARNING: memory leak will occur if overlay removed, property: /soc/sound/seeed-voice-card,channels-ca
```

closed 2021-07-30 by AIWintermuteAI

HinTak, 2021-07-30:
@AIWintermuteAI btw, the reported issue is not fixed or resolved: as I commented above, there is a genuine one-off small memory loss when the driver is first loaded (after boot up). It can and probably should be fixed. However, since it is per boot or per load, and the typical user only needs to load the driver once per boot, losing track of a few bytes per boot does not matter. Unless you see it more than once. People who are likely to see the message and the loss more than once are people like you or me, who are in the driver development side of things and would repeatedly unload and reload (a slightly modified version of) the driver modules frequently. So this is just an small issue unlikely to affect any users in a real way, other than seeing a worrying message about memory loss per boot. The loss is genuine and real, but only per boot. (ie when your machine boots up, with fully available memory, you lose a few bytes at the beginning, but don't continue losing any further for days and months; until you reboot and get all your memory back again, and the process repeats)

AIWintermuteAI, 2021-08-01:
Thanks, I'll make a mental note to add it to Wiki FAQ then.